T-Troll/alienfx-tools #188: Fan Control Taskbar Requests

TheSQLGuru, 2022-08-25:
1) Please provide a user setting to send the application to the taskbar tray when it is closed. This will keep the application open and doing its job and also be easily accessible.

2) Please provide a right-click for the application in both the taskbar and the tray that will drop down the 6 Power Mode choices to make it easier to switch between modes.

T-Troll, 2022-08-26:
I'll think about it, but for now you can use keyboard shortcut `CTRL+ALT+0..6` (0 - manual).

T-Troll, 2022-08-29:
Check [this test version](https://github.com/T-Troll/alienfx-tools/files/9448287/alienfan-gui.zip).

Is it OK for you?

TheSQLGuru, 2022-08-30:
1) The right-click popup intercepts the Windows key, and my laptop's function key. That seems inappropriate, and it prevented me from taking a screenshot.

2) The screenshot would have shown that the Power Mode list only contained Manual. There were spaces for the others, but they had no text - just blank space. 

3) Changing mode via the full application did move the check mark to the appropriate slot when I rechecked the popup.

4) Changing the mode via the popup seemed to function. I say seem because it never changed the actual mode in the application - regardless of whether the application was opened or minimized when the setting was changed. I did observe fan changes though, and the check mark moved as expected. 

Obviously changes in either the app or the popup need to be properly reflected in real time in the other modality. Fill in the text slots, and eliminate the key press intercept, and we should be good to go.

A good start, I would say.

T-Troll, 2022-08-30:
1. Check your system. App does not intercept anything except F17.
2. Yes, it's a bug. But do you still use "Level N" instead of rename it to something more human?
3. Ok.
4. Yes, i forget to set mode and checkbox state after restore (i dont' need to do it before).

T-Troll, 2022-08-31:
1. It's hello from Windows. Some keys blocked if any context menu active.

The rest are fixed, check [7.0.4.1](https://github.com/T-Troll/alienfx-tools/releases/tag/7.0.4.1)

TheSQLGuru, 2022-09-02:
Indeed they are! Kudos sir.

T-Troll, 2022-09-03:
Power modes order is the same as into AWCC (Except manual).
You can also check HWINFO/run tests for PLs.
Usually, one of them clearly hearable - it's Performance/Full speed.

I close this task for now.

closed 2022-09-03 by T-Troll