T-Troll/alienfx-tools #189: Taskbar Tray Icon Right-Click Crashes Fan Control

TheSQLGuru, 2022-08-25:
Win 10 x64 21H2 fully patched. App version v7.0.3

When the app has been minimized to the tray, a right-click (hoping to be able to change power modes - reference the feature request I just made) caused the app to crash.

This is repeatable.

TheSQLGuru, 2022-08-26:
I realize that the app may not be "crashing" but may rather just be "exiting gracefully - just unexpectedly". :-D

T-Troll, 2022-08-26:
Read `Alienfan-GUI` manual - it works as designed for now. Right click on tray menu close app.

closed 2022-08-26 by T-Troll

TheSQLGuru, 2022-09-02:
Well, this item is moot now anyway, because a tray icon right-click now opens the dialog that allows changing the power level. Which is good, because right-click to close a tray app is VERY non-standard! :-D

T-Troll, 2022-09-03:
No, it's quite common for menu-less apps.