T-Troll/alienfx-tools #190: The app wont start corretely after enable the option "Start with Windows" in Alienfx-gui.exe

mario6714, 2022-08-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Open Alienfx-gui.exe and tab settings'
2. Click on 'Start with Windows'
3. Scroll down to '....'
4. See error

**Expected behavior**
The icon in the tray for make changes or update the configuration in the app

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]

**System (please complete the following information):**
 - Model: Dell G15 5510
 - Model details: [e.g. per-key rgb keyboard]
 - OS: Windows 11 Pro Version 21H2 22000.856

**Additional context**
Add any other context about the problem here.

mario6714, 2022-08-29:
Update: I just enabled "start with windows" in alienfx-gui.exe and disabled the option "start with windows" in Alienfx-gui.exe and reboot like 3 times and the app alienfx-gui.exe starts fine

mario6714, 2022-08-29:
Update: I just enabled "start with windows" in alienfan-gui.exe and disabled the option "start with windows" in Alienfx-gui.exe and reboot like 3 times and the app alienfan-gui.exe starts fine

T-Troll, 2022-08-29:
Seems like you use bugged set (first 7.0.0 have a bug there) and don't read release notes for fix - i recommend to do this into it.

closed 2022-08-29 by T-Troll

mario6714, 2022-08-30:
This?

Application behavior control block at right:

    "Start with Windows" - Start application at Windows start. It will not work if application request run as admin level (see below) (default - off).