T-Troll/alienfx-tools #191: GMode dont work

mario6714, 2022-08-29:
**Describe the bug**
G-Mode don't work after i enable G-Mode in the app

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'alienfx-gui.exe in tab "Fans and Power" or in alienfan-gui.exe'
2. Click on 'G-Mode'
3. Scroll down to '....'
4. See error

**Expected behavior**

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]

**System (please complete the following information):**
 - Model: Dell G15 5510
 - Model details: [e.g. per-key rgb keyboard]
 - OS: Windows 11 Pro 21H2 22000.856

**Additional context**
Add any other context about the problem here.

T-Troll, 2022-08-29:
Please download this CLI - [alienfan-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9441618/alienfan-cli.zip)
Unpack it, open administrator CMD and run `alienfan-cli.exe test=2,2 test=3,2`.
Share output with me.

mario6714, 2022-08-29:
Here:

C:\alienfx-tool\Test>alienfan-cli.exe test=2,2 test=3,2
AlienFan-CLI v7.0.2.1
Supported hardware detected, 2 fans, 9 sensors, 5 power states, G-Mode.
command 2 arg 2 - result 2933
command 3 arg 2 - result 1200

mario6714, 2022-08-29:
C:\alienfx-tool\Test>alienfan-cli.exe test=2,2 test=3,2.
AlienFan-CLI v7.0.2.1
Supported hardware detected, 2 fans, 9 sensors, 5 power states, G-Mode.
command 2 arg 2 - result 2933
command 3 arg 2 - result 1200

T-Troll, 2022-08-29:
Check this version: [alienfan-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9448287/alienfan-gui.zip)
Did it work now?

mario6714, 2022-08-29:
I think yes, thanks, what was the problem?

[image]

T-Troll, 2022-08-29:
Not a problem, but a bug into your BIOS. I'll override it.

Wait for next release to update FX GUI as well.

mario6714, 2022-08-29:
Ohh, help if i reinstall the latest version of bios? or i need install a old version?

Ok, for the moment i am going to use only alienfan-gui, thanks.

T-Troll, 2022-08-29:
> Ohh, help if i reinstall the latest version of bios? or i need install a old version?

No. It's just work this way for your system.

closed 2022-08-31 by T-Troll