T-Troll/alienfx-tools #194: Add AW3423DW Support

translucentfocus, 2022-09-01:
This request provides the initial PCAP (change file extension back to .pcap to open in WireShark) for adding Alienware AW3423DW QD-OLED Monitor support.

AW3423DW QD-OLED Monitor Device Information:

> HID\VID_187C&PID_100B&REV_0200
> HID\VID_187C&PID_100B
> HID\VID_187C&UP:FF00_U:0001
> HID_DEVICE_UP:FF00_U:0001
> HID_DEVICE_UPR:FF00-FFFF
> HID_DEVICE

**Methodology for PCAP Collection:**
1. Found a USB root hub that was separated from everything else on the machine to reduce unnecessary PCAP data.  Did this by opening USBPCAP and moving one USB device to different USB ports until that device showed up on a separate hub from everything else.
2. Stopped the AWCC Service from the **Services App** [can also run with **Windows Key** + **R** >> **services.msc**]
3. Unplugged Monitor USB cable from machine
4. Killed AWCC.exe in Task Manager now that the AWCC Service was stopped
5. Started PCAP
6. Plugged in Monitor so that PCAP has USB discovery info
7. Started AWCC Service
8. Opened AWCC
9. Set all 4 lights to a preset Green {RGB: 20,220,18}
10. Set all 4 lights to a preset Red {RGB: 240,0,0}
11. Ended PCAP (may have set lights back to Green)
[AW3423DW.txt](https://github.com/T-Troll/alienfx-tools/files/9465845/AW3423DW.txt)

If this is satisfactory, I will continue [using the very helpful wiki](https://github.com/T-Troll/alienfx-tools/wiki/How-to-collect-data-for-the-new-light-device) to collect the rest of the data

T-Troll, 2022-09-01:
Don't need to play with AWCC/service right now. It needs to be stopped, then we start testing the device only.
VID is Alienware.

Let me check interface, this issue can be a duplicate of #56, please read it.

T-Troll, 2022-09-01:
Yes, it's the same interface as #56.

Can you please run `alienfx-cli probe=a` (answer "n" for light set) and share output to me?

translucentfocus, 2022-09-01:
> Yes, it's the same interface as #56.
> 
> Can you please run `alienfx-cli probe=a` (answer "n" for light set) and share output to me?

```
Dell API not found, Low-level device 0 ready.
===== Device VID_187c, PID_100b =====
Version 512, block size 12
Report Lengths: Output 65, Input 65, Feature 0
Usage ID 0x1, Usage page 0xff00, Output caps 1, Index 64
+++++ Detected as: Alienware,  +++++
Do you want to set devices and lights names? n
Done.
```

T-Troll, 2022-09-01:
Here is the test cli - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466109/alienfx-cli.zip)
Can you, please:
- Stop AWCCService.
- Run `alienfx-cli probe=a` (as before) - check "Detected as:" - should be "Alienware, monitor, APIv6".
- If detection is correct, please issue command `alienfx-cli set-one=0,15,<R>,<G>,<B>`
- This should change all lights color.

translucentfocus, 2022-09-01:
> 
It detects API v6!
```
PS ~\Downloads\alienfx-cli> .\alienfx-cli.exe probe=a
alienfx-cli v7.1.0
Dell API not found, Low-level device 0 ready.
===== Device VID_187c, PID_100b =====
Version 512, block size 12
Report Lengths: Output 65, Input 65, Feature 0
Usage ID 0x1, Usage page 0xff00, Output caps 1, Index 64
+++++ Detected as: Alienware, Monitor, APIv6 +++++
```
Tried changing light colors, appears to not be working properly.
```
PS ~\Downloads\alienfx-cli> .\alienfx-cli.exe set-one=0,15,255,0,0
alienfx-cli v7.1.0
Dell API not found, Low-level device 0 ready.
Done.
```

T-Troll, 2022-09-01:
Ok, detection work correctly.
Can you please use this test version - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466204/alienfx-cli.zip)
Only second command (`set-one`). Did it change the lights?

translucentfocus, 2022-09-01:
> Ok, detection work correctly. Can you please use this test version - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466204/alienfx-cli.zip) Only second command (`set-one`). Did it change the lights?

```
PS ~\Downloads\alienfx-cli (1)> .\alienfx-cli.exe set-one=0,15,255,0,0
alienfx-cli v7.1.0
Dell API not found, Low-level device 0 ready.
Done.
[set-oneCommand.txt](https://github.com/T-Troll/alienfx-tools/files/9466227/set-oneCommand.txt)

```

No luck. No lights changed. PCAP attached (change extension back to .pcap)

T-Troll, 2022-09-01:
Ok, another version (same process) - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466309/alienfx-cli.zip)

translucentfocus, 2022-09-01:
> Ok, another version (same process) - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466309/alienfx-cli.zip)

It worked! All LEDs were set to red. 🥳🎉

PCAP attached as .txt as usual

```
PS ~\Downloads\alienfx-tools-comment-1233653691> .\alienfx-cli.exe set-one=0,15,255,0,0
alienfx-cli v7.1.0
Dell API not found, Low-level device 0 ready.
Done.
```

[3rdalienfx-cliTest.txt](https://github.com/T-Troll/alienfx-tools/files/9466331/3rdalienfx-cliTest.txt)

T-Troll, 2022-09-01:
Great!

translucentfocus, 2022-09-01:
Here's the dark, light, dim pcap as mentioned in the wiki
[DarkLightDIM.txt](https://github.com/T-Troll/alienfx-tools/files/9466372/DarkLightDIM.txt)

T-Troll, 2022-09-01:
So.... New test version - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466631/alienfx-cli.zip)
Can you please check different light/color set:
- command is `alienfx-cli set-one=0,<ID>,<R>,<G>,<B>`. ID = 1,2,3,4 - light number.

translucentfocus, 2022-09-01:
> So.... New test version - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466631/alienfx-cli.zip) Can you please check different light/color set:
> 
> * command is `alienfx-cli set-one=0,<ID>,<R>,<G>,<B>`. ID = 1,2,3,4 - light number.

Lights work! And the colors change as expected.

Possible Issues:

- ID is actually 0,1,2,3 for the lights (order/labels match AWCC) i.e. light 0 is the Alien Head which is labeled as light # 1 in AWCC
- Setting a light causes it to flash white, then set the correct color (even when setting the same color that already was set)
- The first time one runs the ``set-one`` command, it sets _all other lights_ to white and they remain that way until set to something else

T-Troll, 2022-09-01:
No. It's from 1 in my app for now, because it's a bitmask, in fact. Will fix it later.
But the point about white make sense.

T-Troll, 2022-09-01:
Try this cli - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9466870/alienfx-cli.zip)

Warning: it can stop working or have other side effects, let me know.

I try to fix white flash here. Also, on/dim/off should work too, but here is an issue - it will always drop light colors to black for now (and dimming not stored into cli), so use dimming and set light colors after (into one command line!) for check.

translucentfocus, 2022-09-02:
Command:
```
PS ~\alienfx-cli-comment-1233730236> .\alienfx-cli.exe status
alienfx-cli v7.1.0
Dell API not found, Low-level device 0 ready.
Device #0 - Unknown, #4107, Display, VID#6268, PID#4107, APIv6, 0 lights
Done.
```
**Behavior:**
All lights turned off

T-Troll, 2022-09-03:
This version was for test only, it's not intended to work correctly. I just try other way to init device.
And your device does not support hardware power button, so it's the same as set-one (hardware PB is for notebooks - with different color AC/Battery and charging/critlcal/sleep effects).

So... I need one more pcap. Effects, yes. Please run AWCC, set some lights to different effects (please remember colors for it), then share it for me.

PS: Try to play with release gui/cli - it should work with white flash, but stable.

translucentfocus, 2022-09-05:
> This version was for test only, it's not intended to work correctly. I just try other way to init device. And your device does not support hardware power button, so it's the same as set-one [...]

PCAP process:  Lights 1 through 4 (set in order, individually, via AlienFX)
      1. Initial color: Blue (0,0,255)
      2. **Morph** between red (255,0,0) and blue (0,0,255)
      3. **Pulse** blue (0,0,255)
      4. **Breathing** blue (0,0,255)
      5. Repeated for each light

Change attachment to .pcap extension [AlienFXCycles.txt](https://github.com/T-Troll/alienfx-tools/files/9492187/AlienFXCycles.txt)

translucentfocus, 2022-09-05:
> PS: Try to play with release gui/cli - it should work with white flash, but stable.

Gave AlienFX Control 7.1.0 a test.

[image]

I selected **Not Listed** which then causes the program to stop responding.

[image]

Pcap of behavior attached.  Change the extension to .pcap.  
[AlienFX Control.txt](https://github.com/T-Troll/alienfx-tools/files/9492214/AlienFX.Control.txt)

T-Troll, 2022-09-06:
Thank you for effect pcap, i don't need any other for now.
But seems like it's a bug (again!) at clean configuration. I'll check.

PS: Do you really have x15/17, or it's a detection error?

T-Troll, 2022-09-06:
PS: Can you please share your configuration for check - i can't reproduce it! Just open regedit and export `HKCU/Software/AlienFX_SDK` and `HKCU/Software/Alienfxgui`.

translucentfocus, 2022-09-06:
> Thank you for effect pcap, i don't need any other for now. But seems like it's a bug (again!) at clean configuration. I'll check.
> 
> PS: Do you really have x15/17, or it's a detection error?

No, a x15/17 is **_not_** connected.  Only Alienware component being tested/is connected is the AW3423DW.

> Can you please share your configuration for check - i can't reproduce it! Just open regedit and export `HKCU/Software/AlienFX_SDK` and `HKCU/Software/Alienfxgui`.

Configs attached.  As usual, will need to rename the extensions to .reg if you don't want to work with them as text.

[AlienFX_SDK.txt](https://github.com/T-Troll/alienfx-tools/files/9492572/AlienFX_SDK.txt)
[Alienfxgui.txt](https://github.com/T-Troll/alienfx-tools/files/9492573/Alienfxgui.txt)

T-Troll, 2022-09-06:
> No, a x15/17 is not connected. Only Alienware component being tested/is connected is the AW3423DW. It would appear it is a detection error.

It can be not error, but funny side effect. Check your keyboard VID/PID - is it 0D62/CABC?

translucentfocus, 2022-09-06:
> PS C:\Program Files\AlienFX Tools> .\alienfx-cli.exe probe
> alienfx-cli v7.2.0

Seems to hang completely now.  I cleared the REG Keys for Alienfxgui and Alienfx_SDK and they are still blank.

PCAP attached.  Change .txt to .pcap
[Alienfx-cli72.txt](https://github.com/T-Troll/alienfx-tools/files/9492689/Alienfx-cli72.txt)

T-Troll, 2022-09-06:
I don't need pcap anymore, issue is not there!

T-Troll, 2022-09-06:
Try this cli, it's debug - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9492863/alienfx-cli.zip)
Please share the log.

T-Troll, 2022-09-06:
Please also check this gui - if this is not device init it should help.
[alienfx-tools.zip](https://github.com/T-Troll/alienfx-tools/files/9493266/alienfx-tools.zip)
Also detection bug fixed.

T-Troll, 2022-09-06:
About effects:
Seems like this - there are 4 types of command:
```
const byte colorSet[8]{0x92,0x37,0x0a,0x00,0x51,0x87,0xd0,0x04};
		//[3] - command length
		//[6] - command (87 - color, 88 - Pulse, 8c - morph/breath, 84 - timing?),
		//[8] - command type - 4 - color, 1 - morph, 2 - pulse, 3 - timing?
		//[9] - light mask,
		// 3,84 - [10] - Brightness, [11] - ???, [12] - checksum
		// 4,87 - [10,11,12] - RGB, [13] - Brightness (0..64), [14] - checksum
		// 2,88 - [10,11,12] - RGB, [13] - Brightness (0..64), [14] - Tempo?, [15] - checksum
		// 1,8c - [10,11,12] - RGB, [13,14,15] - RGB2, [16] - brightness, [17,18] - tempo, [19] - checksum
```
The only issue now, i can't calculate checksum correctly (i configure out there 8 come from before, but it doesn't work for other commands).

Can you please make another pcap from awcc with different colors and timings (length)  -no matter what lights and kind?

T-Troll, 2022-09-06:
And another task.
Can you please:
- run USBPcap
- THEN run AWCC
- Stop pCap

I want to check how it init the device.

translucentfocus, 2022-09-06:
> Can you please make another pcap from awcc with different colors and timings (length) -no matter what lights and kind?

PCAP attached. (Change extension to .pcap)
[AWCCColorsTimeShifts.txt](https://github.com/T-Troll/alienfx-tools/files/9501284/AWCCColorsTimeShifts.txt)

translucentfocus, 2022-09-06:
> And another task. Can you please:
> 
> * run USBPcap
> * THEN run AWCC
> * Stop pCap
> 
> I want to check how it init the device.

Based on some experiments, it seems the background services do the actually init, so if it helps, here's the PCAP:

- Started PCAP
- Powered off monitor
- Powered on monitor
- Launched AWCC

I did this because, it at least appears like the background services handled the LEDs before AWCC is even opened.

[InitPCAP.txt](https://github.com/T-Troll/alienfx-tools/files/9501304/InitPCAP.txt)

T-Troll, 2022-09-06:
Thank you.

In fact, i'm implement effects already (but a bit weird way), so we only need to test it.
Did `AlienFX-GUI` and CLI work for you now? (check beta versions above)

translucentfocus, 2022-09-06:
> Try this cli, it's debug - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9492863/alienfx-cli.zip) Please share the log.

Where is the log?  I ran it, but can't find a log anywhere.  Is there an option I need to run it with?  Thanks!

Btw, when I run this debug alienfx-cli, it causes the monitor to go black but on.  Windows stops recognizing the monitor. 
 Powering-cycling the monitor fixes it.

```
PS ~\alienfx-cli> .\alienfx-cli.exe probe
alienfx-cli v7.2.0
Scan: VID: 0x187c, PID: 0x100b, Version: 512, Length: 12
Dell API ready, Low-level device 0 ready.
===== Device VID_187c, PID_100b =====
Version 512, block size 12
Report Lengths: Output 65, Input 65, Feature 0
Usage ID 0x1, Usage page 0xff00, Output caps 1, Index 64
+++++ Detected as: Alienware, Monitor, APIv6 +++++
Do you want to set devices and lights names?yes

For each light please enter LightFX SDK light ID or light name if ID is not available
Tested light become green, and turned off after testing.
Just press Enter if no visible light at this ID to skip it.
Probing low-level access... Found 1 device(s)
Dell API version 5.2.0.0 detected
Device #0 - AW3423DW (Display), 4 lights
  Light #0 - Back, Position (0,0,0), Color (0,0,0,0)
  Light #1 - Stand, Position (0,0,0), Color (0,0,0,0)
  Light #2 - Downlights, Position (0,0,0), Color (0,0,0,0)
  Light #3 - Power Button, Position (0,0,0), Color (0,0,0,0)
Probing device VID_187c, PID_100b...Old device name is , Enter device name or LightFX id: 0
Final device name is AW3423DW
```

translucentfocus, 2022-09-06:
> Please also check this gui - if this is not device init it should help. [alienfx-tools.zip](https://github.com/T-Troll/alienfx-tools/files/9493266/alienfx-tools.zip) Also detection bug fixed.

This version behaves just like the last, except it detects no devices at all.  It then freezes and becomes unresponsive.

T-Troll, 2022-09-06:
Well... Thanks.

I mean console log for CLI, not logfile, so you share it already. So now detection not hangs....
.... but what's wrong with gui? Let me check deeper.

> except it detects no devices at all

It's not devices, in fact, but known light schemes.

Device detected, but something going wrong with UI init....

T-Troll, 2022-09-06:
Try this version of GUI - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9501573/alienfx-gui.zip)
It's debug too, and i also change potential issue with init.

translucentfocus, 2022-09-07:
> Try this version of GUI - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9501573/alienfx-gui.zip) It's debug too, and i also change potential issue with init.

Doesn't seem to detect anything & still freezes on launch.  However, I did discover the program will unfreeze if I power cycle the monitor.  Unfortunately, power cycling doesn't fix anything.  PCAP with power cycle attached. 
[Alienfx-gui-1238759317.txt](https://github.com/T-Troll/alienfx-tools/files/9501760/Alienfx-gui-1238759317.txt)

T-Troll, 2022-09-07:
Interesting..... Do you have AWCCService running?
Try to stop it.

T-Troll, 2022-09-07:
Ohh... I found a bug!
Try this one, should work now - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9501823/alienfx-gui.zip)

translucentfocus, 2022-09-07:
> Ohh... I found a bug! Try this one, should work now - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9501823/alienfx-gui.zip)

Awesome! You are doing great!  Getting closer!  Yes! It does work (sort of)!  I can control each light from the Devices and Grid tab.
[image]
However, there's some issues I've noticed such as:

1. On first run, it works, but if I quit AlienFX completely, then run it again, the monitor screen goes black again (leds continue to work) and I have to power cycle.
2. The light #s are not correct as seen below.
[image]
3. I can't seem to assign lights to any zone etc. but maybe I'm doing something wrong.

[image]

T-Troll, 2022-09-07:
Great.
1. Ugh.... Seems like this monitor can be fully controlled this way, including sleep and off! That's why i ask you for AWCC init sequence, let's play with it to find correct command.
2. It's as it should be. BTW, you can type any name there.
3. Read [this](https://github.com/T-Troll/alienfx-tools/wiki/How-to-start-(Beginner's-guide)) first!
4. Define all lights and set it to grid, then press "save lights" and share .csv with me - i'll add it into detection database for other users.

translucentfocus, 2022-09-07:
> 4. Define all lights and set it to grid, then press "save lights" and share .csv with me - i'll add it into detection database for other users.

[AW3423DW.csv](https://github.com/T-Troll/alienfx-tools/files/9503528/AW3423DW.csv)
**Back:**
[image]
**Front:**
[image]

See if that makes sense to you.

translucentfocus, 2022-09-07:
> 1. Ugh.... Seems like this monitor can be fully controlled this way, including sleep and off! That's why i ask you for AWCC init sequence, let's play with it to find correct command.

After some testing, Colors work correctly.  The pulse effect also works, A value of 1 causes it to pulse ~370/sec. A value of 255 causes it to pulse at ~1 pulse per 12 secs.

T-Troll, 2022-09-07:
Thanks for CSV!

You can use smaller grid size (this one for keyboards), but this ok too.
I'll add it to database, so it will be detected at next versions.

So, only pulse works... 
In fact, morph and breathing are the same, breathing just a morph to black under the hood.

We are close to the finish now, all we need now is fix morph (I have some hints what can be wrong with it) and find correct start sequence (current one works, but turns monitor off). I see 2 other into AWCC CSV, so let's try them as well. I'll share the test version tomorrow for check.

T-Troll, 2022-09-07:
Meanwhile, can you please do some testing for Morph?

First, morph require 2 colors (actions)
Second, can you set it any color, but Speed exactly at 2, and Length exactly at 20 (it's what AWCC set)? Maybe i wrong with one of these parameters (first one, i guess).
Third, can you please capture and share pcap for morph set into app? Just for sure it's done the right way?

T.I.A!

T-Troll, 2022-09-08:
Try this CLI - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9523618/alienfx-cli.zip)

It uses another init way (from AWCC), as well as i fix Morph (i hope i found an issue).

translucentfocus, 2022-09-09:
> Third, can you please capture and share pcap for morph set into app? Just for sure it's done the right way?
> 
> T.I.A!

PCAP attached, just change extension to .pcap
[AWCCMorph.txt](https://github.com/T-Troll/alienfx-tools/files/9535116/AWCCMorph.txt)

**Also, here's what I did in AWCC:**
[image]

T-Troll, 2022-09-09:
No, not AWCC, my app!
Anyway, does it work in general?

Thanks anyway!

translucentfocus, 2022-09-09:
> No, not AWCC, my app! Anyway, does it work in general?
> 
> Thanks anyway!

I'm having great difficulties getting ```alienfx-gui.exe``` to start reliably.  Sometimes it just doesn't open.  If I can get it to open and not crash, I'll get a new pcap.

T-Troll, 2022-09-09:
> However, the Downlights (Light # 2 below) does not activate using any configuration.

Uff... Even in GUI?
Ok.... Can you, please:
- Stop AWCC service
- Set it to Manual or Disabled
- Reboot
- Check again

> Also, I'm unsure of the syntax for set-global=<dev>,<type>,<mode>,r,g,b,r2,g2,b2 (what is type vs. mode?) but the following did not change the lights at all.

You don't have this feature.

> Sometimes it just doesn't open. If I can get it to open and not crash, I'll get a new pcap.

Aaargh! What's going on? Of course, it should not be! Drop me a reg files to check how it happened, please!

translucentfocus, 2022-09-09:
> 

> > However, the Downlights (Light # 2 below) does not activate using any configuration.
> 
> Uff... Even in GUI? Ok.... Can you, please:
> 
✅ Stop AWCC service
✅ Set it to Manual or Disabled
✅ Reboot
✅ Check again
> 
> > Also, I'm unsure of the syntax for set-global=,,,r,g,b,r2,g2,b2 (what is type vs. mode?) but the following did not change the lights at all.
> 
> You don't have this feature.
> 
> > Sometimes it just doesn't open. If I can get it to open and not crash, I'll get a new pcap.
> 
> Aaargh! What's going on? Of course, it should not be! Drop me a reg files to check how it happened, please!

**Here's the registry files:**
[Alienfxgui.txt](https://github.com/T-Troll/alienfx-tools/files/9535309/Alienfxgui.txt) [change file extension back to .reg]
[AlienfxSDK.txt](https://github.com/T-Troll/alienfx-tools/files/9535310/AlienfxSDK.txt) [change file extension back to .reg]

Here's the pcap of alienfx-gui (can't seem to get it to launch now).  However, the new alienfx-cli from https://github.com/T-Troll/alienfx-tools/issues/194#issuecomment-1240212981 seems to work for the most part (except the downlights and I don't know the syntax for morph)
[alienfx-gui.txt](https://github.com/T-Troll/alienfx-tools/files/9535312/alienfx-gui.txt) [change file extension back to .pcap]

translucentfocus, 2022-09-09:
> > However, the Downlights (Light # 2 below) does not activate using any configuration.
> 
> Uff... Even in GUI? Ok.... Can you, please:

Just checked, the gui sets the downlights properly... but I still can't seem to get the darn gui to open... it just sets the lights invisibly 😆

T-Troll, 2022-09-09:
> does not set the downlights.

So, only `set-one=0,2,255,0,0` doesn't work?

> except the downlights and I don't know the syntax to test morph

`set-action=0,<id>,morph,r,g,b,255,morph,r2,g2,b2`

>  sets the downlights properly

I checking you config now, so it starts, init, set lights and crash?

T-Troll, 2022-09-09:
I check, config looks correct. Hmm... let me think what it can be.
BTW, you have light dimming on into it.

Try to remove `alienfxgui` (but keep `sdk` - you lost zone settings, but not lights and zones) - did it fix the situation?

Do you remember that you do then it crash first time?

T-Troll, 2022-09-10:
I found and fix this bug, check the next release!
For now, just open registry, navigate to `HKCU/Software/Alienfxgui/Profiles` and change `Profile-gflags-0` from 00040001 to 00000001. Bug was into Grid effects init, so do not enable it at start.

T-Troll, 2022-09-10:
Check [7.2.2](https://github.com/T-Troll/alienfx-tools/releases/tag/7.2.2)

translucentfocus, 2022-09-17:
> Check [7.2.2](https://github.com/T-Troll/alienfx-tools/releases/tag/7.2.2)

Much better!
✅ Opens and detects monitor properly
[image]

✅ My crappy default config I gave you loads correctly 😁
✅ Breath works properly
✅ Morph works correctly

❔ Morph allows 3 colors, but only morphs between 2 colors
[image]

❌ **Spectrum** doesn't appear to work at all
❌ **Rainbow** doesn't appear to work at all
❌ Dragging the sliders causes multiple lights to flash, change color etc. 'glitch out' is a good description
❌ Dragging sliders also causes the monitor to revert to [monitor firmware light settings (see the FX lighting controls in the menu on the left?)](https://youtu.be/XVgUosc-64U?t=381) (lose connection?) might be a good way to describe it.  To fix it, I must click **Save** in AlienFX to 'reactivate' my settings.
[image]
USB Packet Capture:  [SliderGlitch.txt](https://github.com/T-Troll/alienfx-tools/files/9590581/SliderGlitch.txt)

T-Troll, 2022-09-17:
"Colors" page are universal across all hardware.
So it works correctly - some hardware have more effect types, some can allow up to 9 effects in chain.
For your hardware, the rest of the effects works as a simple color set.

Sliders issue is interesting... In fact, "Length" is not used for your hardware at all. It's a bug in case it changes anything!
And yes, it should flash a bit - all changes apply immediately, which is reset effect to beginning.

translucentfocus, 2022-09-19:
> "Colors" page are universal across all hardware. So it works correctly - some hardware have more effect types, some can allow up to 9 effects in chain. For your hardware, the rest of the effects works as a simple color set.

Ah, okay, I thought maybe you had a way of setting the AW3423DW profile to not allow more than two morph colors?

For Spectrum & Rainbow effects, should those work?  In AWCC, I do have a spectrum effect that works.

If those shouldn't work, then I think you can close this issue & declare initial AW3423DW support complete!

Thank you for all your hard work on this! 🥳

T-Troll, 2022-09-19:
> Ah, okay, I thought maybe you had a way of setting the AlienFX AW3423DW profile to not allow more than two morph colors?

I prefer to add multi-color effects then configure out how to do it ^_^

> In AWCC, there is a spectrum effect that works.

You didn't provide me .pcap for this! Do it and i'll add it as well.

> If those shouldn't work, then I think you can close this issue & declare initial AW3423DW support complete!

Except menu popup situation. Still need investigation. Uff.. Tough device you have!
But yes, most functionality is working.

[1 more reply]

closed 2022-09-29 by T-Troll