T-Troll/alienfx-tools #195: Craches on startup

brunosalezze, 2022-09-01:
**Describe the bug**
The app craches on startup.

**System (please complete the following information):**
 - Model: Dell G15 5525 Ryzen
 - Model details:  per-key rgb keyboard
 - OS: Windows 11 stable

**Additional context**
I was able to start the app and see the GUI once, but crached and did not load again. How do I delete the config file?

brunosalezze, 2022-09-05:
Sorry Its a G15 5515 AMD Edition.
The keyboard actually is a 4 zone one. 
The only app I am having issues with is Control, the other ones work great!
I was able to open the app again deleting the registry keys.
How can I help for my device to be suported in future releases?

T-Troll, 2022-09-06:
Oh... I'we as you to share config to me first - i need to pinpoint and fix this! %-(

Make all lights and grids first, then press "Save lights and grids" button and share .csv file with me. I'll add it into device database next release.

closed 2022-09-10 by T-Troll