T-Troll/alienfx-tools #205: Light dimming issue on APIv2

appleff4, 2022-09-19:
**Describe the bug**
light dimming power is active even if I uncheck the dim lights checkbox although I can control the dimming power using the dimming power slider.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'settings'
2. Check the 'dim lights' no effect
3. Slide the 'dimming power slider' lights dim and brighten accordingly

**System:**
 - Model: [Alienware 15 R2]
 - AlienFX: [APIv2, VID_187C/PID_0528]
 - OS: [Windows 10 1909 stable]

T-Troll, 2022-09-20:
Check profile settings - in case dimming active at profile as well, this switch can not work.

appleff4, 2022-09-21:
I Checked it out, same situation, only the dimming power slider can control the brightness of the lights..

T-Troll, 2022-09-21:
Please open `regedit`, export `HKCU/Software/Alienfxgui` and share it with me.

appleff4, 2022-09-22:
Here you go
[ALIENFXGUI.zip](https://github.com/T-Troll/alienfx-tools/files/9630061/ALIENFXGUI.zip)

[HKEY_CURRENT_USER\Software\Alienfxgui]
"AutoStart"=dword:00000001
"Minimized"=dword:00000001
"LightsOn"=dword:00000001
"Dimmed"=dword:00000000
"Monitoring"=dword:00000001
"OffWithScreen"=dword:00000001
"NoDesktopSwitch"=dword:00000000
"DimPower"=dword:00000000
"DimmedOnBattery"=dword:00000000
"OffPowerButton"=dword:00000000
"DimmingPower"=dword:00000080
"ActiveProfile"=dword:00000000
"GammaCorrection"=dword:00000001
"ProfileAutoSwitch"=dword:00000000
"DisableAWCC"=dword:00000001
"EsifTemp"=dword:00000000
"FanControl"=dword:00000000
"CustomColors"=hex:80,dc,40,00,35,b0,d9,00,ce,79,40,00,ff,00,00,00,00,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00
"UpdateCheck"=dword:00000001
"OffOnBattery"=dword:00000000
"Ambient-Shift"=dword:00000000
"Ambient-Mode"=dword:00000000
"Ambient-Grid"=dword:000a000a
"Haptics-Input"=dword:00000000
"ShowGridNames"=dword:00000001

[HKEY_CURRENT_USER\Software\Alienfxgui\Events]

[HKEY_CURRENT_USER\Software\Alienfxgui\Profiles]
"Profile-1"="default"
"Profile-gflags-1"=dword:00000001
"Profile-triggers-1"=dword:00000000
"Profile-effect-1"=hex:00,00,05,00,00,00,00,00,00,00,00,00
"Profile-0"="Profile 0"
"Profile-gflags-0"=dword:00000002
"Profile-triggers-0"=dword:00000000
"Profile-effect-0"=hex:00,00,05,00,00,00,00,00,00,00,00,00

[HKEY_CURRENT_USER\Software\Alienfxgui\Zones]
"Zone-flags-1-65536"=dword:00000001
"Zone-colors-1-65536-4"=hex:02,14,14,ff,ff,00,02,00,00,80,ff,00,02,00,00,00,ff,\
  ff,02,00,00,ff,00,00
"Zone-effect-1-65536"=hex:01,00,50,13,01,00,00,00,00,00,00,00,00,00,00,00
"Zone-flags-1-65537"=dword:00000000
"Zone-colors-1-65537-2"=hex:06,00,00,ff,00,00,06,00,00,00,ff,40
"Zone-events-1-65537"=hex:01,00,00,00,00,00,00,ff,00,00,00,00,00,00,ff,40,00,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,\
  00,00,00,00,00
"Zone-flags-1-65538"=dword:00000000
"Zone-ambient-1-65538-28"=hex:1b,11,01,00,02,03,04,06,05,0d,0c,0b,0a,08,09,07,\
  0e,0f,10,12,13,14,1a,19,18,17,16,15
"Zone-flags-1-65539"=dword:00000000
"Zone-colors-1-65539-1"=hex:00,00,00,80,00,40
"Zone-flags-1-65540"=dword:00000000
"Zone-colors-1-65540-1"=hex:00,00,00,77,dd,33
"Zone-effect-1-65540"=hex:01,01,50,03,01,00,00,00,40,80,ff,00,00,ff,00,00
"Zone-flags-1-65541"=dword:00000000
"Zone-colors-1-65541-1"=hex:00,00,00,39,67,d0
"Zone-flags-1-65542"=dword:00000000
"Zone-colors-1-65542-1"=hex:00,00,00,35,b0,d9
"Zone-flags-1-65543"=dword:00000000
"Zone-colors-1-65543-4"=hex:02,00,00,ff,80,40,02,00,00,00,ff,00,02,00,00,ff,80,\
  40,02,00,00,00,ff,00
"Zone-flags-1-65544"=dword:00000000
"Zone-colors-1-65544-4"=hex:02,00,00,00,ff,00,02,00,00,ff,80,40,02,00,00,00,ff,\
  00,02,00,00,ff,80,40
"Zone-flags-1-65545"=dword:00000000
"Zone-colors-1-65545-1"=hex:00,00,00,ce,79,40
"Zone-flags-0-65537"=dword:00000000
"Zone-colors-0-65537-2"=hex:06,03,64,ff,80,00,06,03,64,00,80,00
"Zone-flags-0-65538"=dword:00000000
"Zone-ambient-0-65538-28"=hex:1b,11,01,00,02,03,04,06,05,0d,0c,0b,0a,08,09,07,\
  0e,0f,10,12,13,14,1a,19,18,17,16,15
"Zone-flags-0-65539"=dword:00000000
"Zone-colors-0-65539-1"=hex:00,00,00,80,00,40
"Zone-flags-0-65540"=dword:00000000
"Zone-colors-0-65540-1"=hex:00,00,00,77,dd,33
"Zone-effect-0-65540"=hex:01,01,50,03,01,00,00,00,40,80,ff,00,00,ff,00,00
"Zone-flags-0-65541"=dword:00000000
"Zone-colors-0-65541-1"=hex:00,00,00,77,dd,33
"Zone-flags-0-65542"=dword:00000000
"Zone-colors-0-65542-1"=hex:00,00,00,35,b0,d9
"Zone-flags-0-65543"=dword:00000000
"Zone-colors-0-65543-4"=hex:02,00,00,ff,80,40,02,00,00,00,ff,00,02,00,00,ff,80,\
  40,02,00,00,00,ff,00
"Zone-flags-0-65544"=dword:00000000
"Zone-colors-0-65544-4"=hex:02,00,00,00,ff,00,02,00,00,ff,80,40,02,00,00,00,ff,\
  00,02,00,00,ff,80,40
"Zone-flags-0-65545"=dword:00000000
"Zone-colors-0-65545-1"=hex:02,00,00,ff,80,00
"Zone-flags-0-65546"=dword:00000000
"Zone-colors-0-65546-7"=hex:02,00,00,ff,00,00,02,00,00,ff,80,00,02,00,00,ff,ff,\
  00,02,00,00,00,ff,00,02,00,00,00,ff,ff,02,00,00,00,00,ff,02,00,00,ff,00,ff
"Zone-flags-0-65547"=dword:00000000
"Zone-colors-0-65547-7"=hex:02,00,00,ff,80,00,02,00,00,ff,ff,00,02,00,00,00,ff,\
  00,02,00,00,00,ff,ff,02,00,00,00,00,ff,02,00,00,ff,00,ff,02,00,00,ff,00,00
"Zone-flags-0-65548"=dword:00000000
"Zone-colors-0-65548-7"=hex:02,00,00,ff,ff,00,02,00,00,00,ff,00,02,00,00,00,ff,\
  ff,02,00,00,00,00,ff,02,00,00,ff,00,ff,02,00,00,ff,00,00,02,00,00,ff,80,00
"Zone-flags-0-65549"=dword:00000000
"Zone-colors-0-65549-7"=hex:02,00,00,00,ff,00,02,00,00,00,ff,ff,02,00,00,00,00,\
  ff,02,00,00,ff,00,ff,02,00,00,ff,00,00,02,00,00,ff,80,00,02,00,00,ff,ff,00

T-Troll, 2022-09-23:
Seems like i found a bug.
Dimming for some devices (v1-v3,v7) is software, so i forget to check some condition.
Check this version, it should fix this - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9630311/alienfx-gui.zip)

appleff4, 2022-09-23:
I checked it, and it's functioning correctly now.
Best regards.

closed 2022-09-23 by appleff4