T-Troll/alienfx-tools #208: Fan speed lowers at high temperature

sanga000, 2022-09-23:
It appears that, despite maintaining the same boost, the fan speed would gradually lower at higher temperatures until it reaches the default 100% cap (4000rpm) at around 85+ celcius for CPU and 77+ celcius for GPU. This ONLY happens in manual power mode however. Setting the power mode to "level 2" (which corresponds to "full speed" in AWCC) does not have this issue and maintains 5100rpm for CPU and 5300rpm for GPU, meaning it should be possible to maintain the high rpm. This problem happens regardless whether AWCC is disabled or not.

Specs:
Alienware M15R5
OS: Windows 10
CPU: Ryzen 7 5800H
GPU: RTX 3060
RAM: 16GB

[image]

T-Troll, 2022-09-23:
What do you have in "Scale" parameter into curve window?
This situation is normal in case no overboost defined (and maybe it missed at your system because you have fake RPM (always x100). Performance mode run faster, then 100% boost in common, overboost CAN help reach this speed. But not for all BIOSes.

PS: BTW, you can rename power mode (except manual).

sanga000, 2022-09-23:
The scale was around 150 the first time I used the "Overboost" button. I tried using Overboost again at high temperature, where it could not boost at all (scale was at 100). I redid it right after a reboot and gave me a scale of 198 (and gave me an extra 100rpm than the first time). The high temperature results (with 198 scale) were the same though (capped at 4000rpm)

wearver, 2022-09-23:
> What do you have in "Scale" parameter into curve window? This situation is normal in case no overboost defined (and maybe it missed at your system because you have fake RPM (always x100). [...]

I have the same problem. I set overboost, works well if i set the curve high than normal, but, if the system gain high temperature, the fan run alone regard the curve. Even the temperature decrease, the fan mantain the RPM high than i set., and stay for few minutres, than the AlienFX take control again. Is the Bios take control of the fans?

T-Troll, 2022-09-24:
You are right. Fan control at modern systems is indirect - you can't really set RPMs, it's BIOS-controlled.
Boost is some addition above BIOS level.
Overboost is the possibility to add even more RPMs, but not all systems support it.
Also, your BIOS shows fake (approximate) RPMs, so my overboost detection can work incorrectly. This did not break anything, but make fan control less precise.
You can play with manual overboost - just run `alienfan-cli setover=<fan number from 0>,<desired overboost level, 100+>`

sanga000, 2022-09-24:
> You are right. Fan control at modern systems is indirect - you can't really set RPMs, it's BIOS-controlled. Boost is some addition above BIOS level. Overboost is the possibility to add even more RPMs, but not all systems support it. Also, your BIOS shows fake (approximate) RPMs, so my overboost detection can work incorrectly. This did not break anything, but make fan control less precise. You can play with manual overboost - just run `alienfan-cli setover=<fan number from 0>,<desired overboost level, 100+>` from Administrator CMD.

I'm not sure if I understand you here. The Overboost function is working fine at idle temperatures. The indicated fan speeds seem to be correct as well under all temperatures, judging from the fan noise (which decreases as the internal temperatures raise) and the temperature at which the computer heats up to (for the same game in the exact same spot, 80+ degrees for GPU with alienfan-gui compared to 75 degrees when set to "Level 2" ("Full speed" in AWCC)).

So from my understanding here, the bios simply doesn't follow the boost level from alienfan-gui when in high temperatures?

wearver, 2022-09-24:
> You are right. Fan control at modern systems is indirect - you can't really set RPMs, it's BIOS-controlled. Boost is some addition above BIOS level. Overboost is the possibility to add even more [...]

It does not work. Sad to hear that this app doesn't work as expected on Dell G15 with Ryzen CPU. The Bios take over the Fans RPM in high temperature, after fews minutes of cool down, is unlock to app take control again...

T-Troll, 2022-09-25:
Ah... Yes, this happened at high temps, after a minute at my system.
Even in g-mode or "Full speed" mode.
The most correct reason i can suggest is power system have not enough power for fans in case both CPU and GPU loaded, so it nerfs it a bit.

Unfortunately, i can'd do anything with it for now.

sanga000, 2022-09-25:
> Ah... Yes, this happened at high temps, after a minute at my system. Even in g-mode or "Full speed" mode. The most correct reason i can suggest is power system have not enough power for fans in case both CPU and GPU loaded, so it nerfs it a bit.

The fans are able to operate above the default 4000rpm cap in "Full speed" (see 3rd pic of original post) despite the high load, so I'm not sure if this is the cause.

> Unfortunately, i can'd do anything with it for now.

Fair enough. Thanks for replying throughout the case.

T-Troll, 2022-09-26:
Do you know what power modes really do at the hardware level? So it looks this case for me - manual nerf fans at high LOAD (which means high temperature and high power usage). 
Seems like profile for "Full Speed" have different power limit for fans (it usually have this), overboost can compensate it for some systems, but not for all of them.

Sorry.

wearver, 2022-09-26:
> Do you know what power modes really do at the hardware level? So it looks this case for me - manual nerf fans at high LOAD (which means high temperature and high power usage). Seems like profile for [...]

which notebook do you use? 

In may case works this way: In normal use the fan curve works 100%, but, a soon as the temperature reach  near 85º,  fan get above 100% and stay for some time.... then, apparently the bios take over and lower the RPM to 85%, but dont lock the RPM control, after cool down, the rpm curve still works.  The bios take all control if the both, cpu and gpu rise above certain temperature, and stay lock for a while .

T-Troll, 2022-09-29:
1. I have a bug with power mode set after resume from sleep, so check latest release (7.3.0).
2. I have m15R1 (main one), M13R2 (to check old model compatibility - but it has direct fan control), m15R6. R1 have the same issue (but overboost helps a lot), R6 never reach 85C ^_^. Also, try to understand - RPM limit is BIOS, boost is just a hint for it. So yes, it can lock it in some cases (but usually lock higher, not lower).

closed 2022-10-18 by T-Troll