T-Troll/alienfx-tools #214: Startup delay option

DinnerBug, 2022-10-07:
**Is your feature request related to a problem? Please describe.**
I based my fan curve off of the gpu core temperature which is only available after running Libre Hardware Monitor. I have Alien FX control and Libre Hardware Monitor set to "Start with Windows" but Alien FX control always starts faster whan Libre Hardware Monitor and so the sensors arent's detected and can't be used.

**Describe the solution you'd like**
I would like the program to only start after Libre Hardware Monitor so maybe add a startup delay option

**System (please complete the following information):**
 - Model: Alienware Aurora r8
 - Model details: egb on the pc case only
 - OS: Windows 10 home

**Additional context**

Thank you :)

T-Troll, 2022-10-08:
I have this option before (for driver load delay), so i just return it back.
Thank you to point it, wait until next release beta.

DinnerBug, 2022-10-08:
Your are a king. Thank you and have a nice day.

T-Troll, 2022-10-09:
Hold on!
First i need to implement it (and you check), then close this issue after release.

DinnerBug, 2022-10-09:
oh, ok

T-Troll, 2022-10-10:
Check this one - [alienfx-tools.zip](https://github.com/T-Troll/alienfx-tools/files/9742913/alienfx-tools.zip).
How it works - add delay in seconds into command line (like `alienfx-gui.exe 5` for 5 seconds init delay). All GUI apps support this option.

DinnerBug, 2022-10-10:
Hey, it works, but I get a Windows prompt to authorize the program.

DinnerBug, 2022-10-10:
[image]
[image]
[image]
[image]
[image]

DinnerBug, 2022-10-10:
Thanks to your help I've been able to figure it out. I'm leaving instructions here (if that's ok with you) in case someone else who doesn't read "README"s swings by the solved issues section.

- In AlienFX Control enable "Start with Windows"
- Open Task Scheduler
- Task Scheduler Library > right click AlienFX-GUI > Properties
[image]

- Triggers > At log on > Edit
[image]

- Enable "Delay task for" and choose how long after logging in the program will start.
[image]

- Save

T-Troll, 2022-10-11:
Thank you for log, i'll use parts of it in manual.
But there is one issue - "Task delay" unstable into Windows 10.
So more solid way is "Actions"->"Edit"->Add delay into "Arguments" field.

DinnerBug, 2022-10-11:
Ok, I'll change that, thank you again.

closed 2022-10-18 by T-Troll