T-Troll/alienfx-tools #215: Morph effect disabled after save or refresh

MateusBMP, 2022-10-08:
**Describe the bug**
If you configure keyboard light to use two colors with Morph actions, set speed and length correctly and click at "refresh" or "save" buttons, the effect stop. Interestingly, when you click on "refresh", keyboard color stop on first color, while clicking on "save" the color freeze on current keyboard color. To refresh effect, change any color, speed or length.

**To Reproduce**
Steps to reproduce the behavior:
1. On AlienFX Control, go to "colors"
2. Select any keyboard zone
3. Go to actions and add two "morph" actions with any different colors, speed 128 and length 5
4. Test the two bugs:
4.1. Click at "save" button to see freeze color bug or
4.2 Click at "refresh" button to see the color changing to first added color and freezing

**Expected behavior**
After click on "save" or "refresh", just keep the effect.

**Screenshots**
Morph effect configuration:
[image]

**System:**
 - Model: Alienware m15R6
 - Model details: Per-Key RGB keyboard
 - OS: Windows 11 Home Single Language stable
 - AlienFX-Tools resease: [7.4.0 (Monitoring effect chains)](https://github.com/T-Troll/alienfx-tools/releases/tag/7.4.0)

**Additional context**
The effect also freezes on the first color after some minutes, but I still can't say after how long. When I know, I'll document that part better.

T-Troll, 2022-10-09:
You have Effect Mode set to monitoring.
In this case, if ANY light for the same device assigned to it, all hardware effects will be stopped.
It's a hardware bug, i mention it into application manual and readme.

MateusBMP, 2022-10-09:
Oh, I'm sorry... However, I have a doubt. If I change to "Any", events configured  into "events monitoring" for other zones too stop to work correctly?

T-Troll, 2022-10-09:
According to grid and zone, you have APIv4 4-zone keyboard, right? (Dell G name it "Alienware keyboard")
For this device, you can only choose one - hardware effects or software effects (monitoring, ambient, etc.).
The issue is this chip stop all hardware effects at any device light update... even in case this light does not participate into it, as you can see.

T-Troll, 2022-10-09:
And a word about effect freeze (in fact - device freeze). It can happen if you change power button color so frequently.
It's a hardware bug as well, AWCC have hard-coded delay in case you change it, i found how to check then device ready. But delay can be from 0 to 10 seconds after power button color update (1.5 sec in common)!

MateusBMP, 2022-10-09:
I understand. There are many details that I had no idea... However, I would like to thank you for your great job and incredible project. Hope to be able to collaborate with him soon, testing or writing code.

T-Troll, 2022-10-10:
Thanks. I always need testers, i don't have all AW/Dell zoo at my table...
Join Discord support server (see readme), we have a "testing" channel for big releases there.

MateusBMP, 2022-10-10:
Thanks. I'm already in, but I will follow more messages there.

closed 2022-10-18 by T-Troll