T-Troll/alienfx-tools #219: "Lights" tab not opening

echomokona, 2022-10-17:
**Describe the bug**
The alienfx-gui.exe (AlienFX Control) program opens directly into the "Fans and Power" tab, and the "Lights" tab cannot be opened. Clicking the tab makes the GUI flash white briefly, then immediately return to the "Fans and Power" tab. 

Version in use: AlienFX-tools 7.4.2.0 (latest as of this writing's moment).
Windows 11, clean installation (no "upgrade" or anything that carries any data over)

**To Reproduce**
Steps to reproduce the behavior:
1. Open AlienFX Control (alienfx-gui.exe)
2. Click on the 'Lights'  tab
3. 
**Expected behavior**
The 'Lights' tab should open and its content be normally visible and in operating state.

**Screenshots**
It would be a few-seconds-long video showing that accessing the 'Lights' tab flashes the GUI and returns to the 'Fans and Power' tab.
**System (please complete the following information):**
 - Model: Dell G15 5515 Ryzen Edition
 - OS: Windows 11, freshly installed, all Windows Updates done.

**Additional context**

**Before reinstalling Windows, the software worked normally.**
In the old Windows installation, I had the possibility at some part of the GUI to specify which device I was using. I don't remember the details, but I could choose "Dell G15 5515". The options in 'Lights' that actually made this keyboard's lights work were Lights 8 through 11, which seemed unusual, but it worked fully (including effects such as Haptics).
Now, all functionality related to Lights is not working.
I don't know if there is a relation, but, in the old Windows installation, I had Alienware Control Center installed in the past at some point (already uninstalled upon trying AlienFX-tools). In this new installation, not AWCC nor any of its components were ever installed so far.

T-Troll, 2022-10-17:
It's correct. This can happen in case no light devices detected.
The issue, in fact, i need to add some message about it to clarify.

So make sure you have USB HID device with VID_187C&PID_0550 present into the system. Should be "Vendor-defined device" into the list.
As a check, you can open CMD and run `alienfx-cli probe` command. It shows all detected (even if not defined) devices.

echomokona, 2022-10-18:
I've reinstalled Windows 11 again, and now AWCC works normally and control the keyboard lights through it - but AlienFX-Tools does not work yet. Before reinstalling Windows at all, everything in AlienFX-Tools was working normally, including the Lights tab and the keyboard lights.

Now, the "Lights" tab can partially be accessed - but directly (and only) into the sub-tab "Devices and Grids", but is unable to perform more Lights-related functionality than this. Screenshot attached. 
[image]

T-Troll, 2022-10-18:
Why don't you back up Alienfx-tools config? You lost all settings after OS re-install as a result!
Now you also need to set lights on grid.

echomokona, 2022-10-18:
Before OS reinstall, I didn't have many configs done in Alienfx-tools. After the reinstall, it feels more like the software isn't recognizing/interacting with the hardware, rather than a simple matter of configs - many parts of the UI (Sub-tabs from Lights tab) are still inaccessible.

T-Troll, 2022-10-18:
It's because you lost grids. No light on grid - no light control, it's based on it.
Anyway, i add config save/restore/remove script into today's release.
Device is accessible (i see the log), so just configure it down according to beginner's guide.

echomokona, 2022-10-18:
Thank you very, VERY much for the super quick replies, and even quicker fixing. 

With the new release, everything appears to be functioning as expected. Even "tricks" with the light modes, such as mixing Breath and Morph to create a "realistic" breathing effect are possible.
Again, thank you a lot, T-Troll, for the attention and for this wonderful piece of software!

closed 2022-10-18 by echomokona

T-Troll, 2022-10-18:
You are welcome. As you see, i don't accept any donate, i just do it for people.