T-Troll/alienfx-tools #222: Cannot use Ambient mode with AW3423DW

MaluNoPeleke, 2022-10-18:
**Describe the bug**
As of release 7.2.2 and accordingly to #194 the Dell Alienware monitor AW3423DW is supported. I just discovered alienfx-tools and want to use it to synchronize the RGBs of this monitor with the screen content (called Ambient mode I guess).
After launching the application and detecting the monitor I set it to "Ambient" but the RGBs just slowly rotate colors, even with a static screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Start alienfx-gui
2. Click on 'Effects mode dropdown' and set it to "Ambient"
3. Click on "Refresh"
4. See error

**Expected behavior**
The RGB should mirror the current screen content color.

**Screenshots**
[image]

**System (please complete the following information):**
 - Model: AlienWare AW3423DW
 - OS: Windows 11 insider

**Additional context**
I cannot select any other tab in the window such as "Colors", "Ambient" or "Haptics". Only "Devices and Grids" is selected/visible by default.

T-Troll, 2022-10-19:
>  but the RGBs just slowly rotate colors, even with a static screen

This happened because you need to set up Ambient - screen zones not defined yet. So colors not changed, using you default settings.

> I cannot select any other tab in the window such as "Colors", "Ambient" or "Haptics". Only "Devices and Grids" is selected/visible by default.

Aha. This is the bug. Let me check, seems like some data is not set correctly after detection. For now, just try to restart app - dit it work correctly after this?

T-Troll, 2022-10-19:
Yes, it was a bug after "Detect" button pressed and data applied.
Fixed in 7.4.3.1.

MaluNoPeleke, 2022-10-19:
Thanks for the quick fix, now I was indeed able to switch to the other areas but after defining ambient zones and clicking on "Refresh" the LEDs still change colors slowly as before.
Anything else I missed?
[image]

T-Troll, 2022-10-19:
Do you have DirectX installed? (install if not yet)
Did the colors of "screen zones" represent real screen colors correctly? (press "Reset"/choose other screen if you have more, then one if not).
BTW, read tools readme - there are some hints why it can not work (in "requirements" and "issues").

I'll check it at my system, all correct.

MaluNoPeleke, 2022-10-19:
Yes, I have DirectX installed and the colors of the screen zones represent real screen colors correctly.
Here is a video of the issue: https://photos.app.goo.gl/qGJdf8Vc6mNLswUv9

T-Troll, 2022-10-19:
Got it.

Seems like something/someone took control about lights. If it was RGB keyboard, i've said global effect active.
Can you please check just plain color set? Did it work?
If not, i need to check you AWCC settings.

PS: Also, it's a good idea to stop AWCCService/chek "Disable AWCC" at Settings. But, seems like the issue is deeper a bit...

MaluNoPeleke, 2022-10-19:
I have checked that option already but that doesn't help.

MaluNoPeleke, 2022-10-19:
Changing Colors doesn't do anything either: 
[image]
These are my other settings:
[image]
[image]
If I run AWCC it looks like this (didn't set up anything there yet):
[image]
[image]
If I change colors there it works as expected immediately.

T-Troll, 2022-10-20:
Good, so it's not a hardware issue, just configuration. I may break it inside tools, or it's a result of incorrect initialization (we spend a lot of time to do it right way then add support).

I need to check what do you have into "FX" settings of the AWCC (choose monitor device first), fortunately i can read German.
Also, it was a post at Alienware reddit - this can be monitor menu issue, latest firmware have some bugs there.

Anyway, we need to check what happened on hardware level later on, so please install USBPcap - i'll provide you test version and tell what to do.

MaluNoPeleke, 2022-10-20:
This is how the FX area looks like (still untouched):
[image]
[image]

T-Troll, 2022-10-20:
Thank you.

What is interesting here - you set lights to color, but backlights are breathing! Please check back panel lights settings (monitor back picture at up-right).

Anyway, seems like color set (or device reset again!) is broken. Can you, please:
1. Close AWCC
2. Open Services
3. Stop AWCCService
4. Run some CLI commands, like `alienfx-cli set-all=255,0,0` or `alienfx-cli set-all=0,0,255`

Did it change monitor light's color?

MaluNoPeleke, 2022-10-20:
This is the output:

> C:\Program Files\AlienFX Tools>alienfx-cli set-all=0,0,255
> alienfx-cli v7.4.3.1
> Dell API not found, 1 low-level devices found.
> Done.

The color always changes to purple for a second, no matter what kind of RGB combination I use.

T-Troll, 2022-10-20:
Thank you, things starting to be clear.

So what we have? CLI change color (GUI share the same code, but more complex way), but something in system or monitor configuration reset it back immediately. And seems like it's not AWCC.

Unfortunately, i doesn't have this monitor at my table, so i can only think and guess what is the reason for this behavior...

... Interesting case! Let me think a bit. Meanwhile, try to Reset lighting and set "Effects" on, then check what happened. And, most probably, we need to check what happened at USB level - both from AWCC and my apps (USBPcap is for this).

MaluNoPeleke, 2022-10-20:
This is how it looks like after reset: 
[image]
The effects setting provides three options: Spectrum, Rainbow Cycle and Comet.
USBPcap is installed, what to do with it?

T-Troll, 2022-10-22:
Ok, let's try,
First, please get this CLI version - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/9843217/alienfx-cli.zip)
It has older device init command (it provides side effect - all lights will be white).
Stop AWCCService and check `set-all` command like before. Did it work now?

If not, we will grab your version of AWCC interface to check.

MaluNoPeleke, 2022-10-23:
No, still same behavior as before, just with a slightly different output on cmd:

> alienfx-cli v7.4.3.1
> Dell API ready, 1 low-level devices found.
> Done.

Which file of AWCC do you need?

@translucentfocus does it work for you?

T-Troll, 2022-10-24:
Thanks for information.
Uff... We need to do it a hard way.

Can you, please:
1. Install USBPcap (you are already do so AFAIR).
2. Run USBPcap
3. Check what USB hub have monitor connected (the list of devices will be print into it's console) and select it
4. Select any file name (f.e. `data.pcap`)
5. Keep it running
6. Open AWCC
7. Set some monitor lights (better use simple colors - red, green, blue, etc)
8. Stop USBPcap (switch to it's window and press CTRL+C)
9. Share file from p.4 here

I want to check how AWCC init it now, as well as possible interface changes.

translucentfocus, 2022-10-24:
> No, still same behavior as before, just with a slightly different output on cmd:
> 
> > alienfx-cli v7.4.3.1
> > Dell API ready, 1 low-level devices found.
> > Done.
> 
> Which file of AWCC do you need?
> 
> @translucentfocus does it work for you?

I haven't tested on the sample we have... but I see you're using Windows 11... does Windows 11 use DX12 for the desktop compositor?  I know DX12 doesn't allow access to required information.  @T-Troll will probably know, but I just wasn't even sure Ambient mode _would_ work in Windows 11.

T-Troll, 2022-10-24:
@translucentfocus, yes, you are right about DX12, i even mention it into readme. But it should work for desktop, at least.
But here we have another issue - light control simple broken at his device. We are trying to configure out why it happened.

JFYI: Here is the issue of monitors support story - https://github.com/T-Troll/alienfx-tools/issues/194

MaluNoPeleke, 2022-10-25:
Here it is:
[data.zip](https://github.com/T-Troll/alienfx-tools/files/9863907/data.zip)

T-Troll, 2022-11-06:
Sorry for delay.

It's a mystery. You have the same device ID as before, but it used a completely different interface to set lights!
But, being honest, more simple one.

I don't know, is it a firmware or different hardware mode.
Anyway, let me implement it.

MaluNoPeleke, 2022-11-07:
Okay, stopped AWCC in the services.msc first, updated AlienFX to latest version (7.4.4.1) and changed colors again: 
[data.zip](https://github.com/T-Troll/alienfx-tools/files/9955272/data.zip)

T-Troll, 2022-11-08:
To make sure - is it a log from AWCC or my GUI?
If first, did AWCC change colors now?

T-Troll, 2022-11-08:
Ohshi... Seems like i found a source of this issue.
Please check this test version (both CLI and GUI inside) - [TestRelease.zip](https://github.com/T-Troll/alienfx-tools/files/9958026/TestRelease.zip)

It should work now.

MaluNoPeleke, 2022-11-08:
https://photos.app.goo.gl/qu5UfAtV7gem2wCd7

T-Troll, 2022-11-09:
Got it. Did the same happened in case you just press "Refresh" button in UI (not in Ambient mode)?

T-Troll, 2022-11-09:
NB: I configure out you have the same light into the 2 groups involved into Ambient. So it set 2 times with different color, which can provide this effect. It's useless, in fact. I will provide check for double light set (not only Ambient one) into the next release.

MaluNoPeleke, 2022-11-09:
Okay, great

T-Troll, 2022-11-09:
But what about just set some color and press "Refresh" button? Did it provide other color flash as well?

T-Troll, 2022-11-10:
Please check [7.4.6](https://github.com/T-Troll/alienfx-tools/releases/tag/7.4.6), it's having previous fixes for your device, as well as lights de-duplication and some other bug fixes.

MaluNoPeleke, 2022-11-10:
Thanks, but it is still hard for me to figure out how to configure it correctly. Lights are flickering purple with ambient mode turned on 
[image]
If I start from scratch, what exactly should I click on? It is not easy to identify for example which areas of the screen/RGBs is currently selected and which not. Maybe you could improve the usability to make it more intuitive?

T-Troll, 2022-11-10:
> Lights are flickering purple with ambient mode turned on

Did you check this?
> But what about just set some color and press "Refresh" button? Did it provide other color flash as well?

>  It is not easy to identify for example which areas of the screen/RGBs is currently selected and which not.

Yes, i don't so happy about current selection as well, but it's hard to do it more contrast. But i'll try.
The concept is easy - you have zone, and map it to follow the screen part color. Screen grid can be resized as well, so tune it to your light's position. I think you need 3x3 grid in your case.

MaluNoPeleke, 2022-11-11:
Yes, checked it with single color mode as well (Effect dropdown on "Off").

T-Troll, 2022-11-11:
Did you get the same effect (other color blink) then effect off and you press "Refresh" button?

MaluNoPeleke, 2022-11-11:
Yes

T-Troll, 2022-11-11:
Splendid!
Seems like i know the source of this... Let me build test version.

T-Troll, 2022-11-12:
Ok, let's try this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9994509/alienfx-gui.zip)
It should only blink violet at start, not every update.

In case it works correctly - great.
In case it not or don't switch colors at all - please do this:
- Make sure AWCCService is running (NOT stopped!)
- Open USBPcap, start recording
- Open CMD into tools folder and issue the command `alienfx-cli high-level setcolor=0,0,255,0,0 setcolor=0,1,0,255,0` (colors should be changed on monitor)
- Share the .pcap file.

This give me a hint how AWCC set separate lights via it's SDK, not AWCC in itself.

translucentfocus, 2022-11-12:
> Ok, let's try this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9994509/alienfx-gui.zip) It should only blink violet at start, not every update.
> 
> In case it works correctly - great. In case it not or don't switch colors at all - please do this:
> 
> * Make sure AWCCService is running (NOT stopped!)
> * Open USBPcap, start recording
> * Open CMD into tools folder and issue the command `alienfx-cli high-level set-one=0,0,255,0,0 set-one=0,1,0,255,0` (colors should be changed on monitor)
> * Share the .pcap file.
> 
> This give me a hint how AWCC set separate lights via it's SDK, not AWCC in itself.

I was able to follow your directions on our test sample.

[Ambient.txt](https://github.com/T-Troll/alienfx-tools/files/9994713/Ambient.txt)

translucentfocus, 2022-11-12:
> It's a mystery. You have the same device ID as before, but it used a completely different interface to set lights! But, being honest, more simple one.
> 
> I don't know, is it a firmware or different hardware mode. Anyway, let me implement it.
There are at least two different firmware versions of the monitor that do change how things function.  It also appears that @MaluNoPeleke is in the EU, so there may be difference between the EU and NA (North America) versions of the monitor.

translucentfocus, 2022-11-12:
> Thanks, but it is still hard for me to figure out how to configure it correctly. Lights are flickering purple with ambient mode turned on [image] If I start from scratch, what exactly should I click [...]

This is my fault.  I can gie @T-Troll a better default config for this monitor.  For example, I made the mistake of calling it **Stand** light, when the firmware controls call it the **Ring** light.  In addition, there's no reason to make it a literal ring in the grid because it is seen as _one_ light.  This means I can simplify the grid substantially.

T-Troll, 2022-11-12:
@translucentfocus Ooohh... welcome back! 

Being honest, this interface is a most problematic among i try to support.

> There are at least two different firmware versions of the monitor that do change how things function.

No, it was my fault - i think @MaluNoPeleke share AWCC log, but it was mine GUI log. I configure it out later and found a bug providing this result.

>  In addition, there's no reason to make it a literal ring in the grid

No, it has a reason. Making precise grid have some benefits for other effect types, especially grid ones. I'm working on it now, it will be much better soon.

Thank you for pcap, i'll check it. 

For now, seems like lights is mixed if set in series, so i need to find a way to reset/update previous set before starting next.

T-Troll, 2022-11-13:
Hmm... If the log @translucentfocus present me from high-level cli - things become way more simple!

Can anyone capture another high-level cli log for me, but the command now will be different:
`alienfx-cli high-level set-one=0,0,255,0,0 set-one=0,1,0,255,0 set-one=0,0,0,0,255`?

Also, please, check - is light 0 blink violet on not into the process?

translucentfocus, 2022-11-13:
> Hmm... If the log @translucentfocus present me from high-level cli - things become way more simple!
>
> Can anyone capture another high-level cli log for me, but the command now will be different: [...]
[set-onetest2.txt](https://github.com/T-Troll/alienfx-tools/files/9997251/set-onetest2.txt)

Hopefully @MaluNoPeleke can also give you logs just to verify I'm not screwing something up 😆

I checked, and light 0 doesn't seem to do anything when I run
`alienfx-cli high-level set-one=0,0,255,0,0 set-one=0,1,0,255,0 set-one=0,0,0,0,255`

MaluNoPeleke, 2022-11-13:
Here is a log of this and the previous command. The last one was setting a color via the GUI.
[light2.zip](https://github.com/T-Troll/alienfx-tools/files/9998134/light2.zip)

T-Troll, 2022-11-13:
Guys, are you sure you send me log from last command (3 x set-one), not previous (2 x set-one)?

I see only 2 light sets (first and second set-one) into it - in both logs (@MaluNoPeleke, i even see you have AWCC service running ^_^, but seems like @translucentfocus  doesn't have it).

Anyway... Open issue is solved, one of the commands i use as "block start" is useless (or even harmful), so i already remove it from yesterday's 7.5.0. It can be the source of violet flash.

So... 7.5.0 should work better. But... There is a probability it can only change color once (not so big one, according to what AWCC do in @MaluNoPeleke example). Please check (you can also use latest CLI command for test, but without "high-level").

MaluNoPeleke, 2022-11-14:
I tried out 7.5.0 and it works fine, even in Ambient mode. Thanks a lot for your efforts!
Do you know what the performance impact is in that mode?

closed 2022-11-14 by MaluNoPeleke

T-Troll, 2022-11-14:
> Do you know what the performance impact is in that mode?

Yes. 1.5-4% of 6-cores CPU depends on screen resolution, and it's grid size.