T-Troll/alienfx-tools #225: G-Mode Brightness Feature

Killerpac, 2022-10-28:
**Is your feature request related to a problem? Please describe.**
When using Alienware Command Centre and Pressing the G-Mode Button the Fans Speed increase as Well as the Brightness of the display Increases but Alienfx-Tools doesn't support that feature for Brightness Increase It is a Necessary Feature Because We cannot Identify The G-Mode Turning On if We have Headphones On 

**Describe the solution you'd like**
Brightness Increase To A Certain Point By a Slider Or Full 

**Describe alternatives you've considered**
Considered Using a Alternative Keymap Tool To increase the Breightness as well as Tuning On the G-Mode but it will use extra memory and cpu to keep it running

**System (please complete the following information):**
 - Model: Dell G15 5515
 - Model details: Orange Backlit
 - OS: Windows 11 22H2

**Additional context**
The 7.0 Update Fixed Many Issues For My Laptop and Have Replaced Alienware Command Centre With AlenFx Continue The Good Work And Also Can you Implement Something About the Nahamic Audio Controls

T-Troll, 2022-10-29:
AWCC doesn't switch brightness, in fact - it switches Windows Power Play to high performance (check).
I don't want to alter other settings by my apps, and prefer to keep it simple and predictable.
Right now, `aliienfan-gui` show current power mode (including G-mode) at application tooltip (move mouse to app tray icon). I can do the same for `alienfx-gui`, but other indication (like pop-up etc) looks annoying for many users (including me).

Killerpac, 2022-10-29:
I don't see any power plan other than balanced in my laptop the high performance is hidden but what you said is true but as i was saying if we are in-game and suddenly wanted to turn on G-Mode there is no way to confirm if the G-Mode is turned on Properly without the Brightness Indicator If headphones in on .

So Like Can you create an option in the ALienfan-gui to trigger thr high prtformance mode if G-Mode Enabled it will be an optional flag where if needed a person can press the toogle to enable the feature

T-Troll, 2022-12-27:
In 7.10.1, i add power mode monitoring into `alienfx-gui`. So now you can set any light to reflect is G-mode enabled or not (as well as any other power mode).

closed 2022-12-27 by T-Troll

Killerpac, 2022-12-29:
i can't find the settings to set it could you guide me as to how its done

T-Troll, 2022-12-30:
Here we are:
- You should use `AlienFX-GUI`, not `AlienFan-GUI`
- Active profile effects mode should be "Monitoring"
- Switch to "Events monitoring" tab, select/create lights zone
- Select Performance/Indicator with source "Power Mode"
- Press "Add" ("+") button to add event
- Select colors for low mode ("From") and high mode ("to")

For "Performance" mode, lights color will be always "From" for Manual mode, and always "To" for G-Mode. Other power modes will have color mixed between them.
For "Indicator" mode, Manual mode color will be always "From", and it will be switched to the "To" depends on "Trigger level" slider (initially any other mode switch it, move slider right to switch at later mode, move to 90+ to switch for G-Mode only).

Killerpac, 2022-12-30:
i think it requires the light part as g15 5515 doesn't have lights so that tab is not visible
[image]

T-Troll, 2022-12-30:
Unfortunately, yes. The rest of the ways (like brightness, numlook, etc) are dirty and annoy many people.

freedxmgxd, 2025-05-20:
> In 7.10.1, i add power mode monitoring into `alienfx-gui`. So now you can set any light to reflect is G-mode enabled or not (as well as any other power mode).

I have a suggestion to improve this if it is not much complicated change the color of taskbar icon when g-mode is enabled. Showing in the text when I put mouse over it's great but with color change I think would be more intuitive.

T-Troll, 2025-05-21:
@freedxmgxd This is not so easy, because i use a bunch of icons for different LIGHT conditions. Mix this with G-mode don't look like the great idea. But i got you point, let me think.