T-Troll/alienfx-tools #226: Fan control crash and close after sleep mode.

Alissonverd, 2022-10-31:
**Describe the bug**
The FanControl app eventually closes after windows come back from sleep mode. So, I started to use AlienFX Monitor to see when my fans are 0 RPM and check if the FanControl are closed.
Sometimes the FanControl doesn't close, but the fans stay at 0 RPM..
Please, check the screenshoot.

**To Reproduce**
Steps to reproduce the behavior:
1. Wait Windows go to Sleep.
2. Use a trigger to comeback from sleep mode (mouse, keyboard..).
3. Check RPM;
4. Check FanControl.

**Expected behavior**
When comeback from sleep mode, check fan RPM and set the correct boost, and if FanControl are closed we need open a new instance.

**Screenshots**
[image]
(left - normal)  (right - after comeback from sleep mode)

**System (please complete the following information):**
 - Model: Dell G15 5511
 - Model details:
       Core i7-11800H
       RTX3060
       Chipset HM570
 - OS:
       Windows 11 Pro 22621.755 updated
       All Dell drivers and Bios updated
       Alienware Command Control (uninstalled)

**Additional context**
Sometimes I the notebook are very hot during sleep mode

T-Troll, 2022-11-01:
Definitely a bug with resume. Let me check.

> Sometimes I the notebook are very hot during sleep mode

See #201. This usually happened if notebook not really sleep or waked up by Windows update.
BTW, do you have "Start with Windows" enabled? Update can reboot notebook...

T-Troll, 2022-11-03:
Interesting...
I can't reproduce your bug, but found something really strange.
You fan percent in Mon is -2. This is really strange value, it can be -1 in case BIOS request fail, but -2 mean some bug into BIOS.

This can be the source of issue (but should not, in fact), i'll check.

T-Troll, 2022-11-04:
Check latest release (7.4.4), i fix one possible crash point, but don't sure.

Alissonverd, 2022-11-04:
> BTW, do you have "Start with Windows" enabled? Update can reboot notebook...
Yes. Only FX Monitor I start manually. but now I set to start with windows.

> You fan percent in Mon is -2. This is really strange value, it can be -1 in case BIOS request fail, but -2 mean some bug into BIOS.
Updated, but fan percent still -2
[image]

Thank you for support and dev this app.
If you want any logfile, could you explain where I get these files?

T-Troll, 2022-11-05:
I check your notebook BIOS dump (i have it).
1. You have no "GetFanPercent" function into BIOS (WMAX command 14 function 6)!
2. "-2" is an error value into your BIOS.
3. Both FX-GUI and Fan-GUI/CLI use overboost-defined value to calculate fan percent (if defined), but FX-Mon use BIOS value (it doesn't have access to overboost config).

So, -2 mystery is solved. I'll patch Mon to threat it as "Error value".

What about GUI crash? BTW, is FX-GUI crash as well?

> If you want any logfile, could you explain where I get these files?

I've tried to keep the app's size as tiny as possible, so no logs into release version. Some in debug, so if this continues, i'll share one to check.

T-Troll, 2022-11-05:
Also, Fans in Manual mode with boost 0 can stop under 50-55C (depend on model), so it's OK.
Seems like Manual set as BIOS default for your system now.

You can alter it - close GUI (but you can keep Mon), open Administrator CMD and issue the command `alienfan-cli setpower=1` - this switch BIOS to auto mode. GUI will respect this for quit/sleep/hibernate.

Alissonverd, 2022-11-05:
>AlienFan-CLI v7.4.4
Supported hardware (3200) detected, 2 fans, 13 sensors, 5 power states, G-Mode.
Power set to 1 (result 0)

Done.

>What about GUI crash? BTW, is FX-GUI crash as well?

The system sleeped 3 times after new version.. Everything ok so far with AlienFan and Mon..
I dont use AlienFX Control because I dont like lights keyboard, so I dont need set it.

This app is very better than AWCC, Thank you for developing it!!!!
have a donate link?? How to pay a coffee for you?

PS...
Writing this reply AlienFan crashed  =(
But fans still running +2k RPM
[image]

T-Troll, 2022-11-05:
> I dont use AlienFX Control because I dont like lights keyboard, so I dont need set it.

You can not set lights if you don't like. But it also about profiles (including different fan profiles), which can be switched by app/power/etc. More advanced about it.

> Writing this reply AlienFan crash =(
> But fans still running +2k RPM

Ooops... So it's a random crash. Let me check.
But seems like it restore thermal profile (or just keep boosts).

> have a donate link?? How to pay a coffee for you?

I don't accept any donates except Alienware/Dell-G hardware for testing.

> Edit: EventViewer Language Portuguese-BR

Thanks, let me take a look. I have a lot of users from Brazil, so i feel like i will understand it soon. ^_^

T-Troll, 2022-11-06:
Try this fan-gui - [alienfan-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9944913/alienfan-gui.zip)
It's a debug version of the next release. I fix something, but not sure whether it's the source of your issue.

Alissonverd, 2022-11-08:
Now AlienFX are stable. No crashes anymore.

But how set to only "Manual" power mode.

closed 2022-11-23 by T-Troll