T-Troll/alienfx-tools #234: M15R7 - Nvme sensors locked at 60 degrees

luiz-fernandoalves, 2022-11-07:
Describe the bug
SSD sensors are always showing 60 degrees.

To Reproduce

Open alienfx-monitor or alienfx-control and check values, they're different from samsung magician values for example. 

Expected behavior
SSD sensors must show the nvme drive temperatures. 

System (please complete the following information):

Model: Intel Core i7 M15R7
OS: win 11 - Build 22621.755 -  Experience Pack 1000.22636.1000.0
Core Isolation - Memory Integrity:  Enabled
AWCC: Not installed 

[image]

T-Troll, 2022-11-07:
Most probably it's not a bug - some SSDs have a fake temperature sensor, it's always return constant value into SMART (where i get it for GUI).

So check your SSDs SMART readings first.

Also, you should have MB SSD sensor as well - it's one of ESIF. Can't say exactly until take a look into BIOS dump (Dell name it, but Intel don't expose names).

luiz-fernandoalves, 2022-11-08:
The only information about my motherboard, from HWiNFO :
Alienware 0TJXCY
Chipset: Intel Alder Lake-P PCH

From SMART: 
[image]

From HWiNFO :
[image]

Actual readings from fx-control
[image]

T-Troll, 2022-11-08:
I see. But it locked into WMI to fake value.
What NVMe driver you are using? Windows default? Samsung?

luiz-fernandoalves, 2022-11-08:
I've used Intel RAID drivers at windows install. Then, windows defaults.

T-Troll, 2022-11-08:
So, you have it at RAID0?

luiz-fernandoalves, 2022-11-08:
Nop. I've used this default option, but there's no RAID 0 setup:

[image]

luiz-fernandoalves, 2022-11-12:
Hey bro ..... I've changed the BIOS option to NVME and now the sensors are working fine.

[image]

T-Troll, 2022-11-12:
Aha, so RAID drivers report fake.
In case you're interested, i take this values from WMI (entry `ROOT\\Microsoft\\Windows\\Storage\\Providers_v2\\MSFT_PhysicalDiskToStorageReliabilityCounter`), which is, seems like filled not by system, but by storage driver.

Anyway, thanks for this bug. i'll add it into readme!

closed 2022-11-12 by T-Troll