T-Troll/alienfx-tools #240: Upgrading from 7.4.0 to 7.5.0 Settings Lost

TheSQLGuru, 2022-11-13:
Alienware X17 R1, fully patched including latest BIOS, 11980HK/3080, 32GB RAM

When I just upgraded from 7.4.0 to 7.5.0, I experienced the following with respect to FC:

1) The application tool a LONG time to start (15 seconds-ish). While doing this, a blank Fan curve window was shown for most of the duration.

2) CPU Boost AC and Battery settings were changed from the values I had set to both being off. I apologize, but I cannot recall the inital values with certainty.

3) I don't know that this is a bug, but I now see that the Fan curve window has scale, boost and a percentage listed in the title bar. 

My 4 values right now are:
169/0/40
156/0/37
252/0/0
250/0/0

a) I found no way to set the scales (presumably temp?), nor any documentation on them.

b) What are the boost and % values? Please add to documentation.

4) The X to reset min/max temp sensor ranges: 

a) resulted in 10-15 second app hang the first time I tried it, and then a hang and then hard crash the second time. Neither of these happened with a subsequent attempt though.

b) Per the .md documentation:

'"X" button above fans list reset currently selected fan curve to default one (0-100 boost).'

c) I VERY much hope you provide an "Are You Sure?" prompt before you clear someone's manual fan control settings!! :-D

5) The application now gives UAC prompt at each start. This has not happened on any prior version.

6) IIRC, 7.4.0 picked up my internal NVMe SSD temp for display, which I understand can give me another way to control fan speeds manually based on its temperature. I like it.

7.5.0 picked up my two EXTERNAL Samsung T5 and T7 drives and added them to the temperature sensors window. Both are 0 (0), because they presumably do not report temps. In any case, I feel you should never add non-internal temperatures to the FC app because they are not helpful with respect to maintaining internal component temps with the fans.

7) KUDOS for all the recent work on the application(s)!!! Part of me wishes I gave a shit about the "bling" of my laptop so I could play around with the FX. ;-)

T-Troll, 2022-11-14:
1. This just CAN't happen. App can hang with no UI shown for a while (in case WMI request delayed), but this... Something wrong with the system. Let me think how to check it... Does CLI have the same delay?
2. It's Windows Power Plan - related. So it follows active Power Plan.
3. Hmm... Seems like last 2 fans can't be controlled.... Or report incorrect data. Try to set it using CLI and see is it works or not.
  a) Scale = Max. overboost level. Use Overboost button/`setover` command in CLI.
  b) Ok, i clarify it a bit.
4. Bug. I check how this can happen, it should be milliseconds. I'll think about clean confirmation (although i prefer not bother user and remove it from any possible places).
5. It can be Windows update (Windows 11?). There are no changes into this part, but they can MAY change Task schedule settings for it.
6. It's not possible to configure out, is it internal or external. Even for internal, sometime driver report fake temperature. I'll add 0-check, so drives without temp will not be exposed.

PS: I always into bargain between add some bells and whistles and keep apps as small as possible. I usually choose second.

TheSQLGuru, 2022-11-14:
1. This just CAN't happen. App can hang with no UI shown for a while (in case WMI request delayed), but this... Something wrong with the system. Let me think how to check it... Does CLI have the same delay?

    I realized that it could have been my DisplayFusion app. It has a number of issues, and I am close to discarding it despite really needing its core functionality.

2. It's Windows Power Plan - related. So it follows active Power Plan.

   Ahh, I think I recall that from our previous conversations. I will just leave it at Off and manage via other means.

3. Hmm... Seems like last 2 fans can't be controlled.... Or report incorrect data. Try to set it using CLI and see is it works or not.
    a) Scale = Max. overboost level. Use Overboost button/setover command in CLI.
    b) Ok, i clarify it a bit.

   I need to clarify so as to correct your notions: My 3rd and 4th set of numbers represent Fan 3 and Fan 4. These are "associated" with my GPU Internal Thermistor. Those fans are DEFINITELY controllable, and they work fine with my manual curves.
 
   WOW! My GPU fans can boost to 250% of normal max?!? I did a quick registry search and didn't find a value I could set to restrict that. I guess I will just lower my curves to reflext the overboost amount I am comfortable with.

   Since you mentioned alienfa-cli, I tried it. I tried the following to attempt to get some help: 

alienfan-cli help
alienfan-cli "help"
alienfan-cli -help
alienfan-cli --help
alienfan-cli /help

I received "Unknown command xxxxxx, use "usage" or "help" for information" for every call. What's the story here. 

4. Bug. I check how this can happen, it should be milliseconds. I'll think about clean confirmation (although i prefer not bother user and remove it from any possible places).

   I understand the confirmation issue. I tried to think of one to suggest, and didn't come up with one. But I did while writing this: place a semi-transparent overlay covering the Fan Curve grid that asks "Delete Manual Fan Curve", with Yes and No buttons below the text. 

5. It can be Windows update (Windows 11?). There are no changes into this part, but they can MAY change Task schedule settings for it.

   Not on Win 11. It is only a minor issue. I wonder if there is a way to remove UAC for individual application starts. Perhaps you are aware of such?

6. It's not possible to configure out, is it internal or external. Even for internal, sometime driver report fake temperature. I'll add 0-check, so drives without temp will not be exposed.

[image]

I would guess in those values there is something that indicates no temperature sensor?

PS: I always into bargain between add some bells and whistles and keep apps as small as possible. I usually choose second.

   I haven't checked the app's various memory usage metrics, but I can't imagine it is very much - unless you really suck at windows app programming. LOL I think the thing that is most difficult/limiting for you is the need to support what must be a HUGE permutation of hardware. You can only abstract/modularize so much, especially when myriad individual pieces-parts need some sort of exceptional handling. 

PPS: I'm interested into x17 BIOS dump, doesn't have one from this model. Can you please share me it in Discord? BTW, it can be the source of 1,3 and 4.

  Just let me know how to get you one, and the Discord channel to get it to you, and I would be happy to help out.

T-Troll, 2022-11-14:
3. It's into `HKCU\Software\AlienFan\Boost-X`, there X is fan number.  Change first byte to 0x64 for 100%, or use `alienfan-cli setover=N,100`, N for fan number.
> Since you mentioned alienfa-cli, I tried it. I tried the following to attempt to get some help:

Ohsi... I forget to change this message. Just run it without parameters.

5. Starting from Task Scheduler can do it. Flip "Start with windows" off and on to re-create app entry.
6. No, this indicates default curve (0,0 - 100,100).

In fact, 4 and 6 is not valid anymore, i rebuilding all internal fan structures now to avoid mess with sensors here and in mon. Also, much simple and faster.

> I haven't checked the app's various memory usage metrics

Not so small, about 20Mb RAM. But it's part of the fun to keep it small. Hardware is not an issue (it's standard and tiny), but UI is. So i trying to use it as standard and as close to plain C as possible.
But this is a part of fun, i got mad skills, useful for my main IoT duties.

> Just let me know how to get you one, and the Discord channel to get it to you, and I would be happy to help out.

As mentioned into project readme - https://discord.gg/XU6UJbN9J5
I have the same nickname there.

T-Troll, 2022-11-16:
Here is next release beta - [alienfan-gui.zip](https://github.com/T-Troll/alienfx-tools/files/10017291/alienfan-gui.zip)

It's A LOT of changes (i refactor all internal structures), it should fix all, except incorrect fan readings.
Please, also, keep in mind fan selection logic changed:
- In case fan unchecked for sensor, it's now keep curve until app restart, so check it back return it.
- "X" button above fans now have confirmation, and remove ALL curves for this fan (related to previous point, in fact).

I still kindly waiting for x17 ACPI dump - i need to check situation with other 2 fans.

T-Troll, 2022-12-12:
I think this one outdated yet.

closed 2022-12-12 by T-Troll