T-Troll/alienfx-tools #241: AlienFX Monitor First Time Use Findings

TheSQLGuru, 2022-11-13:
1) Main Sensors panel:

a) Auto-expand the 4 columns

b) Double-click column divider does not auto-expand column to minimum required width to show all contained data

2) Allow selecting multiple rows and setting Show in Tray options. 

3) When I switched to Use BIOS sensors, all  sensors were removed from panel. I tried Show Hidden checkbox, and saw what appeared to be the same 5 Windows sensors. If they ARE the same, then this is the wrong way to do this and it is covered in #4 below. If they are NOT the same, then #4 covers them as well.

4) Make Win/BIOS sensor check boxes Radio Buttons, so only one can be active at a time? If you continue to allow both, you need to demark which is which since the names are currently the same, and you should stop using the "hidden" functionality to "swap" sets.

5) I could find no function for Reset button.

6) Rename the label Sensor settings to "Selected Sensor Settings".

7) What is the purpose of the Then lower checkbox? I would think it is for removing the alarm if the value goes below some set value. The problem is that you do not provide a second value. I recommend showing a second value and label it "Reset at". If you do this, double the height of the Sensor Settings component and place Reset at check box and value box below the Alarm at ones.

8) An even BETTER way to handle the entire Sensor Settings functionality is to add ALL of this to the Sensors panel. Easy-peasy, lemon-squeezy!! Don't make people click all over the place back-and-forth.

9) OK, I remembered there was a Read Me! :-0 Now I see that you CAN have multiple sensors selected, and that they all show. This definitely means you need to somehow demark each shown sensor type. I recommend adding a column via #8's improved mechanism called Type and using values of W, B and A for Windows, BIOS and Alienware. 

10) #2 above (select multiple rows) is now ESPECIALLY important since it seems that the Reset min/max button will reset ALL rows shown. 
I recommend showing TWO bottons for Reset min/max - one for ALL  and one for Selected.

11) When I mimize the app, it disappears. It is not on my task bar, nor in the tray. It is still running though, via Process Explorer. When I started it up again via Start Menu, I now have two executables running. I think this is a definite no-no. Check for already running process and force the app window to display at its last location. 
UPDATE: its seems that the first app running did not show in task bar or tray. The second one DID drop to the tray when I minimized it. 

12) Running PerfMon, I see that you make these 5 registry calls every second. What is the purpose for this repeated set?

query key: HKLM
open: HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\Perflib
close: HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\Perflib
queryvalue: HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\Perflib\Disable Performance Counters
query key: HKLM

That's plenty for now. I have some real work to do! Kudo's and thanks again for your tooling for the Alienware community!!

T-Troll, 2022-11-14:
1. This is a known bug, VERY hard to fix. I think i do it into next app re-factoring, i have other things i don't so happy about. For now, just minimize-restore app and it will be resized.
2. Ok, let me think.
3. Bug. List not in sync after it... Ok, let me check how it happened, i have this too.
4. No. They have different set of values. In fact, once-used sensors always remembered, just don't updated. Currently, BIOS have only powers and expose temps only if Alieware disabled (Alienware have wider temp set, including BIOS temp). Windows also have RAM load, GPU load, HDD load, network load.
5. It reset min/max for this sensor only.
6. Ok, it's more clear.
7. Direction. If not checked, alarm triggered, then value become higher than setting. If checked - then it becomes lower than setting.
8. No. It's hard to find sensors in this mess, as well as configure out is it active or not. So it should be 2 lists... But maybe i remove inactive by source from hidden.
9. Good point. Type can be useful, i'll add it.
10. THERE ARE two buttons - "Reset min/max" for all and "Reset" for current. First one also accessible from tray menu.
11. Some mystery. Tray icons (both app and tray sensors) are ALWAYS visible from app start, minimized or not. The only ways to have this situation are no tray available (but i handle this and add them then tray started), or Windows not allow adding icon (return error). I don't know how to handle the 2nd one (and it looks the same from code part).
12. Aye. Windows sensors use performance indicators. Seems like i need to stop/start collection if windows sensor's state altered. Thanks.

T-Troll, 2022-11-16:
Here is next release beta - [alienfx-mon.zip](https://github.com/T-Troll/alienfx-tools/files/10017325/alienfx-mon.zip)

It's A LOT of changes (i refactor internal structures as well, and change logic).
No more crazy names (but maybe you need to clear registry - i'll do it into final version), and most of other points fixed (except multiply selection).
Additionally, now both main and hidden list only show sensors from active sources.
Also, clearing sensor name update it to default (BIOS) name.
I add 2-letter code for sensor source/type, and this version also have sensor ID (for debug, i remove it later).

TheSQLGuru, 2022-11-19:
FOLLOW UP TO ORIGINAL FINDINGS:

1.1 How can it be a bug that you can't make the columns auto-expand or expand to fit on double-clicking a column divider, much less one that is "VERY hard to fix"?!? It is a simple windows control! :-) Use a different grid if the one you have doesn't easily allow that. 

1.2 Since you are apparently going to rework the main panel in the next major release, consider adding in (multi-)column sorting. This too should be a standard windows control feature, and thus easy to implement. Some people may want to see things ordered by name, especially when they are using multiple sources. 

4.1 This one is irrelevant since you are going to add a column to the main panel indicating which Source of sensor each row is. 

7.1 That is quite obtuse. Here is a suggestion:

[image]

8.1 " It's hard to find sensors in this mess, as well as configure out is it active or not. ". That is EXACTLY why it needs to be changed!! I am REALLY REALLY pushing you to put all of the INFORMATION about each setting as columns in the main panel. Trust me, that is absolutely the best way to go. Actually EVERYTHING can easily go in the main panel grid. Decent grid controls are easy to program. The checkboxes for alarms can go in two columns: the first would simply be the Alarm value, and if the column contains a value then the alarm is On. The other column would contain the two checkboxes for When Above and When Below. You could put those into separate columns if you wanted, but either way make sure that the check boxes are mutually exclusive (thus radio buttons are a better design choice).

Let me know if you are interested in pursuing this and I can stub out the GUI if you like.

9.1 I am changing what this should be called. It should be Source, not Type. Type, to me, would be fan, temp, etc. Source is a much more descriptive word. 

11.1 I think this may have been yet another artifact of DisplayFusion. I am currently running with that disabled, and when I minimize AlienMon, it goes to the tray as expected. 

NEW FINDINGS

13. I had the app opened, not as admin, Windows as source, cpu load set to show in the tray. I then added Alienware as the source, and the app restarted with UAC prompt. When it was back up, I had TWO cpu load values showing in the tray. When I moved the mouse over them, one disappeared. I have seen other apps so similar things, so this may be a flaw in the tray management routines of windows.

T-Troll, 2022-11-20:
1.1 It was - updates coming other way before. I rework this in latest release completely.
1.2 No, it's not so easy. Windows only provide events for it, but not complex implementation. I'll think about it.
7.1Occam's razor didn't allow me to make 2 checkboxes instead of 1. But i got it can be misunderstanded, so i think about a correct name for it.
8.1 It's funny. No, List Control is not easy to handle and buggy in many places (especially checkboxes). I have a lot of issues at alienfan-gui using it. However, i don't like to have so much columns - first it harder to read, second it's limit name width.

Yes, any suggestion about UI, even if will not implement, is appreciated. Just keep it mind it can only LOOKS easy, but really not so. In fact, Mon was a simple task - i just need a fast way to add sensor icons to tray. Maybe UI need the global re-design, including different concept.

9.1 It's name IS "Source". Type is useful to understand what is it, especially for Windows sensors. F.e. 55 is RAM load, CPU load or Temperature? Or power consume? Default names can not always explain it.

11.1 I meet this situation with alienfx-gui one time during Windows update restart. I made some countermeasures anyway, app should be reachable anyway!

13. Yes. Fixed yet.

TheSQLGuru, 2022-11-25:
BUG: I just did a quick run of AFXM and the Sensors grid starts with the left 4 columns collapsed. When expanded, any/all of them are recompressed to min width on very refresh cycle. 

That's all I had a chance to review.

T-Troll, 2023-02-20:
Outdated now. Please open new issue if you find something else.

closed 2023-02-20 by T-Troll