T-Troll/alienfx-tools #247: AlienFan Lost its Mind

TheSQLGuru, 2022-11-19:
1) on 7.5.0, AlienFan stopped honoring my manual fan curve, and stopped setting fans to speeds it used to on the various Levels.

2) 7.6.1 exhibits the same behaviors. 

[image]
[image]

Level 6 should be WIDE OPEN on my fans. And manual sees the red dot always stuck on the horizontal axis, not tracking my curves. I used Cinebench to stress CPU, and fans never left the 0 point, although the red dot did track the temp horizontally.

I have no idea why this may have changed. I will do an uninstall, make sure the registry is cleaned up, and then reinstalle 7.6.1 and see how it does.

TheSQLGuru, 2022-11-19:
When I uninstalled AlienFX, the following regisitry areas still had AlienFX entries. I think a full uninstall should delete all traces of the application - files and registry. 

HKEY_CURRENT_USER\Software\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\Layers
HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\UFH\SHC
HKEY_CURRENT_USER\Software\Classes\Local Settings\Software\Microsoft\Windows\Shell\MuiCache

Actually, best would be to ask the user if they intend to reinstall/if they want to keep their registry entries related to their settings (like fan curves, etc.)

TheSQLGuru, 2022-11-19:
Hmm, now I find that you didn't delete any of the actual settings registry entries upon uninstallation of the app. That is not good IMNSHO. Ask if user wants to keep those or not. 

When I reinstalled 7.6.1, it obviously picked up the 4 registry entries I had before. The behavior was still broken: Manual did not track my curve, and levels did not boost fans as expected.

I have attached my 4 registry entries them to this post in case they may help you debug the issue.

I have deleted them, and will NOW reinstall 7.6.1 again with a clean registry, and see how it does.
[AlienFX_Registry.zip](https://github.com/T-Troll/alienfx-tools/files/10047822/AlienFX_Registry.zip)

TheSQLGuru, 2022-11-19:
WOW - LOTs of bugs in 7.6.1. I did a COMPLETELY fresh install - after manually deleting all registry entries in HCU\software.

A) After installation, in Start, the AlienFan icon is missing the image. AlienFX and Mon icons are correct. 

[image]

B) Levels 1-6 seems to behave as they used to.

C) FC does not honor manual curves. I tried going above curves with level 6 and then back to manual, and using level 1 and then back to manual. No change. 

[image]
[image]
[image]
[image]

D) AlienFan does not save my custom curves!! Fans 1,2 and 3 do not get saved. Maybe this has to do with the fan red dot seeming to remain right along the default manual line of 0-100/0-100 slope? But fan 4 DOES save, but the fan red dot also stays way above my curve. 

Hmm, check these two images out. In the older apps, CPU would never show a line for fans 3 or 4, but it does now. CPU was fan 1 & 2, GPU was fans 3 & 4. This could also explain why the fan is remaining above my set curves. 

I recommend: 

i) never show yellow curves. Only show curves for the selected pair (cpu/gpu, selected fan number). This is very confusing and clutters the graph. It also leads to the issue of having the two default lines overlap, giving a color that is yellow-green combined. 

ii) only KEEP a green line fan curve if they make some change to the default line. If user doesn't change it, they don't want that element (cpu/gpu) to affect that fan in their manual curves.

E) In addition to fixing whatever bug is making the app not save my manual fan curves, please also provide the ability to copy a fan curve from another sensor/fan pair. I would think it would be common for those that use manual curves to use the same curve more than once. A right-click on the grid could bring up a dialog with all sensor/fan pairs that have a manual curve and radio buttons for the user to pick which one they want. A bonus implementation would provide a small image of each curve.

T-Troll, 2022-11-20:
Ok, some points.

1. Fans can ONLY be controlled at Manual mode. It's BIOS limitation. HINT: if you see boost point always at 0 or 100, this mode doesn't support boost control.
2. All free installers are dumb - they also clear registry at update. Being honest, .msi install is secondary for me, .zip primary, so i don't want to put any effort here anymore. Use `alienfx-config` to clear configuration.
3. I found a bug - disabled curves still participate into boost calculation. Fix it now.
4. It was something wrong with fan IDs at your gear. I make some changes according to the dump you provide, should be ok now about "Not saved" curves (they have same IDs, in fact).

i) No way. It's useful for many things, f.e. for detection which curve set the boost.
ii) Just useless.
E) In common settings, there are no same curves (if you do so, your settings is not optimal) - all fans acting different. So it's a lot of effort for little profit.

Check this GUI - fans should not have mess anymore - [alienfan-gui.zip](https://github.com/T-Troll/alienfx-tools/files/10048377/alienfan-gui.zip)

TheSQLGuru, 2022-11-29:
I think you can close this one now. The later updates seemed to have address the issues here.

closed 2022-11-29 by T-Troll