T-Troll/alienfx-tools #260: How about individually light dimming?

CharlieIXXXXXXXXX, 2022-12-10:
Hi there, I've got an Dell G5 5500. It features an RGB light bar and a backlit keyboard, and I found the brightness cannot be set seperately.
The light bar is simply too bright and if I choose to dim in the setting panel the keyboard then become too dim.

Unfortunately I cannot do the actual programming, thus can not put forward a practical improvement plan.
I would appreciate it if you could add the feature of dimming specific lights.

T-Troll, 2022-12-10:
So, you have 2 devices - box and keyboard, right? What kind of keyboard?
I'll think about separate dimming per-device (for per-light it's more easy just change COLOR brightness than you set it).

CharlieIXXXXXXXXX, 2022-12-10:
Integrated Keyboard. To help you understand better, please see the photo enclosed.
[image]
And my device is not properly detected, it says I am a G5 5515 but I am using a 5500. I manually set up the lights like this: 
[image]
[image]
[image]
Since it is not detected as 2 devices, I don't think dimming per-device will be of any help...
Certainly I tried changing color brightness, however the actual color displayed became much distorted.

T-Troll, 2022-12-11:
Wow! Something new. I was sure 5000 is a desktop...

> And my device is not properly detected, it says I am a G5 5515 but I am using a 5500. I manually set up the lights like this:

No. Did you see the message on top of the detection box - DOES NOT select "similar" models!
Your system utilize quite common v4 chip (ID 550), Many others use it as well.

I have no mappings for your model in my database yet, so i will appreciate if you share yours (but your grid looks not so good now...)

Now about colors. There is a setting named "Gamma correction" at "Settings" tab. It's PWM Gamma correction - so light colors looks the same as on screen (but also brightness decreased a bit). You can disable it and see what happened. Also, you can play with dimming power at the same tab.

> ertainly I tried changing color brightness, however the actual color displayed became much distorted.
Can you explain please?

CharlieIXXXXXXXXX, 2022-12-12:
Alright, let's put the brightness aside, I'll try to deal with it. Sometimes perfectionism isn't that good, I must admit.
I tried to follow your wiki "how to add a new light device", but I had problem locating my USB light device in the device manager.
I remember Ubuntu recognized that well using lspci command ( it showed an "Alienware controller" or something ), however I can't see anything familiar in USBPcapCMD. ( maybe i ran into a corrupted installation, i can't find the gui of the software anywhere, and the cmd line tool didn't offer me useful information )
I'm glad to offer a manually manipulated current.csv...  However I find it hard to define the key of the light bar, it simply has no KEY...
Here it is. It was a 5515 config, and I rewrote the KEY section, hope it will help.
[Dell G5 5500.csv](https://github.com/T-Troll/alienfx-tools/files/10206838/Dell.G5.5500.csv)
[image]

By the way, I found my fans sometimes hit full speed (100%) and then back to normal (approximately 42%) when there is no heavy workloads AT ALL with fan control enabled, the same also happens right after the system bootup. I'll try to keep an eye on that problem, if it becomes more frequent I might open another Issue.
[image]
[image]

T-Troll, 2022-12-12:
> Here it is. It was a 5515 config, and I rewrote the KEY section, hope it will help.

You don't need to put a key (it's for RGB keyboards), but readable name. However, you remove a grid mapping for some reason, it's not good! Can you please just save it without edit?

> I found my fans sometimes hit full speed (100%) and then back to normal

It's normal for your curve. Check temperatures - after OS loading and in case you have some background tasks, it can be 80+ for a while. My app react on it.

T-Troll, 2022-12-12:
> I remember Ubuntu recognized that well using lspci command ( it showed an "Alienware controller" or something)

They know VID list ^_^
If you are tech, run `alienfx-cli probe` - it's all IDs detected. Do not set lights there, by the way.

CharlieIXXXXXXXXX, 2022-12-12:
[image]
It said DELL API not found - perhaps because I uninstalled AWCC since I've got your app. 
( I'm on a Pro edition of Windows and WMIProviderHost would suddenly take up 98% CPU and my machine just stucks. I looked into Event viewer and did a bit of twicking, however it didn't help much. **Thankfully, your app worked just fine.** I mean, no huge CPU consumption has occured so far.)

As for the grids, I found myself stupid when I suddenly realized the box beside the Key botton is available for input ( LOL )
Here it is.
[Current.csv](https://github.com/T-Troll/alienfx-tools/files/10208734/Current.csv)
[image]
I think it will be clear enough with this photo... I mean 01 is the first light bar LED **on the left.**

closed 2023-06-16 by T-Troll