T-Troll/alienfx-tools #261: Unexpected Roaring Fans

CharlieIXXXXXXXXX, 2022-12-12:
**Describe the bug**
Sometimes the fans RPM hits 100% while no heavy workload is running.

**To Reproduce**
I enabled "start with windows", and every single time I boot my machine and the lock screen shows my fans would give a load roar (exactly when the program started, AlienFX Light Effects can be seen,)
Also this happens when the machine is idle, no matter I'm operating on it ( moving mouse, typing etc ) or not. It does not follow a regular basis, maybe 15~20mins it will repeat.

**Expected behavior**
What happened in **To Reproduce**  should not happen.

**Screenshots**
[image]
[image]

**System**
 - Model: DELL G5 5500
 - Model details: 4 zone RGB keyboard with a 12 LEDs RGB light bar, 2 fans
 - OS: Windows 10 Pro 22H2 19045.2311

T-Troll, 2022-12-12:
As i tell at other topic - it's OK for you curve. CPU can have 80+ at OS loading and in case some background tasks running.

But your second screen is interesting. This means boost get return an error. I'll add the patch for this.

T-Troll, 2022-12-13:
Can you please run `alienfx-config backup` and share afx-fans.reg? I see a bug, but don't understand how it happened.

CharlieIXXXXXXXXX, 2022-12-13:
[image]
This time I've got a snapshot of the app gui. It looks weird. But the CPU fan RPM is shown normally.
More interesting though, the CPU fan roared for seconds, and after a small interval, the GPU fan started roaring and the app does not show current GPU fan RPM. It was 0.
I can't drop reg file here, so it's a txt.
 
[afx-fans.txt](https://github.com/T-Troll/alienfx-tools/files/10215966/afx-fans.txt)

CharlieIXXXXXXXXX, 2022-12-13:
Oh for that in GPU 2, I mean boost is 0. I didn't manage to take a screenshot.

T-Troll, 2022-12-18:
> "Boost-1"=hex:00,fc,15

Here are the issue. It was a bug, i fixed it in 7.9.2. So open the latest version, select GPU fan and press "X" button near to "Check max rpm".

closed 2023-01-24 by T-Troll