T-Troll/alienfx-tools #263: AlienFX Control crashes after click Fans and Power

MateusBMP, 2022-12-14:
After clicking to open Fans and Power, AlienFX Control closes.

Steps to reproduce the behavior:
1. Open AlienFX Control
2. Click Fans and Power

**System:**
 - Model: Alienware M15 R6
 - AlienFX Control version: 7.9.1.0
 - OS: Windows 11 stable

**Note:**
I think it's not a version issue as I have the same problem with the previous version. But, how to debug or get more information about the problem? The software have a log file or something like that?

T-Troll, 2022-12-16:
Is it a first start, or you have some data?
Can you please run `alienfx-config backup` and share `afx-fans.reg` it creates?

PS: Just for sure. If you are updating to 7.9.1 from quite old version - check release notes, there are some dragons!

T-Troll, 2022-12-16:
Update: I confirm issue with empty configuration. Wait for test version.

T-Troll, 2022-12-16:
Fixed in [7.9.2.1](https://github.com/T-Troll/alienfx-tools/releases/tag/7.9.2.1).

Please check and close the issue.

MateusBMP, 2022-12-17:
Checked. It is working correctly. Thanks!

MateusBMP, 2023-01-13:
This issues returned in last release, 7.10.4

MateusBMP, 2023-01-13:
I tried to go back to the last installed version, [7.9.2.1](https://github.com/T-Troll/alienfx-tools/releases/tag/7.9.2.1), but the issue persists.

T-Troll, 2023-01-16:
Thanks.
Seems like another configuration issue, please backup config using `alienfx-config backup` command and share `afx-fans.reg` here.

T-Troll, 2023-01-16:
Upd: I check latest version for clean configuration, no issue detected. So i guess something wrong with you config. BTW, i don't change anything about it into 7.10.4, but do it before.

MateusBMP, 2023-01-16:
I'll wipe the entire installation and test again. I think it's a bug in my installation anyway...

MateusBMP, 2023-01-16:
> Thanks. Seems like another configuration issue, please backup config using `alienfx-config backup` command and share `afx-fans.reg` here.

```code
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Alienfan]
"StartAtBoot"=dword:00000000
"StartMinimized"=dword:00000000
"LastPowerStage"=dword:00000002
"UpdateCheck"=dword:00000001
"LastSensor"=dword:00000000
"LastFan"=dword:00000000
"LastGPU"=dword:00000004
"ObCheck"=dword:00000001
"DisableAWCC"=dword:00000000
"KeyboardShortcut"=dword:00000001
"Boost-0"=hex:00,36,13
"Boost-1"=hex:00,7f,13

[HKEY_CURRENT_USER\Software\Alienfan\Powers]
"Power-0"="Manual"
"Power-160"="Balanced"
"Power-161"="Performance"
"Power-162"="Battery Saver"
"Power-163"="Quiet"
"Power-164"="Full Speed"

[HKEY_CURRENT_USER\Software\Alienfan\Sensors]
"SensorName-0-8"="ESIF sensor #7 - CPU"
"SensorName-0-9"="ESIF sensor #8 - CPU"
```

T-Troll, 2023-01-16:
> "Boost-0"=hex:00,36,13
> "Boost-1"=hex:00,7f,13

Here it is. I need to check how did it happen. 
To fix, replace first byte (0) to 0x64 (100).
I'll fix this into next release.

T-Troll, 2023-01-16:
This is the fix (you can keep your data) - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/10429830/alienfx-gui.zip)

But, being honest, i still didn't understand how did you get 0 at max. boost.

MateusBMP, 2023-01-17:
I also didn't understand how I did it, but the **alienfx-gui.zip** solved it. I also run `alienfx-config backup` and now has saved as `"Boost-0"=hex:64,36,13` and `"Boost-1"=hex:64,7f,13`.
Everything is working correctly now. Thanks!

closed 2023-01-17 by MateusBMP

T-Troll, 2023-01-17:
JFYI: This values is max. boost. First byte is boost value, next two is max. RPM at this boost. So it can not be below 100, and in case it 0 - some calculation makes an exception. Later i found how it is possible, it's not your fault.

Thanks for help!