T-Troll/alienfx-tools #265: Inaccurate GPU temperature

CharlieIXXXXXXXXX, 2022-12-15:
It seems the GPU temp detected in alienFX is always lower than that displayed in taskmgr and GPU-Z.
I can only guess that alienFX can not read the data sent from Nvidia driver but use a different BIOS sensor.
[image]
Is there any possibility that alienFX can read the temp **more accurately** ?

T-Troll, 2022-12-16:
This is a known bug for some systems. Dell take incorrect sensor data into BIOS, and they not fix it for months.
As a fix, use one of ESIF sensors (seems like ESIF 3 in your case) instead - they are correct.

CharlieIXXXXXXXXX, 2022-12-16:
After using for a while I found none of the ESIF sensors actually stands for my GPU temp. 
In one situation temp from ESIF sensor 1 might be more close, in another situation ESIF sensor 2 gave me more "accurate" temp. Fun in fact, none of them were even close to the temp shown in taskmgr in some cases.
[image]
I went to Dell.com and found a newer version, which is 1.21.0. 
I will update it and see if they had fixed it.

CharlieIXXXXXXXXX, 2022-12-16:
Now my BIOS version is 1.21.0.
They had not fixed it **at all**.
None of the ESIF sensor value match the taskmgr GPU temperature.
I added a curve in each of the 3 ESIF sensors.
I can't think of a better idea to solve this... Or maybe you could offer some suggestions? I'm all ears.

T-Troll, 2022-12-17:
Well... My tools support [Libra Hardware Monitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor) readings (but start it before tools).
So you can use sensors from it as well (they use different method to retrieve it). This also help to configure out which on from ESIF is really GPU.

Don't rely on task manager, it's quite inaccurate (approximate).

williamcastanho, 2022-12-29:
To me are happening the same problem in Dell G15 5515, but what I realize, in my case, it's the software is using the temps of Radeon and do not get RTX temps. There's a way to change, is it?

T-Troll, 2022-12-29:
I got an answer from Dell.
This is... Optimus. They always take dGPU temp, and then chip is not active... you will see random value.
Can you please check (start game and look at temperatures).

closed 2023-01-24 by T-Troll