T-Troll/alienfx-tools #271: The lights won't light up with the screen

CharlieIXXXXXXXXX, 2023-01-02:
**Describe the bug**
When opening up the laptop lid, the lights won't automatically light up and I need to press Fn+AlienFX or press the Refresh button in AlienFX Control tray menu.
This happens no matter "Lights follow screen state" is on or off.

**Expected behavior**
Maybe the app can automatically Refresh the lights when it detected the screen is on... Just like how it detects when the screen is off.

**System (please complete the following information):**
 - Model: Dell G5 5500
 - Model details: four zone rgb keyboard with a 12 LED light bar
 - OS: Windows 11 Pro stable

**Additional context**
I set the four zone of the keyboard as "indicator". In this way I can dim the light bar only.
The effect is "Monitoring", the keyboard color changes with fan RPM, and light bar color changes with memory.
When the lights are off when the screen is on and memory usage or fan RPM is changed, the lights will light up.
So I reckon that "an automatic Refresh when the screen is on" is needed.

CharlieIXXXXXXXXX, 2023-01-02:
By the way, the "Lights and Grids" of Dell G5 5500 you have attached to the app seems not right.
Here is a copy of my own. 
[Current.csv](https://github.com/T-Troll/alienfx-tools/files/10330800/Current.csv)
It should be like this:
[image]
Or else something may went wrong when the app was saving the .csv file. Your version seems corrupted,

T-Troll, 2023-01-02:
> When opening up the laptop lid, the lights won't automatically light up and I need to press Fn+AlienFX or press the Refresh button in AlienFX Control tray menu.

This is definitely not right. So i'll check how it can happen. Most interesting, i use this function too (and even have the same chip), and it works correct for me. The only difference, i have Windows 10.

> By the way, the "Lights and Grids" of Dell G5 5500 you have attached to the app seems not right.

I copied it from you shared me before. Ok, i'll update, thank you!

T-Troll, 2023-01-03:
It was two bugs at once:
- Colors not refreshed at start if effect disabled
- Colors not saved as default at sleep/hibernate

Here are the fix beta - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/10333859/alienfx-gui.zip), please check.

CharlieIXXXXXXXXX, 2023-01-03:
I regret to say it was of little help. The problem still exists no matter the effect is monitoring or off.
Mind you that my laptop was set to "do nothing" when close the lid. I have to keep a few programs running in background thus I can't let it sleep even in S0.

P.S.: I'm not a native English speaker, so plz don't mind if anything was messed up due to my poor English.
If you get confused, just ask.  : )

T-Troll, 2023-01-03:
> Mind you that my laptop was set to "do nothing" when close the lid

This is IMPORTANT information. Please share uncommon settings related to issue to save my time!

> P.S.: I'm not a native English speaker, so plz don't mind if anything was messed up due to my poor English.

Neither me.

Ok... I Can't reproduce this issue at mine hardware and OS, BTW - screen stay on then i close the lid with "Do nothing".
But i add lid state monitoring, and now it works correctly for me even if "Do nothing" selected (but it can provide issue for systems with 2+ monitors!). Check, please - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/10334165/alienfx-gui.zip)

CharlieIXXXXXXXXX, 2023-01-03:
> This is IMPORTANT information. Please share uncommon settings related to issue to save my time!
My bad. Hopefully my issue wouldn't take up your time too much.
However you might have sent me the wrong file, this version does not do basic work like assigning colors for the lights.
When opening this version, my light bar just stops working at once and the keyboard RGBs behave abnormally.
Enclosed with two very short videos, which may help you understand my situation.

https://user-images.githubusercontent.com/78650751/210299181-137b51a9-044c-44e7-a14e-1abc2f77609e.mp4

https://user-images.githubusercontent.com/78650751/210299185-42d29bdd-fef3-46f3-8df4-ab36bd915334.mp4

CharlieIXXXXXXXXX, 2023-01-03:
And this shows what happened with the previous version.

https://user-images.githubusercontent.com/78650751/210299616-e3852abf-f221-47a1-81a8-10d6025d1608.mp4

My screen and lights would automatically go out and back when the app is quited.

https://user-images.githubusercontent.com/78650751/210299861-ede99cd9-378a-438e-87fb-3c402d4fa5a3.mp4

T-Troll, 2023-01-03:
This can happen - it's a beta, i change some other things too.

But turn off SOME lights from the SAME device is a mystery, a bit...

Try this one, [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/10334318/alienfx-gui.zip)

BTW, check initial bug as well, it should be fixed.

CharlieIXXXXXXXXX, 2023-01-03:
I see the improvement. Now the Keyboard light would light up automatically. But the light bar will not.
As I've mentioned at the very beginning, I set the four Keyboard lights as **indicator**, in this way i can dim the light bar alone.
Maybe it has something to do with this.

CharlieIXXXXXXXXX, 2023-01-03:
BTW, in very rare circumstaces that the light bar will stop working itself. I did many checks, including checking the cables inside the machine -- and I am quite sure it was a software problem. Restarting the app won't work, but rebooting the machine would.
I will also keep a close watch on this, and report it when I think I've got more information about what will cause that to happen.

CharlieIXXXXXXXXX, 2023-01-03:
No, it did not work. I thought it seemed to work only because the fan RPM and CPU temperature was changing when I open the lid.
When the machine is in a more stable state ( several mins after booting ), it would not light up as I open the lid.

T-Troll, 2023-01-03:
> I set the four Keyboard lights as indicator

Enough. Please run `alienfx-config backup` and share `afx-lights.reg` and `afx-gui.reg`. I'm a bit tired to guess what else you forget to tell me.

> I thought it seemed to work only because the fan RPM and CPU temperature was changing when I open the lid.

This mean you configuration doing this. Default color is black or undefined (2nd, i guess, looking on test).

CharlieIXXXXXXXXX, 2023-01-04:
It's kind of you to help me solve the issue, Here they are.
[afx-gui.txt](https://github.com/T-Troll/alienfx-tools/files/10341278/afx-gui.txt)
[afx-lights.txt](https://github.com/T-Troll/alienfx-tools/files/10341279/afx-lights.txt)
However during the backup process i received something like 3 "Operation successful" followed by an "Error: System cannot find the defined registry value."
[image]

T-Troll, 2023-01-04:
Thanks. I check your settings, and i should say it works correct at mine hardware (which have the same chip).
But i need to mention some (at Settings tab):
- You have "Lights follow screen state" off. This means lights always on, and not off with screen or lid.
- You have "Keep power/indicator on" on. This means the same for power/indicator lights.
- You have "Dim power/indicator" off. This means it will ignore dimming (but i think it's what you want).
- Also, try to enable "Light names on grid". It helps to understand which light is it.

So.... Enable "Lights follow screen state" and disable "Keep power/indicator on". All should work correctly.

> I've got 3 files, not 4. But that seems irrelavant, right?

It's ok, seems like you never use `aliefx-mon`, so didn't have settings for it.

PS: South or North China?

closed 2023-01-24 by T-Troll