T-Troll/alienfx-tools #272: cli for linux?

eatdust, 2023-01-02:
Hi there,
landed here from alienfx github issue page. I am wondering how hard would it be to compile alienfx-tools, or a subset of it, under linux? You have solved the hid interfacing, may be a simple cli tool working under linux would render all folks@alienfx happy, me included!
Cheers,
Chris.

T-Troll, 2023-01-03:
Well... I only use Linux in VM or cross-translation for ARM, so i can't do it myself.
But you can port CLI/SDK yourself, if you wish.

What you need to do:
- I use Windows types (like `WORD`, `DWORD`). It needs to be mapped to Linux ones.
- Modify detection functions (`AlienFXCheckDevice`, `AlienFXInitialize`) to scan HID Linux way.
- Modify `PrepareAndSend` to use Linux USB IOCTRLs, not Windows.

After this step, basic functions like "set one light"/"set one light effect" will work.
But for mass-updates, you also need to convert `Mappings` class - it's using registry now, and needs to be altered to `/etc` config file - mass-updates require light ID's list.

closed 2023-01-24 by T-Troll

eatdust, 2023-01-24:
Thanks for the details. I'll certainly try to include this within openrgb, it detects some subsets of the hid associated with the alienware keyboards.

T-Troll, 2023-01-24:
Feel free to use my SDK - check how i detect various Alienware chips (BTW, it's not only Alienware) and interfaces will be the same (but keep in mind - some controlled by report, some by feature, some by direct data!).
AW keyboards can have both Alienware and Chicony IDs (or even Darfon one for internal notebook RGB keyboards), it's listed into alienfx-controls.h