T-Troll/alienfx-tools #285: One of my fans sounds like a sign wave

meemkoo, 2023-02-08:
**Description**
I was configuring my fan curves, and as I get the fans to get louder, at least one of them starts to sound as if its speed is going down and then up. The change stays at a consistent interval, like a sign wave.

**Expected behavior**
I guess i expected it to not sound like the little math goblins put sign waves in my fans.

**System**
 - Model: AlienWare G5000 Desktop.
 - Model details: No keyboard with lighting, just a front panel light
 - OS: Windows 10 Stable

**Additional context**
[image]

[image]
The only sensor with check boxes for both of the fans is the one shown in the 2nd screenshot.

If necessary I could provide a YouTube video with the sound.

meemkoo, 2023-02-08:
Wow I just got the bright idea to restart my computer and revert to defaults. That fixed it.

closed 2023-02-08 by meemkoo

T-Troll, 2023-02-08:
You get bouncing - then one of the temperatures running around switch point and modify boost all the time.

PS: JFYI - boost is OVER BIOS RPM, not real RPM. So set it for high temp only, or fans will be always on high run and never stop.

meemkoo, 2023-02-09:
@T-Troll Thank you!