T-Troll/alienfx-tools #290: Lost Mouse

JazzyJazzjazzy, 2023-02-15:
Hi, installed version 8.1.0.1, restarted and no dewces were available in the DETECT DEVICE window. It was blank.

Restarted again same result.

Restarted again BSOD.

Restarted again to DESKTOP and had NO mouse.

Deleted Start Up task from Task Scheduler, restarted.

Still no mouse. Thoughts?

JazzyJazzjazzy, 2023-02-15:
Okay, cleaned up and got DEVICEs DETECTed. Selected Alienware 51M... and now see keyboard. DIM switch still not  working.

T-Troll, 2023-02-16:
It was a bug with light off (lights stop changed at all after off/on), but not with dimming.
2nd one is interesting, like version mismatch. But i check version is ok.

T-Troll, 2023-02-16:
Check latest update.

JazzyJazzjazzy, 2023-02-16:
Downloaded latest  update and no change noted. I have selected all of the keys, 107, and nothing happens when I change color or effects for the keyboard. Tron changes are being applied.

T-Troll, 2023-02-16:
Seems like i broke V5 support. Roll back to 8.0.2 or help me to diagnose.
(you will need USBPcap for this).

T-Troll, 2023-02-16:
Hmm... Maybe i found the issue.
Try this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/10752592/alienfx-gui.zip)

JazzyJazzjazzy, 2023-02-16:
Hello sir, I am happy to try this version. I will be home in an hour and will post back. Thank,.

JazzyJazzjazzy, 2023-02-16:
Works GREAT now! Your uploaded alienfx-gui.zip fixed it. Thank you.

T-Troll, 2023-02-16:
> DIM works in SETTINGS Window properly but there is no response when right clicking on the taskbar icon and selecting/deselecting DIM. The state will change in the SETTINGS window, check and uncheck, but the light's state does not change.

I'll check. But also check dim settings at "Profiles" tab.

> Should I check option, USE BIOS SENSORS? Does ENABLE FAN CONTROL also need to be checked?

It's different. BIOS sensors a wider (including power), and does not require fan access. Fan sensors have some in common with BIOS, but different as well. The first setting is for Events effect mode only.

So you can close this issue, if you wish, but i need your help. You have V5 keyboard, and i want to optimize it a bit. Can you help me?

JazzyJazzjazzy, 2023-02-17:
Just noted that ALIEN HEAD, RING TOP & BOTTOM not working.

T-Troll, 2023-02-17:
Calm down!

It's another device with APIv4 (that's why you have 2 of them).
Device is fully functional in latest version (i have it as well), so it can be 2 reasons of malfunction:
1. Incorrect light IDs - it will happen if you are use profile from different system
2. Chip hang (it can happen with v4 chip in case you change power button color frequently).

So first, check "Devices" tab - device should present, and selecting light from it should turn it green.
In case device not present - check other light IDs, if any one of them works - you have incorrect mappings, do it manual way.
If device not present or all lights fail - do a chip reset:
- Shutdown (NOT sleep) system
- Remove all cables, including power cord
- Wait 15 seconds
- Connect back and start
You will see rainbow lights in BIOS in case reset successful.

JazzyJazzjazzy, 2023-02-17:
"Calm down!" No need, my friend since I am not excited, lol.

Your suggestion regards removing all cables and rebooting solved it. Working great now, thanks.

What did you need done?

T-Troll, 2023-02-17:
> Your suggestion regards removing all cables and rebooting solved it. Working great now, thanks.

It's not good, in fact. This mean tools hang the chip. I do some counter-measures into code.

> What did you need done?

I'm into preparation of some test CLI versions for you, will deploy it here tomorrow. I want you to run some commands into each of them and tell me what happened.

JazzyJazzjazzy, 2023-02-17:
Happy to do so, my friend. Will keep my eye on your site. Thanks.

T-Troll, 2023-02-18:
Here are a bunch of test apps - [alienfx-cli_v5_test.zip](https://github.com/T-Troll/alienfx-tools/files/10772786/alienfx-cli_v5_test.zip)

Can you, please:
1. Unpack it somewhere, open CMD to this folder, then:
2. Run this command - `alienfx-cli_nostatus status`
3. Stop `AlienFX-GUI`.
4. Check output - look for string "Device #N, ......., Keyboard, ....". Remember the number N.
5. Run command `alienfx-cli_nostatus setdim=N,0`. Are keyboard lights turn off?
6. Run command `alienfx-cli_nostatus setdim=N,255`. Are lights back on?
7. Repeat 4-5 for all files into archive (`_nofirst`, `_nosecond1).
8. Share 4-5 results (yes or no) with me.

JazzyJazzjazzy, 2023-02-18:
Hello, I ran it once. Kindly clarify what you mean when you say run it on each file in the archive please. Thank you.
[Test 1.txt](https://github.com/T-Troll/alienfx-tools/files/10772965/Test.1.txt)

T-Troll, 2023-02-18:
Ohh... I mess a bit (then edit list).
1. Replace "N" in commands with keyboard device number (0 in your case).
2. Instead of repeat 4-5, you need to repeat 5-6. It's 3 files in archive, you need to check it with ALL of them.

T-Troll, 2023-02-18:
Thanks! I only need for device 0, and don't need logs, but ok anyway.

So... Simplest protocol works, no more heavy init needed. I'll do this for next version, drop you a beta here.

JazzyJazzjazzy, 2023-02-18:
Lol, we aim to please...

T-Troll, 2023-02-20:
I publish new release, all should be fixed. Check dimming, please.

JazzyJazzjazzy, 2023-02-20:
DIM works great! Thank you.

closed 2023-03-07 by T-Troll