T-Troll/alienfx-tools #295: Thoughts/Notes on AlienFx

JazzyJazzjazzy, 2023-02-20:
Thank you for the new release, sir. 

Version 8.1.1.0 uses 2 megabytes more ram than previous version (3.x vs 1.x megabytes).

Would be good to use SHIFT, CNTL A, or click AND drag to select ALL the keys at once in SELECT LIGHTS FROM LIST (ZONES).

COLORS Window RESIZE Sliders are inoperative/greyed out. On GRIDS Page, sliders work fine. Is this normal procedure?

COLORS Window, beneath Action Settings, The MODE is set COLOR and greyed out. Are there other OPTIONS here to select or only COLORS?

My TRON items still do not respond to EFFECTS. The proper colors are displayed on the lights but items remain only one of the colors selected, BACKGROUND. Note, when I select RANDOM on the LIGHTS Page, My TRON items DO change color. 

When I DETECT DEVICES, TOUCHPAD is included but my touchpad does not have lights. I cannot delete this entry. Do I need to create custom light grid? 

DIM LIGHTS option works fine. No problem here. To this end, have you considered separating your versions? Is it reasonable to discontinue supporting older models in each new release and to have two separate versions? 

I refer to what AWCC does. One version supports up to a certain level and another that picks up from where that version leaves off? 

Pop up TOOL TIPS might be helpful in the SETTINGS Window for some of the items.

Thank you for your effort and your time.

T-Troll, 2023-02-20:
> Version 8.1.1.0 uses 2 megabytes more ram than previous version (3.x vs 1.x megabytes).

It's OK. Price for stability.

> Would be good to use SHIFT, CNTL A, or click AND drag to select ALL the keys at once in SELECT LIGHTS FROM LIST (ZONES).

I check what i can do. There are 2 modes of list, now i use first one.

> COLORS Window RESIZE Sliders are inoperative/greyed out. On GRIDS Page, sliders work fine. Is this normal procedure?

It's OK. Grid size can only be changed at "Devices&Grids" tab to avoid mess with zones.

> COLORS Window, beneath Action Settings, The MODE is set COLOR and greyed out. Are there other OPTIONS here to select or only COLORS?

This is OK in case no action present/selected.

> My TRON items still do not respond to EFFECTS. The proper colors are displayed on the lights but items remain only one of the colors selected, BACKGROUND. Note, when I select RANDOM on the LIGHTS Page, My TRON items DO change color.

Increase WIDTH - i mention it into release notes/readme before. WIDTH define number of phases for morph.

> When I DETECT DEVICES, TOUCHPAD is included but my touchpad does not have lights. I cannot delete this entry. Do I need to create custom light grid?

You are got mappings from incorrect device. Just select it from light list there and press "Reset" button.

> To this end, have you considered separating your versions? Is it reasonable to discontinue supporting older models in each new release and to have two separate versions?

No. One of the ideas behind these tools was i support ANY hardware in one app. 
V1/V2 difference is fan access method only (some systems feels better with old V1 - it uses some hack tricks), but both support the same hardware.

> Pop up TOOL TIPS might be helpful in the SETTINGS Window for some of the items.

A lot of effort and app size grow for small profit. Press "Help" in top menu - it opens manual with all options explained.

JazzyJazzjazzy, 2023-02-20:
RAM Usage is back to 1.x ram.

FINALLY got the lights to work properly, duh... I added  new group in LIGHTS and moved the TRON keys to it. Works great now. TRON Lights are shown in Devices and Grids but did not transfer over to LIGHTS like the LIGHTS did. Thank you.

"Increase WIDTH - i mention it into release notes/readme before. WIDTH define number of phases for morph.
BTW, for Tron lights, effects defined at the "Color" tab using actions."
[image]

I did this and the keyboard had no other entries besides the keyboard. No Tron, etc. Also re-detected DEVICES and saw no change in keyboard options. 

Unchecking POWER/INDICATOR Lights in SETTINGS does not extinguish the Power Light.

Just comments, my friend. I am very happy with current results.

T-Troll, 2023-03-07:
> Question: Can I sync the TRON items and the KEYBOARD to same effects or by using a single control?

You can create one zone for all of them and set it.

> Unchecking POWER/INDICATOR Lights in SETTINGS does not extinguish the Power Light.

Yes - checking it at this tab, just report about type. It handled other way - then off/dim lights. Power/indicator can not set as off or dim with others. Power button is also tricky - it should have at least 2 colors (AC/BATT), so Color tab can not allow you to set only one.

> AlienFX Configuration pop out window hangs when saving configurations.

It's a simple .cmd calling regedit. So it's about your system settings.

closed 2023-06-16 by T-Troll