T-Troll/alienfx-tools #297: AlienFx Control not starting

jasonhpchu, 2023-02-24:
Using 8.1.2

As soon as I start AlienFx Control, I get two UAC prompts,  one to install some stuff, and another to actually run this, then nothing.
No window starts.
I can see the AlienFX Control process running in task manager, but no gui.

AlienFX Monitor is fine.
AlienFan Control is fine.

Having trouble the the AWCC, hoping this can replace that junk.
(Do I need to remove AWCC first before using this?)

TIA

T-Troll, 2023-02-24:
Hmm... Tools does not install anything, so you should have exactly 1 prompt. Let me check what happened.
Meanwhile, start it as Administartor - this should work.

T-Troll, 2023-02-24:
Unfortunately, i can't reproduce it (and, in fact, only 1 UAC prompt available in code at start).
Can you, please, share what EXACTLY it requests to install?
UI can be consequence, both accept and deny UAC provide it (but some functions will be disabled in case you deny).

jasonhpchu, 2023-02-24:
alright, here's exactly what happens:
First there's this:
[image]
Then I quickly see this happening:
[image]
Then finally I get this:
[image]

T-Troll, 2023-02-25:
Ohhh... We meet it before.
The issue is.... application shortcut. Windows 11 do not use common link file, but do it trough fake installer you see.
Let me check how do we solve it before... or create new direct shortcut.

NB: If you have Intel system, FIRST application start can be delayed up to 5 seconds.

jasonhpchu, 2023-02-25:
I see, ok, maybe I'll use the zip instead of the main then.  I'll report back.

T-Troll, 2023-02-25:
Doesn't need to use zipped - just make new shortcut from installed version folder.

jasonhpchu, 2023-02-25:
Alright, I used the zip, I feel it's cleaner than the msi.
Copied those files for the DPTF.
Run as admin.
Nothing happens.
AlienFX Control just sits there in the Task Manager, no GUI.

Guess I should mention, I have an Aurora R13, windows 11.
Running BIOS 1.10.0 (currently their latest one, installed it trying to debug the AWCC not launching).

Just let me know what you need me to capture to help debug.

T-Troll, 2023-02-25:
I check it on R13 before... Hmm... BIOS update?
So it hangs on detection, most probably into fan/power.
To check, can you please open Administrator CMD at folder you install/unpack it and run:
`Alienfan-cli rpm`. Then share the output to me.
If it stuck, stop it by CTRL+C and tell me - i'll share test version.

jasonhpchu, 2023-02-25:
Alienfan-cli rpm
AlienFan-CLI v8.1.2
Supported hardware (4400) detected, 4 fans, 14 sensors, 4 power states.
Fan CPU 0 RPM: 794
Fan 1 RPM: 1497
Fan 2 RPM: 933
Fan CPU 3 RPM: 903

alienfx-cli probe
stuck, had to ctrl-c

T-Troll, 2023-02-25:
Wow! Light hangs on device enumeration!
Thanks, i'll fix it ASAP.

T-Troll, 2023-02-25:
Here we are. Please unpack this debug alienfx-cli and run as before - `alinfx-cli probe` - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/10830504/alienfx-cli.zip)

It has a lot of debug information, but most interesting is "Testing device #..., ID=......" - it should loop on it.
I'm interesting what IDs it will have then looped (it should be the same one for many devices).

jasonhpchu, 2023-02-25:
alienfx-cli v8.1.2
Testing device #0, ID=2341708408400
Testing device #1, ID=2341708408832
Testing device #2, ID=2341708408064
Testing device #3, ID=2341708407680
Testing device #4, ID=2341708409024
Testing device #5, ID=2341708407776
Testing device #6, ID=2341708407296
Testing device #7, ID=2341708408448
Testing device #8, ID=2341708408160
^C
It stops after that #8

T-Troll, 2023-02-25:
Oh! So it stops after 8 and not continue? Thanks!

T-Troll, 2023-02-25:
Here is another debug CLI version with attention to probing - [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/10832482/alienfx-cli.zip)
Can you please run it the same way?

T-Troll, 2023-02-26:
Wow! Quite interesting. Thanks, i'll add some more checks.

T-Troll, 2023-02-26:
Check this one, please. This is the only thing remain... [alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/10832746/alienfx-cli.zip)

jasonhpchu, 2023-02-26:
Just a thought, would this be related to why AWCC doesn't launch properly?
Maybe somehow the BIOS update or something made these probes not communicate properly with the software causing it to hang/crash?

Anyways, here you go: 
alienfx-cli v8.1.2
Probing ID#1988781504352
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_5426&PID_1297
Probe CAPS OK
Probe done, type -1
Probing ID#1988781505504
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50489
Probe CAPS OK
Probe done, type -1
Probing ID#1988781505984
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_5426&PID_1297
Probe CAPS OK
Probe done, type -1
Probing ID#1988781505552
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_4152&PID_4781
Probe CAPS OK
Probe done, type -1
Probing ID#1988781505840
Detail data probe OK
Detail data OK
Probing ID#1988781504880
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#1988781505072
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#1988781505648
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#1988781506080
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_6268&PID_4105
Probe CAPS OK
^C

T-Troll, 2023-02-26:
Hmm... Do you have Alienware monitor?

jasonhpchu, 2023-02-26:
Yes, Alienware AW2721D monitor, and Alienware Aurora R13 desktop.
The monitor has 4 lights? (Looking at the monitor menu). The power button, the back alien symbol, a bottom light, and a stand light (which I don't have the stand, as it's on a mount).
The desktop should just be the power button and the ring around the power button.  I don't have the model with the inside light.

T-Troll, 2023-02-26:
So... Monitor is the source of the issue (i see it by PID/VID).
It's the only device need to be initialized immediately (so i send init command to it)... But command send to device hangs.
You can try to disconnect it's USB and run - all should work correctly.

Ok, let me check how to avoid it. 
It can be monitor settings (some of them can disable light control from OSD) or just unsupported model (i didn't test 27-series, 34-series only).

T-Troll, 2023-02-26:
This is cli version without monitor init - check it, please. Should not hang now. But it's interesting will monitor work or not.
[alienfx-cli.zip](https://github.com/T-Troll/alienfx-tools/files/10832823/alienfx-cli.zip)

jasonhpchu, 2023-02-26:
Now that you mention it's the monitor, when I was searching about my AWCC issue, I did see someone reporting issue with another Alienware monitor.  Maybe related.

Both alienfx-cli status and alienfx-cli probe returned the same thing.
Here's the result:
alienfx-cli v8.1.2
Probing ID#2526594215088
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_5426&PID_1297
Probe CAPS OK
Probe done, type -1
Probing ID#2526594214128
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50489
Probe CAPS OK
Probe done, type -1
Probing ID#2526594215472
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_5426&PID_1297
Probe CAPS OK
Probe done, type -1
Probing ID#2526594214560
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_4152&PID_4781
Probe CAPS OK
Probe done, type -1
Probing ID#2526594215568
Detail data probe OK
Detail data OK
Probing ID#2526594214992
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#2526594215184
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#2526594215616
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#2526594215232
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_6268&PID_4105
Probe CAPS OK
Probe done, type 6
Scan #8: VID: 6268, PID: 4105, Version: 6
Probing ID#2526594214464
Detail data probe OK
Detail data OK
Probing ID#2526594215712
Detail data probe OK
Detail data OK
Probing ID#2526594215808
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50489
Probe CAPS OK
Probe done, type -1
Probing ID#2526594213888
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_6268&PID_1360
Probe CAPS OK
Probe done, type 4
Scan #12: VID: 6268, PID: 1360, Version: 4
Probing ID#2526594214752
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50489
Probe CAPS OK
Probe done, type -1
Probing ID#2526594214368
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50489
Probe CAPS OK
Probe done, type -1
Probing ID#2526594215136
Detail data probe OK
Detail data OK
Probing ID#2526594215280
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50489
Probe CAPS OK
Probe done, type -1
Probing ID#2526594213936
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#2526594214800
Detail data probe OK
Detail data OK
Probing ID#2526594214944
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_4152&PID_4781
Probe CAPS OK
Probe done, type -1
Probing ID#2526594248464
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#2526594247936
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50475
Probe CAPS OK
Probe done, type -1
Probing ID#2526594246928
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_1133&PID_50489
Probe CAPS OK
Probe done, type -1
Probing ID#2526594247600
Detail data probe OK
Detail data OK
Handle OK
Attributes OK, VID_4152&PID_4781
Probe CAPS OK
Probe done, type -1
State update: PID: 4105, brightness: 255, power: 0
State update: PID: 1360, brightness: 255, power: 0
^C

T-Troll, 2023-02-26:
Yes, works OK now. What about monitor lights changes? (described above)

T-Troll, 2023-02-26:
`status` should have lines like this at the end:
```
Dell API not found, 2 devices found.
Device #0 - AW410k/510k, Keyboard, VID#1266, PID#6504, APIv8, 108 lights
```
Anyway, this is beta GUI and CLI - it should work correctly at your  system now - [Tools.zip](https://github.com/T-Troll/alienfx-tools/files/10832923/Tools.zip)

I also remove debug print from CLI.

jasonhpchu, 2023-02-26:
ok, I think something's still hanging it.
so earlier, when I ran the status, it didn't give me those device details, it hangs there.
I was tweaking around trying to get rid of AWCC (it gotten into a stupid state, the uninstall does a repair, and it hangs there), but guess it triggered something, and the CLI finally spit out the status:
Dell API not found, 2 devices found.
Device #0 - , Display, VID#6268, PID#4105, APIv6, 0 lights
Device #1 - , Desktop/Notebook, VID#6268, PID#1360, APIv4, 0 lights

So I tried to do setone, it hanged, and had to ctrl-c.
Tried to do status again, hangs no more device details.

So I restarted, status works again.
Tried your new CLI and GUI, nothing shows again, task manager shows it's running at the back.
Tried the status again, hangs with no device details.

One weird observation...after running the GUI (even the release version), I notice my desktop fans spin up faster.
Not sure if it's the hanging process making things heat up, or if it somehow changed my fan speed profile.

T-Troll, 2023-02-26:
Seems like you have hardware issue with your monitor - it freezes system on every command send to it!
So remove USB cable from it, other lights should work well. You can also try AWCC without this cable.

> I notice my desktop fans spin up faster.

It's because GUI turn on manual fan control.

jasonhpchu, 2023-02-26:
Bingo my friend.
I unplugged, and AWCC starts.
Great, now another problem to resolve.
Thanks.

T-Troll, 2023-02-26:
You are welcome.
BTW, try to change/reset monitor settings and replace the USB cable - this CAN help.
Anyway, you can switch to my tools as well - it can do more, then AWCC.

jasonhpchu, 2023-02-26:
You won't believe how stupid I feel the solution was.
I powered off the monitor, powered it back on, and now AWCC and your utility works.
I've left it on 24/7, and letting it go to sleep when my computer goes to sleep, because the USB hub on it powers off if I turn off the monitor, and I have my keyboard and mouse dongle plugged to the monitor.  If I power off the monitor, the keyboard and mouse won't be able to wake up the sleeping computer.

I'd guess there must be some minor glitch on the monitor's internal software, memory leak or something, so a power off/on resets it.

Thanks again for your help.

I will probably end up using your utility over AWCC, since all I really use AWCC for is the lights and fan speed.

closed 2023-02-26 by jasonhpchu

T-Troll, 2023-02-26:
Do you remember the old joke? "Hit with a hammer - $1. Know the place where to hit - $999".
We are spent 2 days to configure out where to hit.
So it's not stupid, and you learn a lot of new things into the process.

You can continue to use AWCC, it's much simple - my tools are for enthusiasts mostly.
But my tools have seriously less footprint (about 2M RAM and 1M on HDD only), better fan and power control (multiply sensors and curve points, switch by keyboard shortcut, etc), and light effects (in fact, monitor support was added with the help of the guys who wants ambient light from it).
Can have some bugs too, but, as you can see, i do my best to fix it as fast as possible.

Anyway, good luck! Interesting case, i'll add some checks for this situation into the next release!