T-Troll/alienfx-tools #301: Trying to Collect Data for my X17 R1 Laptop

TheSQLGuru, 2023-03-02:
Hey guy. Sorry for ghosting. Life has been . . . suboptimal.  :-( 
[image]
[image]
[image]
[image]

I tried walking through your Wiki page on how to provide you the information needed to produce an exact match for my computer lighting, but I have many issues. 

1) I have multiple keyboards, and a butt ton of HID devices. How am I supposed to know which one of them is my laptop's physical keyboard? See attached pic.

2) I have a trackpad device shown in two images. One has the Properties open, and there is no PID. I am pretty confident that you don't want DELL/0A8F because that is on multiple components. What's the story?

3) Even more odd is that the trackpad is not "under" a USB anything when viewed by Connection. It shows up under the I2C HID Device, which subordinate to the Serial IO I2C Host Controller. And as you can see in the USBPcap pic, no I2C device shows up there to choose. 

So it seems I need some more hand-holding. It also seems that someone should put some pictures in the Wiki page for this process to help folks like me. :-)

T-Troll, 2023-03-02:
Well... I see all devices at #2 - 6/1 is internal devices, 9 is external keyboard.
Hint: look at "vendor-defined device".

Touchpad doesn't have a separate light device and served by common tron device with PID_0550 (or 551 for new gear).

TheSQLGuru, 2023-03-02:
I have it just about figured out now!! It's been AGES since I had WireShark open! :-D I am about to do a dimming and brightening (something you didn't ask for) of the track pad, and then cycle through effects for the track pad. Then I can upload the files for you.

T-Troll, 2023-03-02:
APIv4 it uses support per-light hardware brightness even in mine SDK. Just locate it's light ID.
Or you want to support other hardware?

TheSQLGuru, 2024-08-16:
Sorry for leaving this hanging. I don't have this laptop any longer, so you can close the ticket.

closed 2024-08-20 by T-Troll