T-Troll/alienfx-tools #303: Fan Curve Displays Only One Curve

JazzyJazzjazzy, 2023-03-05:
This is probably not an issue but a configuration question. I duplicated your curves displayed in your AlienFan instructions. Configured as depicted, I have only one curve displayed and not the second yellow curve. I duplicated the curves in the GPU Thermistor also. Do I need to create curves for all sensors or just the sensor that I am using? Why do I not see the ye
[image]
llow curve? Thank you.

T-Troll, 2023-03-05:
Yellow curves are from OTHER sensor - GPU in my case. Green from selected one.
So if you don't set GPU sensor, you will only see one.

I can share you .reg file with my full configuration, but keep in mind - i have SSD close to GPU, so i need to cool it down too.

JazzyJazzjazzy, 2023-03-05:
I set the GPU curve, you can see this in the photo. I also placed check mark in CPU Thermistor, unchecked GPU on right side in same window, set GPU Thermistor curve and unchecked CPU there. Still only one curve showing. 

Yes, please send .reg entry and i will review it. Thank yoou.

T-Troll, 2023-03-05:
[fans.zip](https://github.com/T-Troll/alienfx-tools/files/10891001/fans.zip)
Here it is. But be careful:
- I can have different SensorID, so some fan curves can not have real sensor (but will be visible anyway)
- This .reg have my sensor names, so remove from reg or backup yours.

JazzyJazzjazzy, 2023-03-05:
Okay, I downloaded your .reg file and changed the NAMES to match what I saw in  my registry. I deleted one entry because I did not have that sensor. (Deleted "SensorName-0-7"="CPU Participant (GFX)")

It appears that you only created curves for the two sensors, CPU THERMAL THERMISTOR & GPU INTERNAL THERMISTOR. You checked both CPU &GPU boxes on both.

With the exception of"SensorName-0-2"="SSD Thermistor", all other sensor's check boxes are unchecked.  I am guessing that you left this one checked because of your aforementioned SSD.

Question: 
1. If the highest value of each curve is the one that is honored when the program is running, then why do you need the  second curve on the above "0-2" sensor?

2.Why can I not check only the CPU check box on CPU THERMAL THERMISTOR 
and only the GPU check box on the GPU INTERNAL THERMISTOR?

3. Why do the curves change when I select CPU and then GPU in the right window?

A note, both curves appear on each graph respectively using your .reg file. This was not the case on mine. I include my original .reg, as a text file and your .reg file with the changed names for your reference. Thank you.

[My Sensors.txt](https://github.com/T-Troll/alienfx-tools/files/10892135/My.Sensors.txt)

[Your file with My Changed names.txt](https://github.com/T-Troll/alienfx-tools/files/10892134/Your.file.with.My.Changed.names.txt)

T-Troll, 2023-03-06:
1. Because my SSD near GPU and fried by it. So i need to cool it down, if hot, or it starts throttle.
2. Because you have ONE, JOINED, cooling system. So spin up GPU fan for CPU-only and CPU fan for GPU-only helps to cool it down.
3. Because curves in window represent active FAN, selected into right list. For ALL sensors, which control it.

Your control have 1 sensor for CPU fan, and 2 for GPU. So it's OK, one curve if CPU fan selected, 2 if GPU fan.

JazzyJazzjazzy, 2023-03-06:
Understood. Kindly CLOSE this topic if you wish, thank you.

T-Troll, 2023-03-06:
You give me an idea how to modify UI to make this clear for user.
So i keep it open until i do it.

JazzyJazzjazzy, 2023-03-06:
Sounds good sir. May I offer that the curves that are yellow have the corresponding sensor label displayed as a possible solution? Thank you

T-Troll, 2023-03-06:
Labels is not so easy to implement good way.
I plan to change UI the other way - checkboxes will be not on fan, but on sensors. So you will see selected sensors after select fan.

T-Troll, 2023-03-07:
Check latest release.

closed 2023-03-07 by T-Troll

JazzyJazzjazzy, 2023-03-07:
Hello sir,  I have the latest release. It is a much clearer interface and easier to work with it. It is easy to see all the individual sensors that are in effect for our curves. Thank you!

T-Troll, 2023-03-07:
Thanks. As i said, initially this was fan-driven, later i made it sensor-driven but don't change UI.
Anyway, thanks for your questions, this help to clarify.

JazzyJazzjazzy, 2023-03-07:
Sir, YOU Are The Man! Peace.