T-Troll/alienfx-tools #317: AlienFan persistently drops out of monitoring phase.

JazzyJazzjazzy, 2023-03-19:
AlienFan persistently drops out of monitoring phase and CPU and GPU overheat. RPM values remain at 500 and do not change,

This has been the case on initial startup and resume from sleep. Changing the profile from MANUAL to anything else and back to MANUAL usually resolves issue but it returns.

Also, noted that POWER MODEs are now numbers, ie "Level 163". Is this by design? Thank you.

T-Troll, 2023-03-19:
Thanks, it's serious.

You are not the first to report me about stuck fans, seems like latest BIOS make this shit.
Can you remind me your systemID (visible in fan cli)?

> Also, noted that POWER MODEs are now numbers, ie "Level 163". Is this by design? Thank you.

Ah, yes. Power mode name now have not the number, but real hardware power mode ID (internal change). It can be from A0 to AB hex, A0 balanced, usually, AB is G-Mode/Performance.

T-Troll, 2023-03-20:
Check latest release. Did it fix this issue?

JazzyJazzjazzy, 2023-03-20:
Latest release appears to have fixed this issue. Will post back if anything else noted, Thank you.

JazzyJazzjazzy, 2023-03-20:
Spoke too soon, issue still persists in latest HOTFIX,

T-Troll, 2023-03-20:
So it's random. Ok, tell me your SystemID, please...

JazzyJazzjazzy, 2023-03-21:
I remember now. I also remember that I had no Device ID. You thought that strange  at the time.

C:\Program Files\AlienFX Tools>alienfan-cli
AlienFan-CLI v8.2.5
Supported hardware (0) detected, 2 fans, 11 sensors, 6 power states.
Usage: alienfan-cli command[=value{,value}] {command...}

JazzyJazzjazzy, 2023-03-21:
C:\Program Files\AlienFX Tools>alienfan-cli getpower
AlienFan-CLI v8.2.5
Supported hardware (0) detected, 2 fans, 11 sensors, 6 power states.
Power mode: Manual (0)

C:\Program Files\AlienFX Tools>alienfan-cli setpower=1
AlienFan-CLI v8.2.5
Supported hardware (0) detected, 2 fans, 11 sensors, 6 power states.
Power mode set to Level 1 (163)

C:\Program Files\AlienFX Tools>alienfan-cli getpower
AlienFan-CLI v8.2.5
Supported hardware (0) detected, 2 fans, 11 sensors, 6 power states.
Power mode: Manual (0)

C:\Program Files\AlienFX Tools>alienfan-cli getpower
AlienFan-CLI v8.2.5
Supported hardware (0) detected, 2 fans, 11 sensors, 6 power states.
Power mode: Level 1 (163)

C:\Program Files\AlienFX Tools>alienfan-cli setpower=1
AlienFan-CLI v8.2.5
Supported hardware (0) detected, 2 fans, 11 sensors, 6 power states.
Power mode set to Level 1 (163)

JazzyJazzjazzy, 2023-03-21:
Last item is after resume from sleep.

T-Troll, 2023-03-21:
> Power mode: Level 1 (163)

If last 2 after resume, this means you are in Balanced mode, which is not reset to manual!
Interesting...

JazzyJazzjazzy, 2023-03-21:
just confirming that last two are after resume.

T-Troll, 2023-03-21:
Ok, Wonder how it happened.
Please remind me you model.

JazzyJazzjazzy, 2023-03-21:
> Ok, Wonder how it happened. Please remind me you model.

Alienware m17 R4

JazzyJazzjazzy, 2023-03-23:
Update: Sir, AlienFan is properly working at start up and resume from sleep. Thank you for your extra efforts. Peace.

T-Troll, 2023-03-23:
At least. Great.

closed 2023-03-23 by T-Troll