T-Troll/alienfx-tools #320: Ver. 8.2.5.1 not Displaying "Pop-Up" TEMPS & BOOST On Fan Graph

JazzyJazzjazzy, 2023-03-22:
Ver. 8.2.5.1 not Displaying "Pop-Up" TEMPS & BOOST On Graph when changing Fan Curves. 

This feature is helpful when setting and matching curves of the different sensors.

Do not know if you removed this from this build so will not say is "bug". Thank you.

T-Troll, 2023-03-22:
OMG... What a day for me... One guy blame me i broke keyboard (no, it's impossible). Even thinking about close this project.
No, it's displaying it at my gear. Both FX and Fan GUI. But i'll check what it can be.
BTW, what about fan stuck? Did it fixed?

T-Troll, 2023-03-22:
Aye, catch it! Will be fixed next update.

JazzyJazzjazzy, 2023-03-22:
Good day sir, I saw the comment regards the keyboard, lol. Sorry for your grief. PLEASE do not stop developing this project as it is the best thing since sliced bread. It can be frustrating and disheartening when folks attack your work in such a fashion, especially when you devote so much of your time and efforts for FREE. 

I wonder if he has tried to connect the keyboard to another machine or completely uninstalled all related software and rebooted? So many questions...

Anyways, I appreciate you and your works. Peace.

T-Troll, 2023-03-23:
Fixed in [8.2.5.2](https://github.com/T-Troll/alienfx-tools/releases/tag/8.2.5.2).

JazzyJazzjazzy, 2023-03-24:
Perfect! Don't change a Thing! Thank you.

closed 2023-03-24 by JazzyJazzjazzy