T-Troll/alienfx-tools #322: Aurora R7 won't show light control options.

timmytot666, 2023-03-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
Open program and the lights tab doesn't show up.

**Expected behavior**
A clear and concise description of what you expected to happen. Should show a light control tab and it doesn't.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**System (please complete the following information):**
 - Model: Alienware Aurora R7
 - Model details: Tower
 - OS: Windows 11 Stable Release

T-Troll, 2023-03-22:
Correct. 
R7 and R6 the only uncommon systems which have ACPI-controlled lights (all other hardware have USB-controlled).

Unfortunately, it's still under development for latest Fan SDK V2 (yes, lights controlled by fan subsystem at this gear), but you can use Fan V1 version i provide among other release files - it supported there (BTW, this fact mentioned into every release notes).
Please read readme notes into V1 archive first!

timmytot666, 2023-03-22:
I'm sorry.  I totally read the readme but I didn't understand which version I needed sorry!

T-Troll, 2023-03-22:
It's OK.

Also, i need to warn you - Antivirus will complain about V1 version badly (i use some hack methods to load unsigned driver). Check [WIKI page](https://github.com/T-Troll/alienfx-tools/wiki/Why-antivirus-complain-about-some-alienfx-tools-components%3F) why it happened.

You can help me to support it into newest SDK, if you have time.

TechnicalLuck13, 2023-03-31:
I have the same issue with the Alienware 15 R4. Should also go back?

T-Troll, 2023-03-31:
No, R4 have 2 USB devices, so the source of your issue is different. Is it visible in Device Manager? What `alienfx-cli probe` prints?

TechnicalLuck13, 2023-04-01:
the -cli.exe files have refused to function for me regardless of the version I've used.

TechnicalLuck13, 2023-04-01:
Ah, I joined the discord cuz I figured something like this you'd rather deal with there and saw your pinned message. If you aren't able to respond for a while I totally get it. I'll just keep troubleshooting on my side. I'm Luck13 on the discord if it's easier there.

T-Troll, 2023-04-02:
CLI = Command Line Interface. So first open CMD first (any for FX, Administrator for Fan) then run it from it - you will see the log.
But in case we are talking about `AlienFX-CLI` - this is bad news for you - most probably, you will see something like "Both high and low-level devices not detected, quit" - this mean no light chip detected among PnP hardware, so you need to check what you have into Device Manager.

TechnicalLuck13, 2023-04-02:
Have command prompt open as I try to run AlienFX-CLI ? Tried that even running both as administrator. I don't see "Both high and low-level devices not detected, quit". I don't see anything. It's as if nothing even tried to run. Nothing in the CMD either. Where am I looking in Device manager and what for. Although somehow I doubt it has to do with recognized devices because on 8.1.2 which I'd been running previously I could control my lighting. I even tried rolling back to 8.1.2 but... would you rather make a thread in the discord support server? I have a feeling this problem is gonna hide from us.

TechnicalLuck13, 2023-04-02:
It randomly fixed itself?! Uhm. Problem solved?

T-Troll, 2023-04-03:
Check about restart/reboot.
Yes, you can lose devices not because of my app, but because Windows update and hardware fault.
About device manager - you should have "Vendor defined device" among "Human interface devices" with VID_187C&PID_0550 - this is tron lights. Check among "Unknown devices" as well. I don't remember exact keyboard lights device IDs, but it should be one with VID_0d62.

PS: This is well-known bug about monitor can hang both AFX and AWCC in some cases. Also, it's a way to reset light system - see afx readme ("Known issues").

closed 2023-06-16 by T-Troll