T-Troll/alienfx-tools #325: Alienware 17 R1 (Windows 11): AlienFX Control crashing when trying to open it (Please set lights first, Faulting module name: KERNELBASE.dll)

mickael28, 2023-03-25:
**Describe the bug**
* AlienFX Control crashes when trying to open it after 1st installation.

**To Reproduce**
* I just downloaded the latest - https://github.com/T-Troll/alienfx-tools/releases/tag/8.2.6
* Tried to launch `AlienFX Control` on Windows 11 (Alienware 17 R1)
* I get a message on left corner saying:
```
Please set lights first
```
* Then the GUI crashes
* It has not launched a single time, always getting the error.

**Expected behavior**
* The GUI to stay on the screen to be able to configure it.

**Screenshots**
[image]

[image]

**System (please complete the following information):**
 - Model: Alienware 17 R1
 - Model details: standard config
 - OS: Windows 11 Home
     - Version: 22H2
     - OS Build: 22621.1413

**Additional context**
* The event viewer captures the following error:
```
Faulting application name: alienfx-gui.exe, version: 8.2.6.0, time stamp: 0x641e6485
Faulting module name: KERNELBASE.dll, version: 10.0.22621.1413, time stamp: 0x5129184b
Exception code: 0xc000041d
Fault offset: 0x000000000002bd6d
Faulting process ID: 0x0x2334
Faulting application start time: 0x0x1D95F472249DD9F
Faulting application path: C:\Program Files\AlienFX Tools\alienfx-gui.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report ID: 52809847-d5d0-4513-ae58-6c5e09ddc983
Faulting package full name: 
Faulting package-relative application ID: 
```

```
[-](https://github.com/T-Troll/alienfx-tools/issues/new?assignees=&labels=&template=bug_report.md&title=#) <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
[-](https://github.com/T-Troll/alienfx-tools/issues/new?assignees=&labels=&template=bug_report.md&title=#) <System>
  <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" /> 
  <EventID>1000</EventID> 
  <Version>0</Version> 
  <Level>2</Level> 
  <Task>100</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x8000000000000000</Keywords> 
  <TimeCreated SystemTime="2023-03-25T18:25:17.9024881Z" /> 
  <EventRecordID>2394</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="5640" ThreadID="4416" /> 
  <Channel>Application</Channel> 
  <Computer>DESKTOP-4</Computer> 
  <Security UserID="S-1-5-21-721532812-1001" /> 
  </System>
[-](https://github.com/T-Troll/alienfx-tools/issues/new?assignees=&labels=&template=bug_report.md&title=#) <EventData>
  <Data Name="AppName">alienfx-gui.exe</Data> 
  <Data Name="AppVersion">8.2.6.0</Data> 
  <Data Name="AppTimeStamp">641e6485</Data> 
  <Data Name="ModuleName">KERNELBASE.dll</Data> 
  <Data Name="ModuleVersion">10.0.22621.1413</Data> 
  <Data Name="ModuleTimeStamp">5129184b</Data> 
  <Data Name="ExceptionCode">c000041d</Data> 
  <Data Name="FaultingOffset">000000000002bd6d</Data> 
  <Data Name="ProcessId">0x2334</Data> 
  <Data Name="ProcessCreationTime">0x1d95f472249dd9f</Data> 
  <Data Name="AppPath">C:\Program Files\AlienFX Tools\alienfx-gui.exe</Data> 
  <Data Name="ModulePath">C:\WINDOWS\System32\KERNELBASE.dll</Data> 
  <Data Name="IntegratorReportId">52809847-d5d0-4513-ae58-6c5e09ddc983</Data> 
  <Data Name="PackageFullName" /> 
  <Data Name="PackageRelativeAppId" /> 
  </EventData>
  </Event>
```

T-Troll, 2023-03-25:
Interesting.

I tried to reproduce it at my 15R1 (same hardware), but it starts with clean configuration without any issue.
According to these details, seems like it Windows 11 related, i'll check what it can be.

T-Troll, 2023-03-26:
Ok, i need some data from your system.

Can you, please:
- Open Administrator CMD
- Run `alienfan-cli RPM` and share output
- Run `alienfx-cli probe` (answer "n" for the question) and share output

mickael28, 2023-03-26:
Thanks for the guidance. I'm getting this:

```
# alienfan-cli rpm
AlienFan-CLI v8.2.6
Hardware not compatible.
```

```
# alienfx-cli probe
alienfx-cli v8.2.6
Dell API ready, 1 devices found.
===== Device VID_187c, PID_0524 =====
+++++ Detected as: Alienware M17x, APIv2 +++++
Do you want to set devices and lights names?n
Done.
```

Reg files for `alienfx-config backup`:

1. afx-fans.reg
```
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Alienfan]
"StartAtBoot"=dword:00000000
"StartMinimized"=dword:00000000
"LastPowerStage"=dword:00000000
"UpdateCheck"=dword:00000001
"DisableAWCC"=dword:00000000
"KeyboardShortcut"=dword:00000001
"KeepSystemMode"=dword:00000001
"DPTF"=dword:00000000

[HKEY_CURRENT_USER\Software\Alienfan\Powers]
"Power-0"="Manual"

[HKEY_CURRENT_USER\Software\Alienfan\Sensors]
```

2. afx-gui.reg
```
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Alienfxgui]

[HKEY_CURRENT_USER\Software\Alienfxgui\Profiles]

[HKEY_CURRENT_USER\Software\Alienfxgui\Zones]
```

3. afx-lights.reg
```
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Alienfx_SDK]
```

4. afx-mon.reg
```
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Alienfxmon]
"AutoStart"=dword:00000000
"Minimized"=dword:00000000
"UpdateCheck"=dword:00000001
"wSensors"=dword:00000001
"eSensors"=dword:00000000
"bSensors"=dword:00000000
"Refresh"=dword:000001f4

[HKEY_CURRENT_USER\Software\Alienfxmon\Sensors]
```

T-Troll, 2023-03-26:
Oh, so you have old M17xR1, not modern m17R1 (it's from here - `+++++ Detected as: Alienware M17x, APIv2 +++++`). And install W11 into it...

Bad news - fan control will not work for your system. Fan V2 will never do, Fan V1 can, but me and you need some effort for this.
Good news - light control is supported.

So... First, try today's release - i do something can potentially solve your issue. I also check EXACT configuration tomorrow (using mine M13 with same light chip), and return with update - did it work for me or not.

mickael28, 2023-03-26:
Thanks, today's release, 8.2.6.1, failed for me with the same issue (crash in event viewer and 'please set lights first' message when trying to launch it)

But I think the `alienfx-cli probe` didn't detect my model correctly? I don't think it's the old model 17xR1, it's reported on my BIOS and system information as `Alienware 17` and it's got the look of the more models, not the older, thicker version:

[image]

It's similar to this design from 2014, which when I checked a while ago I thought it was an Alienware 17 R1:
https://www.worthpoint.com/worthopedia/alienware-gaming-laptop-17-p18e001-1791412896

At the bottom of my laptop it mentions:
* P18E001 , it case that makes any difference

T-Troll, 2023-03-26:
`probe` report what light chip have into its hardware. So yes, your model can be 17RX, but the chip is from M17x.

Thanks for info, seems like i lost a way in fan controls - i check this today.
Now you can try this:
- Open regedit
- Navigate `HKCU\Software\Alienfxgui`
- Create DWORD key named "FanControl" with value 0
- Try to run GUI

T-Troll, 2023-03-27:
So, here is debug version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11073486/alienfx-gui.zip)
I alter some APIv2 processing, this can help.

Please check it, in case it crashed - download and run `DbgView` utility (from Microsoft website), then run this again - it will provide debug info of what happened, please share `DbgView` output.

mickael28, 2023-03-27:
Same crash and message with the attached `alienfx-gui.zip`.

A couple of `dbgview` logs:

[DESKTOP-4R3D570-2nd.LOG](https://github.com/T-Troll/alienfx-tools/files/11073715/DESKTOP-4R3D570-2nd.LOG)
[DESKTOP-4R3D570-1st.LOG](https://github.com/T-Troll/alienfx-tools/files/11073716/DESKTOP-4R3D570-1st.LOG)

Just in case, I tried with the manually created `FanControl = 0` entry and same crashed too (with latest alienfx-gui or with the original one from 8.26.1)

Thanks so much for the ideas trying to help with the problem! 

All the attempts get captured in the event viewer with the initial error:
[image]

T-Troll, 2023-03-27:
Aha! We catch it finally!
Crash happened into USB light device driver, then i try to check light with ID#0.
Can be incorrect command or something wrong with ID. Let me check.

T-Troll, 2023-03-27:
Can you please try this debug version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11074660/alienfx-gui.zip)?

I change something, and also have more debug info, so `DbgView` log will be useful.

mickael28, 2023-03-27:
Windows defender deletes the file detecting it as a virus, do you think is it ok in that debug version? (I didn't run it yet to see what you think)

[image]

T-Troll, 2023-03-27:
Defender and SmartScreen is a crap working to block what M$ don't like, but not protect from real threat.
Check this (or do yourself) -https://www.virustotal.com/gui/file/b4e5d07145fd475c51dd53133bf78fecef8f2ef95f17eb3f89017471c5fcddc7?nocache=1

mickael28, 2023-03-27:
Finally defender allowed me to run it after adding a few allowed exceptions.

Same error/crash and `DgbView` output is quite similar to previous one. Not sure if you'll be able to find out there something useful. Attaching it.

[DESKTOP-4R3D570-3rd.LOG](https://github.com/T-Troll/alienfx-tools/files/11078194/DESKTOP-4R3D570-3rd.LOG)

T-Troll, 2023-03-27:
No, it's not the same and important!
Thank you!

T-Troll, 2023-03-27:
Check this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11083155/alienfx-gui.zip)

Seems like i found the reason (yes, it was into your V2 interface), hope i not broke it, so check, please:
- Is lights changing?
- Is timings ok (can be very slow, like some seconds delay for every reaction)?

mickael28, 2023-03-28:
Hey man, it seems you got it! I've not followed what the issue was on my laptop but on the latest GUI it launched. Will that change be included in future updates too?

Not sure what's supported yet on my laptop. I need to experiment after reading the beginners guide but I'm going to be away on holidays soon for a few weeks so it might take a while.

One doubt though with the brief testing I could do today. I saw the Grids tab was recognising lights in different parts of the hardware as blocks (eg, the 4 different parts of the keyboard, like 4 different areas in a grid), but it didn't recognize any key individually.

Do you know if that means that on the 1st tab for Lights, I can just change the colour for those 4 areas of the keyboard in my grid but I won't be able to assign lights to individual keys on this specific laptop?

T-Troll, 2023-03-28:
Glad to hear it! Of course, it will be part of the next release and so on.

Grids are optional, you may not use it for your hardware - it's for per-key RGB keyboards mostly. But you CAN use it for basic spatial effects and gauges.

For quick check, use buttons on top of lights list, not the grid - it's a way to check is this light connected (it will change color) or not. And yes, read Beginner's guide for details, please!

Device tab is for telling the app what lights you have in general. Other tabs about how to control it. So first set lights at Devices, then assign it to color in Colors/Monitoring/etc.

mickael28, 2023-03-28:
Got it, thank for all the help troubleshooting what was going on. It was good you were able to catch the issue :)

T-Troll, 2023-03-28:
Yes, it affects most of old devices (v2/v3). I rarely use my old 13R2, so didn't catch it.
So thank you for point me this!