T-Troll/alienfx-tools #331: Audio problem with the use of alienfx

meetlicious, 2023-04-03:
Hi , im having audio issues with use of the alienfx , when i open it the audio crack/pop with 1 1,5 second  until i kill the process or restart the laptop 

my model is g15 5525 , ryzen 5 6600h rtx 3050 and windows 10 last version , everything works fine just the audio bug/problem .

Let me know if i can submit something to help in the troubleshooting. 

Thank you !!!

T-Troll, 2023-04-03:
Hmm... It looks quite strange.

Instead of AWCC, i don't mess with audio and even touch it, except Haptics effect active (i copy system output stream in this case, so it shouldn't provide any mess as well).

Also, CPU load by my tools is not high enough as well to provide this effect from system overload.

Most probably, it's some audio driver/system/player issue, just triggered by other running app, not mine only.

So, for troubleshoot check other player apps, make sure you have the latest drivers FROM DELL SUPPORT SITE (it's important!), as well as try other players. And yes, disable Haptics effect if you have one - in this case audio access will not be even initialized.

meetlicious, 2023-04-03:
Well , this is quite strange . 

Its a Brand New model with 3 days of usage , i make a fresh install of w10 just to make shure something that came front fábric is conflicting . 

If i dont open the alienfx this “bug” dont appear in w10 , w11 awcc and the alienfx make the same “bug”. 

The áudio problem exist in every áudio source mp3,Spotify,YouTube,games, in w11 even the system sound .

i Will make some more tests .. 

Thank you !!!

T-Troll, 2023-04-03:
Got it. What do you have enabled into AlienFX? Fan control? Light effects? Plain light set?

meetlicious, 2023-04-03:
Only the fan control , nothing more , tomorrow i will test some more thing in the app and return it here !! 

Thank you !!

meetlicious, 2023-04-03:
I found the "problem" here , if i select the power mode to manual , the "bug" immediately start and just stop if i close or disable the fan control, if i keep on the "levels" just work fine until now in 1 hour of test . Using both AlienFX and Alienfan - GUI 

Oh , when switching the levels mode , the crack in audio appear when select.

Thank you !!

T-Troll, 2023-04-03:
Wow! Interesting!
"Levels" (Power mode) = "Thermal profiles" into AWCC. So manual mode make this bug.
Well... Try to roll back BIOS, seems like something wrong with latest one.

T-Troll, 2023-06-11:
> Maybe it's related to WMI calls?

Seems like. I don't mess with audio in itself, but can be side effect. Let's investigate it deeper, let me think and ask you to run some CLI commands to pinpoint.

> https://github.com/AlexIII/tcc-g15

I've contacted him already... He is just starting the way i pass then develop it.

T-Troll, 2023-06-16:
> I am currently looking for a program that can solve the unreasonable fan curves (it's so loud during light work even on battery) that doesn't negatively affect my experience.

You can't. Fan control is indirect, so BIOS decide fan RPM (and this for good in most cases).
But you can try to play with different power modes, some have a lowered fan curves.
.. And switch to manual/performance by running app (`AlienFX-GUI` can do it).

Amrsatrio, 2024-02-23:
Hi, sorry for the late followup. I forgot to add that the stutter happens even when I'm changing settings in AWCC, so do you think it's unavoidable in my model?

T-Troll, 2024-02-24:
Yes, seems like some hardware mode changes provide this.
Try to play with audio drivers (downgrade, use universal/patched etc), most probably the issue into it.