T-Troll/alienfx-tools #333: Unable to update from 8.0.0

XRone99, 2023-04-06:
Everytime I try to upgrade version from 8.0.0 to any other more recent version the new version crashes as soon as I open it.
There's not any kind of error displayed on screen, the program just closes itself.

**System Information**
 - Model: _Alienware x17 R2_
 - Model details: [CPU: Intel Core _i7-12700H_; GPU: Nvidia _RTX 3080Ti Laptop_; Keyboard: _MX Cherry RGB_; RAM: 16GB _DDR5_]
 - OS: [Edizione	_Windows 11 Home Versione 22H2_  Data installazione: ‎05/‎10/‎2022   Build sistema operativo 22621.1485 
 Esperienza Windows Feature Experience Pack 1000.22639.1000.0]

**Additional context**
We already got in touch under 8.2.5 version comment thread, this is the latest backup I made (06/04/2023)
[alienfx-tools 8.0.0 backup.zip](https://github.com/T-Troll/alienfx-tools/files/11171424/alienfx-tools.8.0.0.backup.zip)

T-Troll, 2023-04-07:
Here are debug version for you - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11177340/alienfx-gui.zip)
You will need Microsoft tool named `DbgView`.
Run `DbgView` (from admin), then run app (from admin).
Please share output from `DbgView`.

XRone99, 2023-04-08:
I just did what you asked this is the output
[LORENZO-ALIEN.LOG](https://github.com/T-Troll/alienfx-tools/files/11182874/LORENZO-ALIEN.LOG)

T-Troll, 2023-04-08:
Oh, i see. Interesting.... It's somewhere into effects.
Let me check and think.

XRone99, 2023-04-09:
There's something else I could do to help debug?

T-Troll, 2023-04-12:
I share new debug version for check today.

T-Troll, 2023-04-12:
Here is new debug version, please run the same way (i need DbgView log) - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11210683/alienfx-gui.zip)

I've added some debug info to pinpoint the fault point, and also some checks. Seems like an issue somewhere into APIv5 (keyboard) control code.

XRone99, 2023-04-12:
[LORENZO-ALIEN.LOG](https://github.com/T-Troll/alienfx-tools/files/11212419/LORENZO-ALIEN.LOG)

Here's the log, program crashed ad always
Let me know if I can do anything else to help

T-Troll, 2023-04-12:
A mystery.
Good news it's not APIv5, it's working well.
Something wrong with string creation, some entity has forbidden name...
Ok, let me check you config.

T-Troll, 2023-04-13:
First, i not sure your current config is the same as your backup before (it's saved then elevation prompt appear).
Anyway, it should not be this.

Here is another debug - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11218219/alienfx-gui.zip) - please check it the same way.

In case it crash, try to check some other ways:
- Open `RegEdit`
- Navigate to `HKEY_CURRENT_USER\Software\Alienfxgui`
- Change `Monitoring` (because i see your default profile using Ambient) and `ShowGridNames` to 0.
- Start again

PS: I see you have `Gamma correcton` disabled. Well... It will make lights a little brighter at the center colors, but it will not fit screen colors in many cases.

XRone99, 2023-04-13:
[backup.zip](https://github.com/T-Troll/alienfx-tools/files/11218620/backup.zip)
[LORENZO-ALIEN.LOG](https://github.com/T-Troll/alienfx-tools/files/11218625/LORENZO-ALIEN.LOG)
[LORENZO-ALIEN-EditedREG.LOG](https://github.com/T-Troll/alienfx-tools/files/11218626/LORENZO-ALIEN-EditedREG.LOG)

Before editing register I produced the first log and the program crashed as always, after that I edited register as you said and the program worked (second log)

XRone99, 2023-04-17:
Followed the first point of your list (deleted AlienfxGUI key) and the debug version you sent me worked fine until I minimized it
Meanwhile I set up some setting manually as they were before the update
Now after some attempts to start it the situation returned to previous (so it doesn't even start and crushes diractly)
I wonder if there's something with te fact that the program starts minimized...
By the way here's log.
Sorry for late reply but in those days I've been away from home
[LORENZO-ALIEN 3.LOG](https://github.com/T-Troll/alienfx-tools/files/11247786/LORENZO-ALIEN.3.LOG)

T-Troll, 2023-04-18:
Hmm... Going minimized, just unload active UI and start auto profile switch.
But according to delay between refresh and exception, this can be other.

So let's make the last effort - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11257802/alienfx-gui.zip)
Try this version (i don't need logs if it crashed, it will be the same).

We have crazy situation - your configuration works well at my gear, and some other guys didn't complain about crashes on the same gear as you have. According to exception, it looks like a combination of settings + your locale.

So... For next steps, we only have a choice of the remote debug at your machine. Contact me in Discord (link to the server at project readme) about it.

XRone99, 2023-04-18:
Ok, I just downloaded this version: 

1. It seems to work but it erased all the settings of ambient lighting (it's not a problem i'll set them back)
2. Now I'm going to set up manually my previous setup and see if it makes app crash
3. Set up ambient lighting in a couple of profiles and still seems to work properly
4. Going to reboot and see if it starts properly
5. System rebooted normally and AlienFX GUI started with windows as expected
Everything seems to work pretty good, I don't know what you did but I think we can say the issue has been fixed

T-Troll, 2023-04-19:
I'm glad to hear it, but hold on!
All i do.... remove tray tooltip. Seems like it provide crash on your settings then generates tool tip text. The good question is why, so we need to investigate it.

XRone99, 2023-04-19:
What can I do to help?

(Besides i found something that could be a little bug: i created 3 profiles to provide different lighting and fan behaviour in different situation but for some reason 2 of these are bond; if I modify Fan profile on one the other gets modified too, I don't know if it's some kind of bond setting or it's a real bug)

T-Troll, 2023-04-20:
Well... Here are release with some changes into tooltip generation. Please check it - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11279840/alienfx-gui.zip)

It's two possible broke point inside it, this is the check for one (and new format as well).

About profiles, make sure you have "Fan settings" selected for profile at "Profiles" tab, or it will share the same system fan profile. If you check it, a copy of system fan profile will be copied, and you can edit and apply it separately.
NB: You will edit system fan or profile fan regarding of this setting and what profile selected into top UI line.

PS: I found a bug into Windows. Then i add Italian locale for testing, it also installs Italian keyboard unexpectedly. And it can't be removed! (well... i remove it after some effort). Report this to M$.

XRone99, 2023-04-20:
This version seems to work well, here's log
[LORENZO-ALIEN 4.LOG](https://github.com/T-Troll/alienfx-tools/files/11283786/LORENZO-ALIEN.4.LOG)

About fan issue the situation fixed itself for some reason ("Fan setting" was already selected on each of the 3 profiles I have)

T-Troll, 2023-04-20:
Glad to hear it.
So... I can tell you what happened. You config provides a moment you have no active profile at all, and W11 ask to generate tooltip at this time. I didn't check this situation before, so app crashes.

I think fan was not an issue, you just edit incorrect profile or profile without "Fan settings" active.

The version i send is a release one, so you can continue to use it. I'll patch it at next release as well.

closed 2023-04-20 by T-Troll

XRone99, 2023-04-20:
Thank you for support, I'm happy to have been of help