T-Troll/alienfx-tools #334: Link CPU turbo boost setting with the profile

GiowGiow, 2023-04-07:
**Is your feature request related to a problem? Please describe.**
Currently, the CPU turbo boost setting works independently from the profile setting. Which I think is counterproductive for some use cases, for example:

I can link applications to a specific profile, which means for GPU-demanding applications it would be best to have G-Mode on, for max fan speed and 15w more for the GPU, and the CPU turbo boosts off to keep temps down as my laptop has the CPU and the GPU dye sharing the same heat pipes. After that application is finished execution it would be best to have the laptop back in its normal state with the default profile.

**Describe alternatives you've considered**
Have the profile save the CPU mode on AC and on Battery settings and apply them when the profile changes.

**System (please complete the following information):**
 - Model: Dell G15 5515 
 - Model details: [e.g. per-key rgb keyboard]
 - OS: Windows 11

T-Troll, 2023-04-07:
Yes, it is. 

Because it's Windows Power plan-related, so in case you change plan it will change too!
So the reason i don't want to mess with power plan monitoring and serious modification (it's dangerous), but you always can have any settings in action just switching Power Plan.

GiowGiow, 2023-04-08:
Currently we have the "CPU mode AC" and the "Battery" settings 
[image]
My suggestion is to save and apply those when you change the "Active Profile"
[image]
Is there any dangerous behaviour that could happen with this change?

closed 2023-06-03 by T-Troll