T-Troll/alienfx-tools #335: G15 5511 3060 can't open alienfx-gui and alienfan

Suezqen, 2023-04-09:
I'm a new hand on here. I tried to open the tool after installing the MSI file, but only the monitor can open it. These two cannot be opened, and both have been granted administrator permission to request an answer!

T-Troll, 2023-04-12:
So it crashes after start immediately?
Ok, let me check, but be ready to got debug version from here to investigate.

Suezqen, 2023-04-12:
I don't know why when I downloaded AWCC back and then installed it, it actually worked. What I saw on our forum was that I needed to completely remove AWCC and its components before I could use this software. A few days ago, after using your installation package without any results, I downloaded AWCC back. The GUI is now functioning properly. I also have a question, do you need to completely remove awcc and its components?

T-Troll, 2023-04-12:
No, you don't need it (and yes, AWCC can be installed automatically by Windows update).
You need to do one of 2 things to prevent cross-control:
- Stop and disable AWCCService in Services
- Or just check "Disable AWCC" in Settings.

closed 2023-04-17 by T-Troll