T-Troll/alienfx-tools #337: Sensors Not Correctly Dispalyed

JazzyJazzjazzy, 2023-04-15:
My listed Sensors in AlienFx-Fan does not correspond to what is in my registry. Registry Entries are below and .png shows what is displayed in AlienFx-Fan.

"SensorName-0-0"="ESIF sensor 1"
"SensorName-0-1"="ESIF sensor 2"
"SensorName-0-2"="ESIF sensor 3"
"SensorName-0-3"="ESIF sensor 4"
"SensorName-0-4"="ESIF sensor 5"
"SensorName-0-5"="ESIF sensor 6"
"SensorName-0-6"="ESIF sensor 7"
"SensorName-1-1"="CPU Internal Thermistor"
"SensorName-1-6"="GPU Internal Thermistor"
"SensorName-2-0"="SSD sensor 1"
"SensorName-2-1"="SSD sensor 2"
[image]

T-Troll, 2023-04-16:
This means you lost ESIF sensors, but it was into the system before.
ESIF sensor data provided by Intel DPTF (Dynamic Tuning Framework driver), so make sure you have it installed.
Also, some DPTF drivers do not provide this data to system (just collect and use inside the driver), so better check you have drivers from Dell's support site, not from Intel one (or just find one which provide it).

How to check you have DPTF data? Open `perfmon`, select "Performance monitor", press "+" button. Then the list populated, look for "ESIFDeviceInformation" entry. In case it presents into the list, all work correctly.

JazzyJazzjazzy, 2023-04-16:
Hello,

In PERFMONITOR, I do not have ESIF Entry. Goes from Energy Meter to Event Log, skipping ESIF. 

I reinstalled Intel-Dynamic-Tuning-Intel-Dynamic-Tuning-Driver_90GWR_WIN64_8.7.10600.20700_A12.EXE from DELL site and observed No Change in my display.

Also used [alienfx-tools.msi] to reinstall AlienFx as opposed to the .zip and still No Change.

T-Troll, 2023-04-16:
As i said, some DPTF does not support it.
You can try `fan_sdk_v1,zip` version (READ README FIRST!), but i don't sure it supports fan control at your system.

JazzyJazzjazzy, 2023-04-16:
Your program has been supporting my laptop since I started using it months ago. Now, why suddenly it just stops supporting it? 
It is a relatively new Alienware m17 R4 with i7-10870H, RTX 3070.

T-Troll, 2023-04-16:
V1 is older SDK and some models not supported. But AFAIR R4 is.

JazzyJazzjazzy, 2023-04-16:
Why would I have to resort to using V1 when laptop previously running fine on most up to date version? What has changed?

T-Troll, 2023-04-16:
Good:
- You will have more sensors
- Lower CPU usage for fan control
Bad:
- Antivirus will be crazy about it

BTW, i check [this](https://github.com/T-Troll/alienfx-tools/wiki/Supported-and-tested-devices-list) - yes, R4 is supported.

JazzyJazzjazzy, 2023-04-16:
Thank you for this information but I think I will pass. When you UPDATE your software is when I notice inconsistencies creep in or things get broken. Kindly recall most of my previous issues were observed after an update. This current issue just "popped up" out of now where after my computer has been fine for many months. I do not think a 10th Generation cpu is that old to need V1 SDK. Once. I suggested separate versions in order to avoid this, however, you wrote that you  wish to maintain commonality and I understand and respect that. I will download previous versions to see if any one of them still work. If not, then I will continue using current version as my fans do reliably respond to the visible sensors. Thank you.

closed 2023-04-16 by JazzyJazzjazzy

JazzyJazzjazzy, 2023-04-16:
Sir, I wanted you to see my CPU & GPU Temps while running MSFS 2020 in a 29 degree Celsius room. I could not have achieved these values without your software as I have played this game for the past year and I know what temps have been. I am grateful for all of your work. Thank you. Maybe if I can find the V1 SDK, I will give it a try.

[image]

JazzyJazzjazzy, 2023-04-26:
[image]
Repair installed W11 and now all sensors are visible. I suspect that it was my W11 install and not your software causing issues. Apologies. Thank you.