T-Troll/alienfx-tools #340: AlienFX Control(alienfx-gui.exe) and Alien Fan(alienfan-gui.exe) both do not detect the CPU temperature correctly

meemkoo, 2023-04-30:
**Describe the bug**
The bug is that both programs (listed in the title) cannot seem to detect the temperature properly or adjust fan speed accordingly.

**Expected behavior**
I expected for the 2 programs to detect the temperature and apply it to the fan curves.

**Screenshots**
Both of these screenshots display Core Temp and MSI afterburners monitoring window both showing temps of 52~ degrees, while both alienfx-gui and alienfan-gui show 25 degrees
[image]
[image]

**System:**
 - Model: Dell G5 5000
 - Model details: 2 fans, one CPU fan and a side fan(I don't know which one is fan 1 and fan 2) 
 - OS: Windows 10 stable

**Additional context**

- As shown in the screenshots, I installed MSI afterburner which may have messed cause my very situation. However I find this unlikely as I originally installed it a couple months ago to replace AWCC(Which leads into my next point). However(x2), I found out it was for GPUs but I kept it installed because it worked for my GPU, until I reinstalled windows 3 weeks ago. However(x3) I installed it again

- Another notable point I should make is that for, well ever since I bought this computer its fans have just been a pain with the only two programs(so far) that are able to control them being alienfx-tools and AWCC. Probably because, even though its not a laptop the (goofy) proprietary-prebuilt motherboard having the fans configured wired.

T-Troll, 2023-04-30:
Wow!

No, it's not a bug, but misleading a bit.

You have really crazy BIOS, i see things like this first time among all models!

Let me describe you how it works and comment:
- First set of sensors detected from proprietary AWCC BIOS methods. All systems have 2-3 sensors, first always CPU, 2nd always dGPU. You have 5(!) of them, and seems like CPU is not the first (3rd, AFAIC).
- I'm also trying to detect fan type based of the sensor it uses (BIOS have this info). But in your case they both belong to uncommon sensors with the type i can't detect, so it's just "fan" as a result.
- You have only 2 ESIF sensors (and names not retrieved). Do you have G-letter CPU (without iGPU)? BTW, ESIF 1 is package temp, ESIF2 is CPU cores temp.

So... Just connect curves to correct sensors (and rename sensors to something understandable if you wish), and it will work correctly.

Also, it will be useful for me if you run some commands using `alienfan-cli` (but i need to share some test version for you) to understand BIOS sensor mappings and IDs.

PS: It's a combination of MB and BIOS - Dell don't provide direct fan EC control other guys do, indirect control only (and this is right in many cases). So only tools based on proprietary calls will work.

jpatchMC, 2023-04-30:
hey this might be unfounded as we have different systems.  I have a m15 r5.  not too long ago microsoft released a dell firmware update for my system that causes both awcc and alienwere fan control to read my cpu at 80 something and the gpu at 2, but hwmonitor would report different (likely the real) temperatures. If its that and not fan control I hope you have an easier time rolling back. I will be reimaging when my semester is over

T-Troll, 2023-04-30:
@jpatchMC Yes, BIOS update can do this. Maybe they provide more data preparing newest AWCC release. It will be nice to have your BIOS dump (the same for @meemkoo) to check what do they change.

Fan control will work anyway, but you may need to create curves for other sensors as well.

closed 2023-06-16 by T-Troll