T-Troll/alienfx-tools #341: Alienfan-gui stops controlling CPU fan after computer sleeps or is plugged in

luked144, 2023-05-01:
Been using this program for a few months now, I feel like this has been an issue for a while now. It's annoyed me to the point of posting it here as a bug, perhaps it is just user error but I doubt it as I've since reformatted the computer, updated the BIOS once again etc and redownloaded the program and the problem persists.

It seems like after a certain amount of time using the computer, or if I leave it to idle so it sleeps and then start using it again, or if I plug the computer in after using it on battery, the CPU fan just doesn't turn on anymore. I haven't been using anything GPU intensive so I do not have any results regarding the GPU fan, this is all in regards to the CPU fan. I have the fan curve set as a straight line (40 degree = 0% fan, 71 degrees = 100% fan), about as simple as it gets. The program will work after freshly starting the computer up, it will work if I close the program and reopen it, but after an indeterminant amount of time of the program being open (seems to be about a couple hours), or if the computer sleeps and then is awoken, or if the computer is plugged in from battery, it will still read the correct temperature from the sensors, but the fans will not turn on. Sometimes it will say that the fans are at some RPM (currently it says the CPU fan is at 3000 RPM), but the fans are not spinning whatsoever. Most times it just says the fans are at 0, which I suppose is accurate as they aren't spinning, but they SHOULD be in regards to the fan curve and the temperature of the CPU.

The simple fix is to just close and reopen the program or to restart the computer but one can probably understand how annoying that is to do every couple of hours, especially if you're in the middle of something.

I am using version 8.2.6.5

The computer is a G15 5515, AMD Ryzen 7 5800H, 16 GB RAM, Nvidia 3060, Windows 11

T-Troll, 2023-05-01:
Hmm... Interesting. In case temperatures updated, my procedure working correctly.
Where is red dot at fan curve window?

But seems like something wrong with BIOS - fan control is indirect, all m app doing is just a BIOS hint, not control fans directly. As well as RPM - i just read it from BIOS, so in case it reports 3000 i have no way to check it's true or not.

You can also have an issue with the fan in itself, it can be dusty and can't spin correctly at low speeds.

Try to set lower boost between 35 and 50 (5-10), this can prevent fan from stop by BIOS.

luked144, 2023-05-01:
The red dot seems to be accurate, I have been checking with HWMonitor. It seems like everything about the program is working as you intended. I wasn't intending to insinuate that it was your program that was the problem, it seems like these laptop's BIOS are an issue for a lot of people looking through other complaints. It's just frustrating I suppose, and I understand it's not really something you can control or do anything about.

I appreciate you making this program nonetheless, it is certainly better than AWCC not starting the fans at all until ~80 degrees lol. I don't believe it's dust or anything physical as this laptop is rather new and I don't use it very often as is (partially because of these problems).

I have adjusted the fan curve as per your suggestion; hopefully that will rectify the problem, perhaps it is something as simple as the fans not being restarted by the BIOS and not letting them stop would indeed fix that.

Thank you for your reply, and for the program!

T-Troll, 2023-05-02:
Ok.

Please open Administrator CMD, run `alienfan-cli rpm` and share me output (i'm interested in your systemID). I'll try to make override of this bug (it happened at some other systems as well, so we know how to do).

Also, as a small fix, try to use Alt+Ctrl+1 then Alt+Ctrl+0 then it happened (this change power modes). Did it help?

closed 2023-06-16 by T-Troll