T-Troll/alienfx-tools #344: CPU Fan doesn't spin on sartup

XRone99, 2023-05-15:
**Describe the bug**
Very often on system startup CPU fan doesn't spin even if the program says it's spinning at some RPM

**To Reproduce**
Steps to reproduce the behavior:
1. Start the system
2. AlienFX GUI starts itself with windows on tray
3. I see cpu temperature slowly rise and fan doesn't fisically spin even if the program shows some RPMs
5. If i restart AlienFX GUI or change "Power Mode" from manual to any other value and the go back to manual everything works perfectly

**Expected behavior**
AlienFX GUI should be able to control CPU fan without the need from me to restart it or force the cange of Power Mode

**System (please complete the following information):**
 - Model: Alienware X17 R2
 - Model details: i7-12700H, RTX 3080Ti Laptop, 16GB RAM
 - OS: Windows 11 Home 22H2

**Additional context**
I use a custom fan curve and it works perfectly fine if I do that thing to make AlienFX get unstuck

T-Troll, 2023-05-16:
Try this one - it should help - [fan_gui.zip](https://github.com/T-Troll/alienfx-tools/files/11485890/fan_gui.zip)

XRone99, 2023-05-20:
Ok, I've been using this version you gave me for some days with no problems BUT as I said the problem already was intermittent
In fact this morning it happened again, as always switching mode and going back to manual did the fan start spinning correctly as always

T-Troll, 2023-06-03:
I do some into latest release version.
If this didn't help again, seems like issue into BIOS, and you should configure out the system other way - i tell you how.

XRone99, 2023-06-03:
So I should download the latest version?
(By the way in the last 2 weeks or so fans have worked fine)

T-Troll, 2023-06-03:
It has some other fixes about it, so try.

XRone99, 2023-06-03:
Thank you, i'll try it and let you know <3

XRone99, 2023-06-10:
I've tested latest version for some days and I can tell that the problem is still there; as always it is discontinuous, sometimes it work, sometimes fans are blocked on 700rpm (sometimes is CPU1, other times is CPU2 or GPU 1 and so on; it seems to be random)
As always changing between profiles does work and restores function

T-Troll, 2023-06-11:
Hmm....

Latest version ALWAYS switch between power modes then resume from sleep. So i really don't understand how it is possible to still hang fans. Maybe timing is an issue....

XRone99, 2023-06-12:
Today I've noticed something:
It's NOT that the program doesn't control the fan spinning, actually it does
The thing is that Fan Max RPMs are lower then actual maximum until I change Power Mode to another profile and then switch back to Manual
For example CPU2 fan will spin at 2000/2500 RPMs on manual mode 100% boost BUT if I switch to another profile and then go back to manual it will spin at 3000 RPMs
And I can tell this difference matters in terms of temperatures (and also is audible even without looking at RPMs)

I wish this could be helpful

T-Troll, 2023-06-13:
Oh... Look at this - https://www.reddit.com/r/Alienware/comments/147sckl/fan_speed_not_working/

Seems like something broken into BIOS for some models...

About you issue... All i can do is delay power mode switch - MAYBE this can help. Now i switch mode to Balanced then app start/resume, then switch to mode you define into profile. But now it will be done immediately, so i have an idea some delay between can help.