T-Troll/alienfx-tools #346: M18R1 I cannot control the lights of F1&F12

knknwxc, 2023-05-18:
I cannot control the lights of F1&F12.
It will be lighting on when i choose the color.A well later it will be extinguished.
First the phenomenon showed on F1,F12,volume_mute,mic_mute,windows_lock.
But i disable the effect and the other keys works fine except for F1&F12.
Thank you

T-Troll, 2023-05-18:
Hmm.... Seems like m16/18 uses different keyboard control chip, so this can be a result of some new features (like key-controlled light). Please check - did it light up, for example, after Fn press or so?

In general, we need to study how AWCC control it and make some corrections into my code.
In case you are ready to help me with this, i'll share a short manual for it.

knknwxc, 2023-05-18:
I'd like to help you.
What website or online drive is convenient for you to watch the video to.
I will show you the phenomenon

T-Troll, 2023-05-18:
> I'd like to help you.

Ok. So.... First open CMD and issue this command `alienfx-cli probe`. Copy output for me (it's light chips IDs). Answer "N" for the question.

Second, read [this](https://github.com/T-Troll/alienfx-tools/wiki/How-to-collect-data-for-the-new-light-device) wiki page.
We don't need to follow it directly, because your chip is APIv5(?) variation.

Install USBPcap, stop `AWCCService`, run USBPCap, start it back (to grab init sequence). Then open UI and check settings - did it have something about it? If yes, switch it. If no - just dim lights, go dark and back to light. Stop USBPcap and share a log here.

> What website or online drive is convenient for you to watch the video to.

Youtube is ok. In case you are in China, you can use yandex.disk (yandex.com) for share video file.

knknwxc, 2023-05-18:
[USBPcap.zip](https://github.com/T-Troll/alienfx-tools/files/11508338/USBPcap.zip)
===== Device VID_187c, PID_0551 =====
+++++ Detected as: Alienware AW-ELC, APIv4 +++++
===== Device VID_0d62, PID_aab0 =====
+++++ Detected as: DELL Technologies Keyboard, APIv5 +++++

I adjusted the color of the keyboard in awcc and found no bright or dark options.
So I chose several basic colors to change the keyboard lights.
There is also an additional document, and I have made changes to several of the light effects that come with AWCC.

I hope what I have done is correct. If you have any requirements, please continue to contact me

T-Troll, 2023-05-18:
Yes, both chip are new. BTW, last one not Dell, but Chinese Darfon ^_^.
You can contact me at Discord support server (listed into project readme) for faster connection. But please keep in mind, i'm into GMT+5.

knknwxc, 2023-05-18:
Using discord in China requires special skills, unfortunately my profession does not allow me to do so.
The data I collected is in the zip file I replied to last time, and it may be that the name is quite confusing.
If you have any necessary data, you can continue to contact me.

T-Troll, 2023-05-18:
Ah, yes. I use SSH tunnel for it last time i visit your country ^_^
Ok, let me study pcap from zip, i'll be back with the results.

T-Troll, 2023-05-24:
I check files - seems like you choose wrong USB concentrator - there is no data inside (hint: files should be at least 50-200kb)
Can you please do it again selection correct hub (check device list USBPCap show at start)?

isaacsummers, 2023-09-27:
I also have this problem but only with a monitoring action set up and software effects toggled on

T-Troll, 2023-09-27:
Are you sure it's belong to this issue?
Can you please describe it deeply - what do you do and what happened?

isaacsummers, 2023-10-07:
i believe it belongs to this issue or one similar, i have an alienware m16. when toggling software effects on with nothing assined all the lights seem to work fine. However if i create a performance counter and assign it to some lights the backlight or any assigned colors do not work for the F1, right WIN key, microphone mute, and F12 key

T-Troll, 2023-10-08:
Interesting... Looks like you have it altered for the same zone, but how did it happen?
Can you share me your configuration? Open CMD, run `alienfx-config backup`, share `afx-gui.reg` and `afx-lights.reg` (no personal data inside, just your config - you can open it into notepad to check).

JazzyJazzjazzy, 2023-12-20:
> Interesting... Looks like you have it altered for the same zone, but how did it happen? Can you share me your configuration? Open CMD, run `alienfx-config backup`, share `afx-gui.reg` and `afx-lights.reg` (no personal data inside, just your config - you can open it into notepad to check). I have Alienware m18, I9-13900HX, RTX 4080.

I have the same problem of no light for F1, F12, RT WIN, and MIC keys. I thought you might like additional reference material.
[afx-gui.reg.txt](https://github.com/T-Troll/alienfx-tools/files/13733304/afx-gui.reg.txt)
[afx-lights.reg.txt](https://github.com/T-Troll/alienfx-tools/files/13733306/afx-lights.reg.txt)

Applejuice-ger, 2024-01-16:
Hello, I have similar problems, it is problematic for me. First of all, sorry for my bad english, this is not my native language and my school knowledge is old...

Im using an m18 R1 ; German Keyboard Layout ;

After using alienfx-tools the F1, F12, Windows Hold and some Function Keys (Mute, Mic Off) are dark after a reboot.
Reconfiguring lights with Alienware Command Center dont helps, after reboot first lights are dark, then Alienware loads settings and colors of keyboard (error-free), but a few seconds again the specific keys going dark again!
I done a complete reset of my system and clean windows 11 reinstall, the issue remains!

No matter what colors I set in the AWCC, the special buttons always go dark again. The only setting that still works flawlessly is "static blue" in the AWCC. However, as soon as the colors are specifically assigned, the problem occurs.

Apparently alienfx-tools has written not quite correct settings directly to a programmable hardware memory area of the keyboard controller or similar? 
In any case, the problem can no longer be solved on the software side with Alienware tools, but is permanent.

Dear @T-Troll , I hope there is a chance to correct the function of the m18 R1 keyboard lighting again, I am happy to help with research and troubleshooting, I just need information on what to do.
Of course I would also be happy to pay a small compensation for the work and effort once the problem has been solved. 

Here is a photo I took while booting, the problem is clearly visible.
[image]