T-Troll/alienfx-tools #347: Still no way to enable the lights on Alienware

mbelaunzaran68, 2023-05-19:
Reinstaled command center aboutn 50 times already. 
Installed your AlienFx-Tools, but nothing is working

I ran the .\alienfx-cli Prob and status and the result is

PS C:\Program Files\AlienFX Tools> .\alienfx-cli.exe high-level status
alienfx-cli v8.2.5
Dell API not found, devices not found.
Both low-level and high-level devices not found, exiting!

This is an alienware 15 R3
On windows 10

Is there a solution for this

Kind regards

T-Troll, 2023-05-24:
> Dell API not found, devices not found.
> Both low-level and high-level devices not found, exiting!

This means no one light device detected into the system.
AFAIR R3 have separate light chip board with 2 chips (so it should be 2 devices). Check cable from it to MB, as well as it can be broken itself.

mbelaunzaran68, 2023-05-29:
Checked internal cables all seem to be ok. and prior running an update on AlienFX was showing it detected 1 device

mbelaunzaran68, 2023-06-02:
I checked and there seems no cable issue. and Once I downloaded FX Tools I ran it and showed there was 1 device. Then I reloaded Alienware Command Center and no shows cero.

T-Troll, 2023-06-03:
Please confirm i understand you correctly:

- You have 15R3 (NOT m15R3)
- My tools detect light device (but latest version didn't?)
- AWCC (BTW, what version?) did not

Am i correct?

T-Troll, 2023-06-16:
Look for light device present into Device manager (among HID devices).
I can't control non-existing light device, sorry.
It should have `VID_187C`, and use common "vendor-defined" driver.

In case it not found among both HID and unknown devices, you have a hardware issue i can't fix, sorry.