T-Troll/alienfx-tools #348: alienfx-cli crash when I open it everytime

yuhao71, 2023-05-27:
I am a new starter to use this software,I just install the msi file in the E disk without other measure I need to do
when I save the light config,it crash and can not open it again,and can‘t open again and again
can you help to see why?

yuhao71, 2023-05-27:
[image]

T-Troll, 2023-05-28:
1. Breath require 2 colors and some timing (sliders).
2. Anyway, it didn't supported for APIv5 hardware, but you can emulate it via Grid effects.
3. But most interesting fact, you have 0 lights defined for keyboard device! I didn't configure out how it is possible yet.

So, can you, please:
- Open CMD into app folder
- Run `alienfx-config backup`
- Share 4 .reg files it generates here

yuhao71, 2023-05-28:
OK,I am out for work for a few days,I will upload in 2023.06.01

yuhao71, 2023-05-31:
hao can i transfer the reg files to you?

T-Troll, 2023-06-03:
Pack in .zip and attack here.

closed 2023-06-07 by yuhao71