T-Troll/alienfx-tools #352: Descriptions of Power Modes

cemkaya-mpi, 2023-06-07:
**Is your feature request related to a problem? Please describe.**
Not really a problem, but the power modes in fan control are G-mode, Manual and some numbers such as Level 160, Level 161 etc. 

**Describe the solution you'd like**
Instead of Level 160, it would help to see USTT_Balanced. Instead of Level 161, USTT_Performance and so on.

**Describe alternatives you've considered**
The following is from [another AWCC alternative](https://github.com/AlexIII/tcc-g15) project on github. @AlexIII is the maintainer and credit goes to him. He recovered the following enum from disassembling the original software:
```
public enum ThermalMode
{
    Custom = 0,                    
    Quiet = 150,
    Balanced = 151,                 
    Performance = 152,
    FullSpeed = 153,
    USTT_Balanced = 160,
    USTT_Performance = 161,
    USTT_Cool = 162,
    USTT_Quiet = 163,
    USTT_FullSpeed = 164,
    USTT_BatterySaver = 165,
    G_Mode = 171                   
}
```
Could these definitions be used in the fan control gui instead of Level 160 etc. ?

**System (please complete the following information):**
 - Model: Dell G15 5525
 - Model details: 4 Zone RGB Keyboard
 - OS: Windows 10

T-Troll, 2023-06-11:
You can just rename it as you wish - type, and it will be saved as a mode name.
Funny, he still not configured out G-mode is not a thermal only, but more.
I'll contact him about it.

cemkaya-mpi, 2023-06-14:
I could not save the name, but I am probably doing something wrong. I typed "Balanced" for Level 160 and pressed Save. I still see level 160. I am using version 8.2.7.0

T-Troll, 2023-06-14:
Ooops! It's a bug.
I confirm it, will fix ASAP.

T-Troll, 2023-06-14:
[alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11744907/alienfx-gui.zip)
Check this version, mode names will be saved now. BTW, doesn't need to press "save" every time.

closed 2023-06-14 by cemkaya-mpi