T-Troll/alienfx-tools #357: Alienware x17 R2 - Keyboard Lights not working after boot!

labelexa, 2023-06-17:
Hi,

Until yesterday, everything was working fine. Then, yesterday, I downloaded the latest version 8.3.0, the lights were working fine. _I **only** use the **Alien-fan.exe**_ to control my fans, so I don't open any other application. The keyboard lights were set to a green-ish color. I went to sleep, and today as I started Windows the keyboard lights were not lighting up. 

Important: when I press the power up button, all lights (keyboard and tron) lit up, the keyboard lights do the rainbow animation until the laptop boots to any operating system. (Yes, I tried an WinPE iso and Ubuntu.)

Using the alienfx-cli.exe, I get the following:
```asdf
alienfx-cli v8.2.6.2
Dell API not found, 2 devices found.
Device #0 - RGB Keyboard, VID#3426, PID#51900, APIv5, 84 lights
  Light ID#0 - ESC
  ...
Device #1 - Main lights, VID#6268, PID#1360, APIv4, 16 lights
  Light ID#0 - KB left
  ...
1 groups:
  Group #0 (84 lights) - All
Done.
```
Also on device manager I can see all the 5 HID vendor-defined devices.

Then I tried to set all the colors to green with:
```asdf
alienfx-cli.exe lowlevel
alienfx-cli.exe setall=0,255,0
alienfx-cli.exe highlevel
alienfx-cli.exe setall=0,255,0
```

And the tron lights change accordingly, however, the keyboard lights don't lit up.

Do you know what's happening and how can I fix this?

P.S.: I went back to version v8.2.6.2 

Thanks in advance

T-Troll, 2023-06-17:
Well... Rainbow at start means the light system was reset.
Seems like you have AWCC, which run at boot and update firmware.
Fan GUI does not interfere with lights any way, it has no code for it....

> Device #1 - Main lights, VID#6268, PID#1360, APIv4, 16 lights
>  Light ID#0 - KB left

You set light IDs from other device (something with 4-zone keyboard lights). I write BIG LETTERS into beginner's guide and even detection dialogue, it's a BAD idea. Now you need to clean all lights and set it up again using EXACTLY the same device profile or manual setup.

```
alienfx-cli.exe lowlevel
alienfx-cli.exe setall=0,255,0
alienfx-cli.exe highlevel
alienfx-cli.exe setall=0,255,0
```

It doesn't work this way. You need to combine parameters into one line, like `lowlevel setall=...`. 
BTW, high-level will only work if AWCC running, it utilizes Dell's library - and you don't have it running - `Dell API not found`.

labelexa, 2023-06-17:
I don't have and I never used anything other than alienfan-gui.exe. Well I tried both:
```
alienfx-cli.exe lowlevel setall=0,255,0
alienfx-cli.exe highlevel setall=0,255,0
```
As I said, the tron lights change as expected but the keyboard don't. I installed AWCC 😑 and it can't make the keyboard lights lit up too. 
```asdf
D:\alienfx-tools>alienfx-cli.exe highlevel setall=0,255,0
alienfx-cli v8.2.6.2
Dell API ready, 2 devices found.
Dell API selected
Done.
D:\alienfx-tools>alienfx-cli.exe lowlevel setall=0,255,0
alienfx-cli v8.2.6.2
Dell API ready, 2 devices found.
Device access selected
Done.
```
I will try to clean all lights and set all of them up again, but I don't know if I will know how to do or if it will work. I was expecting that after install this crapware (AWCC), the keyboard lights would lit up again.

closed 2023-06-18 by labelexa