T-Troll/alienfx-tools #359: M16 r1 Alien FX not detected

BattleBusted, 2023-06-26:
Hi,

Could you help me please in AWCC i'm no more able to adjust my light of my keyboard and also the light at the rear of my screen. Even with your soft. Look like a driver is missing. Do you have an idea ?

BattleBusted, 2023-06-26:
[image]

Is that normal I have only oone light ?

BattleBusted, 2023-06-26:
I installed your soft then i restart my pc and this happen ... Indeed I had two devices, after the reboot only one and my stadium and power button light off ... look like a driver missing. Could you please help me to fix it ?

I have M16 r1

T-Troll, 2023-07-08:
Check device manager - you should have Vendor-defined device across HID devices with VID_187c.
In case you lost it/have across "Unknown" - it's a hardware issue, i can't do anything abut it.

aliraza1944, 2023-08-24:
@T-Troll Is this the guide to share data about new hardware. https://github.com/T-Troll/alienfx-tools/wiki/How-to-collect-data-for-the-new-light-device

I havfe an alienware m16 R1.

T-Troll, 2023-08-25:
Yes. But you don't need to check color set - it's the same for m16/m18. I mostly interested at initial setup (then you just run AWCCService/change color profile) to pinpoint command for Fn row.

aliraza1944, 2023-09-11:
@T-Troll I am not sure if this is helpful but this is the csv for alienware m16 r1
[alienware_m16_keyboard.csv](https://github.com/T-Troll/alienfx-tools/files/12579815/alienware_m16_keyboard.csv)

T-Troll, 2023-09-12:
Thank you, it's very helpful. I'll include it into the next release!

HunterZ, 2023-11-10:
Here is a light mapping CSV for the Alienware m18 as well:
[m18.zip](https://github.com/T-Troll/alienfx-tools/files/13315429/m18.zip)

T-Troll, 2023-11-10:
Thank you, i'll add it into the next release!

luxexcudo, 2024-01-29:
Hi,
i got 2 Alienware m16 R1 AMD for me and my wife, we are Linux users and want to get rid of Windows.
So we have the problem not to be able to control our LED Stuff because of the unsupport for Linux from Dell.
I also Put a request to OpenRBG,.. and had to collect some information before.
Maybe you also have use for them,...

Name of device:
Dell Alienware m16 R1 AMD

Link to manufacturer's product page:
https://www.dell.com/de-de/shop/game/amd/spd/alienware-m16-r1-amd-laptop

Machine INFO:
https://linux-hardware.org/?probe=134a43f7ee

LEDs are controlled via USB
[Linux-HWInfo-Keyboard.txt](https://github.com/T-Troll/alienfx-tools/files/14084628/Linux-HWInfo-Keyboard.txt)
[Linux-HWInfo-USB.txt](https://github.com/T-Troll/alienfx-tools/files/14084629/Linux-HWInfo-USB.txt)
[Linux-LSUSB-Out.txt](https://github.com/T-Troll/alienfx-tools/files/14084630/Linux-LSUSB-Out.txt)
[Linux-LsUSBTree-Out.txt](https://github.com/T-Troll/alienfx-tools/files/14084631/Linux-LsUSBTree-Out.txt)

Additional Information:

The Laptop has 4 LED "Parts":

    the Logo on the lid
    on the backside the so called Arena
    The PowerButton ...
    The Powerbutton changes it's color only if you safe to default while the other Parts changes color imediatelly by assigning it
    ... The Powerbutton has 2 Main States: "Wired" and "on Battery"
    4.The Keyboard

... in because of USBCap/Wireshark Problems i was only able to capture the root hub of the devices, it includes the Bluetooth device which i have disabled before captureing ...

Captured Files:
[Logo_and_Arena.zip](https://github.com/T-Troll/alienfx-tools/files/14084640/Logo_and_Arena.zip)
[Logo_and_Arena.pdf](https://github.com/T-Troll/alienfx-tools/files/14084642/Logo_and_Arena.pdf)

[PowerButton_.zip](https://github.com/T-Troll/alienfx-tools/files/14084655/PowerButton_.zip)
[PowerButton.pdf](https://github.com/T-Troll/alienfx-tools/files/14084656/PowerButton.pdf)

[Keyboard - select .zip](https://github.com/T-Troll/alienfx-tools/files/14084662/Keyboard.-.select.zip)
[Keyboard.pdf](https://github.com/T-Troll/alienfx-tools/files/14084663/Keyboard.pdf)

... and at last the link to the OpenRGB Issue on Gitlab:

https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/3877

T-Troll, 2024-02-19:
Pal, i'm using Windows, and WSL enough for me.
All YOU need to do is the port light SDK (and, possibly, CLI) for Linux. It's quite easy, you only need to alter detection and IOCTRLs.

PS: Yes, OpenRGB can control tron (v4 API) from your gear, but not v5 (keyboard).