T-Troll/alienfx-tools #362: Unexplainable crash

XRone99, 2023-07-08:
**Describe the bug**
The program has been crashing inexplicably for a few days;
I have not updated the program nor windows; the only new thing I did was install Skype (the first program crash I noticed while making a video call)
It seems to be pretty randomic, sometimes it works fine for hours, sometimes it crashes as soon as i try to launch.

**To Reproduce**
It just happens some seconds after I launch the program 

**System (please complete the following information):**
 - Model: Alienware X17 R2
 - Model details: MX-Cherry KB, i7-12700H, RTX 3080Ti (Laptop), 16 GB DDR5
 - OS: Windows 11 stable 22H2

**Additional context**
I leave LOG in attached files

[LORENZO-ALIEN.LOG](https://github.com/T-Troll/alienfx-tools/files/11990923/LORENZO-ALIEN.LOG)

T-Troll, 2023-07-09:
Ugh-ogh.

First, did it happen with FX-Gui or Fan-GUI?
Second, seems like it's tooltip generation issue.

Can you please share your configuration (by running `aliengx-config backup` and send 4 .reg files it generates) to diagnose the source?

XRone99, 2023-07-09:
It happens with AlienFX-GUI
(Yes I noticed that crash happens when hovering on the icon, just before information are shown)

I'm gonna try to edit register and see what happens

[afx-fans.zip](https://github.com/T-Troll/alienfx-tools/files/11994797/afx-fans.zip)

XRone99, 2023-07-09:
Edited register, now it seems like it's not crashing anymore but now I no more have the Fan Control Sheet in the application :-(

T-Troll, 2023-07-09:
So... As i think, it's an issue with power mode or fan name (that's why disable fan control to make it work).
I check in your data what exactly (that's why i need it).
I'll return with a fixed version ASAP.

XRone99, 2023-07-09:
Thank you very much, you're always so much helpful!

T-Troll, 2023-07-09:
To make sure, please check this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11994829/alienfx-gui.zip)
Please enable Fan Control in settings for check first, then mouse over tooltip.
This helps me to understand what exactly happened.

XRone99, 2023-07-09:
After some little trouble on download (because windows defender wasn't collaborating LOL) I checked the version
It works fine now, it doesn't crash when I hover the cursor on the icon as it doesn't show the fan infos anymore
Fan control seems to work fine also

T-Troll, 2023-07-09:
Aha! So it was trouble with fan name or data...

Try this one - i add some checks about it - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/11994857/alienfx-gui.zip)

XRone99, 2023-07-09:
Well... that one crashed

T-Troll, 2023-07-09:
Not good anyway.... Issues with fan list. Ok, you can use it now, i'll check meanwhile.

XRone99, 2023-07-09:
By the way, I don't know if this is important but I say it anyway, I just realized that after the last reset of my pc I forgot to copy DPTF content into system32 directory
I don't know if this could be involved 
What I know is that before reset I had files in correct directories but FX-GUI never managed to show names on the sensors list

T-Troll, 2023-07-09:
I mention it into your config.
You can:
1. Copy files
2. Close Fan/FX GUI
3. Open `regedit`, navigate `HKCU/Software/Alienfan`
4. Change `DPTF` to 1
5. Start GUI

It will be 2-5 second delay at start, but it should get ESIF names.

XRone99, 2023-07-09:
Strange behaviour: I just edited DPTF in Register and after that started GUI, Fan control disabled by itself apparently
So I closed again gui and went back to reg/software/alienfxgui/FanControl and put it back to 1 again but still no fan control was working
So I went again into register, disabled DPTF an re-enabled FanControl (yes after updating DPTF value to 0 also FanControl was turned 0) and now it's working again but without DPTF

T-Troll, 2023-07-09:
Thanks, let me check.

XRone99, 2023-08-31:
Hi, this morning I tried to check if 8.4.0 version still has the same old problem of crashing on mouse hovering... and it has
Could you make it a selectable option perhaps?
So i can update from 8.3.0 without crash and without needing to ask you for a special version everytime?

Thank you in advance, you're alway so kind!

T-Troll, 2023-09-03:
Ohh... Seems like some desktops have a mess into hardware fan configuration i can't pinpoint yet.
Wait a moment, i can provide a debug version for you, to check what happened into DbgView.

T-Troll, 2023-09-03:
Here we are: [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/12505400/alienfx-gui.zip)
Please:
1. Download DbgView from [here](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview)
2. Unpack, run
3. Run this GUI version
4. You will see some debug into dbgview already
5. Move mouse to icon, crash ^_^
6. Share what you have into DbgView

XRone99, 2023-09-03:
[LORENZO-ALIEN 030923.LOG](https://github.com/T-Troll/alienfx-tools/files/12505443/LORENZO-ALIEN.030923.LOG)
 Here it is!

T-Troll, 2023-09-03:
Ohshi....
Here it is:
```
Printing done, length 128
```
Max. tooltip length - 127 (Windows limit). I made a check about it, but this is not enough.... 
So..... As a quick fix, rename your profile (make it 2 chars shorter). It will stop crash.
I'll fix this somehow into the next release, just need to time to think what to drop from in into this situation.

XRone99, 2023-09-03:
Incredible, i changed name from "Predefinito" to "Default" and now it doesn't crash anymore!!

T-Troll, 2023-09-03:
Yes. As i tell, in your case it generates loooong tooltip, and it is not handled correctly by Windows.
So i will limit it correctly and may drop some info in this case at next release.

closed 2023-09-25 by T-Troll