T-Troll/alienfx-tools #363: Fan Control holds at 6000 RPM max.

davies-w, 2023-07-15:
**Describe the bug**
Can't reach max RPMs without unchecking sensors.

**To Reproduce**
Steps to reproduce the behavior:
1. Fan Control GUI
2. click CPU Fan, select sensor. 
3. Edit Curve to Max Boost at zero C
4. CPU will not exceed Boost of 100 (on Chart), and caps RPM at ~6K.

However, if you uncheck the sensor, the Fans will spin up to their max RPM, @ 8000RPM, cooling significantly.
Boost shows at 255 on the chart.

**Expected behavior**
I'd expect Max RPMs at Max Boost

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
It's a 2020 Alienware  laptop  (m15 r2?)

**Additional context**
The sound of the fans on AWCC at max settings suggest they hit ~8K, so I was surprised that this program wouldnt let it.

davies-w, 2023-07-15:
Thank you though, I hate AWCC with a passion!

davies-w, 2023-07-15:
Found this from Reddit BTW, in case its useful ! https://www.reddit.com/r/Alienware/comments/tqn9ql/alternative_fan_control_program_for_alienware/

T-Troll, 2023-07-17:
Hmm.... Seems like i have a bug, but in the other. I check.

Boost 255 mean you have Performance mode enabled. It provides more RPMs in common (among other changes), but can't be controlled.
You MAYBE can reach the same using "check max boost" button.

Screen scale is in boost percent, always from 0 to 100. See "scale" at top of it to define real max. boost possible.